notro/fbtft #215: Support for Waveshare 3.5 Spotpear

sudobrew, 2015-01-05:
Hello,
I have been attempting to set up a Waveshare 3.5 Spotpear , but all I have been getting is a blank bright white screen. I'm was hoping someone with a bit more experience would have more insight.

http://www.amazon.com/Waveshare-Raspberry-Resistive-Interface-Rapsberry-pi/dp/B00OZLG2YS

From the manufacturer:
1,17 - 3.3V
2,4 - 5V
3,5,7,8,10,12,13,15,16 - NC
6,9,14,20,25 - GND
11 - TP_IRQ (Touch Panel interrupt)
18 - LCD_RS (Instructions / Data Register Selection)
19 - LCD_SI / TP_SI (SPI data input)
21 - TP_SO (SPI data output)
22 - RST (Reset)
23 - LCD_SCK / TP_SCK (SPI clock)
24 - LCD_CS (LCD chip selection)
26 TP_CS (Touch Panel chip selection)

I have tried a few variations of:
sudo modprobe fbtft_device custom name=fb_ili9341 speed=48000000 gpios=dc:18,reset:22 bgr=1 rotate=270 width=480 height=320

But so far all I get is the HDMI to freeze and the LCD remains bright white.

I was following this forum: https://github.com/notro/rpi-firmware/issues/6 but the solutions provided don't seem to work with this different display. I have the image provided from the manufacturer, but I would rather use the standard Raspian image.

sudobrew, 2015-01-05:
I was thinking, since I have the custom image from waveshare, is there an easy way to extract the initialization sequence if I temporarily load the image on the raspberry pi?

notro, 2015-01-06:
Do you know if they use fbtft?
Try:
```
$ dmesg | grep graphics
graphics fb1: XXX frame buffer, 480x272, 255 KiB video memory, fps=20
```
If they do, what does ```lsmod``` show?

sudobrew, 2015-01-06:
Thank you for the response
```
$dmesg | grep graphics
fb1: flexb frame buffer, 480x320, 300kiB video memory, 4 kiB DMA buffer memory, fps=20, spi0.0 at 16MHz
```
```
$lsmod
joydev 0 
ads7846 0
ads7846_device 0
fbtft_device 0
flexfb 3
fbtft 2 (flexfb, fbtft_device)
syscopyarea 1 (fbtft)
sysfillrect 1 (fbtft)
sysimgblt 1 (fbtft)
fb_sys_fops 1 (fbtft)
i2c_dev 0
snd_bcm2835 0
snd_soc_pcm512x 0
snd_soc_wm8804 0
snd_soc_bcm2708_i2s 0
regmap_mmio 1 (snd_soc_bcm2708_i2s)
snd_soc_core 3 (snd_soc_pcm512x,snd_soc_wm8804,snd_soc_bcm2708_i2s)
snd_compress 1 (snd_soc_core)
regmap_i2c 2 (snd_soc_pcm512x,snd_soc_wm8804,snd_soc_core)
snd_pcm_dmaengine 1 (snd_soc_core)
regmap_spi 3 (snd_soc_pcm512x,snd_soc_wm8804,snd_soc_core)
evdev 4
snd_pcm 3 (snd_bcm2835, snd_soc_core, snd_pcm_dmaengine)
snd_page_alloc 1 (snd_pcm)
snd_seq 0
snd_seq_device 1 (snd_seq)
snd_timer 2 (snd_pcm,snd_seq)
leds_gpio 0
led_class 1 (leds_gpio)
snd 7 (snd_bcm2835, snd_soc_core, snd_timer, snd_pcm, snd_eq, snd_seq_device, snd_compress)
spi_bcm2708 0
i2c_bcm2708 0
```

I then ran:
```
$dmesg | grep spi

fbtft_request_gpios: 'reset' =GPIO025
fbtft_request_gpios: 'dc' =GPIO024
flexfb_verify_gpios_dc()
fbtft_init_display()
fbtft_reset()
init: write(0xB0) 0x00
init: write(0x11)
init: mdelay(250)
init: write(0x3A) 0x55
init: write(0xC2) 0x44
init: write(0xC5) 0x00 0x00 0x00 0x00
init: write(0xE0) 0x0F 0x1F 0x1C 0x0C 0x0F 0x08 0x48 0x98 0x37 0x0A 0x13 0x04 0x11 0x0D 0x00
init: write(0xE1) 0x0F 0X21 0x2E 0x0B 0x0D 0x05 0x47 0x75 0x37 ox06 0x10 0x03 0x24 0x20 0x00
init: write(0xE2) 0x0F 0X21 0x2E 0x0B 0x0D 0x05 0x47 0x75 0x37 ox06 0x10 0x03 0x24 0x20 0x00
init: write(0x36) 0x28
init: write(0x11)
init: write(0x29)
Display update: 1691kB/s (177.354ms),fps=0 (0.000 ms) 
fbtft_register_backlight()
fbtft_register_backlight(): led pin not set, exiting
graphics fb1: flexb frame buffer, 480x320, 300kiB video memory, 4 kiB DMA buffer memory, fps=20, spi0.0 at 16MHz
fbtft_device: spidev spi0.1 500kHZ 8 bits mode=0x00
fbtft_device: flexfb spi0.0 16000kHZ 8 bits mode=0x00
ads7846_device: Deleting spi0.1
ads7846 spi0.1: touchscreen, irq 187
input ADS7846 Touchscreen as /devices/platform/bcm2708_spi.0/spi_master/spi0/spi0.1/input/input1
```

Interesting to see that they use 24 and 25 which is different than their documentation

fergie4000, 2015-01-06:
```
modprobe flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
modprobe fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 gpios=reset:25,dc:24
```

sudobrew, 2015-01-06:
Thank you fergie4000! That worked perfectly!

I grabbed the config.txt and cmdline.txt from the factory image and added these lines to my configs:

config.txt
```
gpu_mem=128
```
cmdline.txt
```
fbcon=map:1 fbcon=font:ProFont6x11
```

fergie4000, 2015-01-06:
Not a problem. Recently went through this trying to get the same screen working on my Raspi.

kurosov, 2015-01-06:
just got this display myself and having a little trouble setting it up on raspbian.

Before i get into it does anybody know if raspbmc will be able to work fine with this display as I'm hoping to stick a small display on a raspbmc pi in the future.

kurosov, 2015-01-08:
Ok, i've got this screen working with the command line but attempting startx gives the error message "Fatal server error: No screens found".

Not sure what i've missed.

kurosov, 2015-01-09:
I now have this display running on both Raspbian and  Raspbmc although often the Raspbmc display goes a bit funny. I'm trying to sort that out before i even attempt to get touch working on Raspbmc.

drkrieger01, 2015-01-14:
Hello folks :)
I just purchased one of these Waveshare 3.5" displays and am also having issues getting it running on my B+. I'm running Rasberian as I do not want to use the supplied image. 
I tried running the commands in the shell that fergie4000 suggested, but still just get a blank white display. 
Is there something else I need to do? I've run a firmware update and verified that fbtft is installed and up to date.

sudobrew, 2015-01-14:
Here are the steps that I took to get it working. The screen should start working at step 5 when you do the test.

1.Expand the filesystem
```
sudo raspi-config
```
2.Change fb0 to fb1
```
sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf
```
3.Enable SPI if needed
```
# remove or comment out the spi blacklist line
sudo nano /etc/modprobe.d/raspi-blacklist.conf
```
4.FBTFT drivers as loadable modules
```
sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update
sudo reboot
```
5.Test
```
sudo modprobe flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
sudo modprobe fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 gpios=reset:25,dc:24
```
6.Perminantly add Driver
```
#Add to file /etc/modules
flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 gpios=reset:25,dc:24
```
7.Additional settings
```
#config.txt
gpu_mem=128
```
```
#cmdline.txt
fbcon=map:1 fbcon=font:ProFont6x11
```

drkrieger01, 2015-01-14:
When I run the second command in step 5, I get this error:

ERROR: could not insert 'fbtft_device': Invalid argument

Any ideas?

notro, 2015-01-14:
This will probably tell you why:
```
$ dmesg | tail
```
fbtft_device writes messages to the kernel log.

drkrieger01, 2015-01-14:
Sorry for delay, had to run to work. 
This is what I get: 
dmesg | tail
[   26.790736] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   30.811120] bcm2835-cpufreq: switching to governor ondemand
[   30.811160] bcm2835-cpufreq: switching to governor ondemand
[   35.085853] wlan0: authenticate with 6c:aa:b3:11:ac:38
[   35.174888] wlan0: send auth to 6c:aa:b3:11:ac:38 (try 1/3)
[   35.176448] wlan0: authenticated
[   35.178121] wlan0: associate with 6c:aa:b3:11:ac:38 (try 1/3)
[   35.183354] wlan0: RX AssocResp from 6c:aa:b3:11:ac:38 (capab=0x431 status=0 aid=1)
[   35.198350] wlan0: associated
[   35.323892] Adding 102396k swap on /var/swap.  Priority:-1 extents:1 across:102396k SSFS

Also, would it matter if I was running this in a putty session? Will it still enable the screen on the unit?

notro, 2015-01-14:
You have to load fbtft_device first (sudo modprobe ...). Then run dmesg and give me the part containing fbtft related messages (skip tail, it might drop the first messages).

notro, 2015-01-14:
> Also, would it matter if I was running this in a putty session? Will it still enable the screen on the unit?

That works fine.

drkrieger01, 2015-01-14:
For some reason my OS isn't booting anymore, I'm going to do a fresh Raspberian install to the SD card and try the steps Sudobrew has listed. I was trying them in the exact order listed, but still hung on on line two in step 5 with that error message.

drkrieger01, 2015-01-15:
And it appears that I was running NOOBS. I'm going to reload it with Raspberian only.

drkrieger01, 2015-01-15:
Well, I have made some progress... the screen is no longer white, it is now black (backlight is on). I can see it initalize when booting up the unit, just doesn't show anything on it. 

Here's a full dmesg for you. Any ideas?
Incoming wall of text: 

dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.12.34+ (pi@raspi2) (gcc version 4.8.3 20140106 (p                            rerelease) (crosstool-NG linaro-1.13.1-4.8-2014.01 - Linaro GCC 2013.11) ) #1 PR                            EEMPT Sun Dec 7 22:39:06 CET 2014
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr                            =00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruc                            tion cache
[    0.000000] Machine: BCM2708
[    0.000000] cma: CMA: reserved 8 MiB at 17800000
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] On node 0 totalpages: 98304
[    0.000000] free_area_init_node: node 0, pgdat c0662838, node_mem_map c071b00                            0
[    0.000000]   Normal zone: 768 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 98304 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pag                            es: 97536
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1680                             bcm2708_fb.fbheight=1050 bcm2708.boardrev=0x10 bcm2708.serial=0xccdc23b4 smsc95x                            x.macaddr=B8:27:EB:DC:23:B4 bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708                            .disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0                            x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,11520                            0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 374204K/393216K available (4636K kernel code, 266K rwdata                            , 1476K rodata, 147K init, 710K bss, 19012K reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xd8800000 - 0xff000000   ( 616 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xd8000000   ( 384 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc06003d8   (6113 kB)
[    0.000000]       .init : 0xc0601000 - 0xc0625e84   ( 148 kB)
[    0.000000]       .data : 0xc0626000 - 0xc0668a38   ( 267 kB)
[    0.000000]        .bss : 0xc0668a44 - 0xc071a328   ( 711 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:394
[    0.000000] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 4                            294967ms
[    0.000000] Switching to timer-based delay loop
[    0.000000] Console: colour dummy device 80x30
[    0.000000] console [tty1] enabled
[    0.001170] Calibrating delay loop (skipped), value calculated using timer fr                            equency.. 2.00 BogoMIPS (lpj=10000)
[    0.001227] pid_max: default: 32768 minimum: 301
[    0.001738] Mount-cache hash table entries: 512
[    0.002577] Initializing cgroup subsys memory
[    0.002685] Initializing cgroup subsys devices
[    0.002723] Initializing cgroup subsys freezer
[    0.002755] Initializing cgroup subsys blkio
[    0.002902] CPU: Testing write buffer coherency: ok
[    0.003349] Setting up static identity map for 0xc046b378 - 0xc046b3d4
[    0.006608] devtmpfs: initialized
[    0.022740] VFP support v0.3: implementor 41 architecture 1 part 20 variant b                             rev 5
[    0.024323] NET: Registered protocol family 16
[    0.029670] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.030471] cpuidle: using governor ladder
[    0.030524] cpuidle: using governor menu
[    0.030958] bcm2708.uart_clock = 0
[    0.032773] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.032828] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.032862] mailbox: Broadcom VideoCore Mailbox driver
[    0.032958] bcm2708_vcio: mailbox at f200b880
[    0.033082] bcm_power: Broadcom power driver
[    0.033121] bcm_power_open() -> 0
[    0.033146] bcm_power_request(0, 8)
[    0.533872] bcm_mailbox_read -> 00000080, 0
[    0.533915] bcm_power_request -> 0
[... 168 lines cut ...]
[   19.670932] flexfb spi0.0: init: write(0x11)
[   19.680658] flexfb spi0.0: init: mdelay(250)
[   19.999647] flexfb spi0.0: init: write(0x3A) 0x55
[   20.009899] flexfb spi0.0: init: write(0xC2) 0x44
[   20.020022] flexfb spi0.0: init: write(0xC5) 0x00 0x00 0x00 0x00
[   20.031411] flexfb spi0.0: init: write(0xE0) 0x0F 0x1F 0x1C 0x0C 0x0F 0x08 0x                            48 0x98 0x37 0x0A 0x13 0x04 0x11 0x0D 0x00
[... cut ...]
[   20.379425] fbtft_device:      flexfb spi0.0 16000kHz 8 bits mode=0x00
[   26.664730] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data                             may be corrupt. Please run fsck.
[   30.326184] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   31.906097] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE                            1
[   36.917500] Adding 102396k swap on /var/swap.  Priority:-1 extents:2 across:2                            134012k SSFS

drkrieger01, 2015-01-15:
Update: So, the GUI works (if I run startx). But no display while outside of the gui. 

Is there any way to display the system when not in gui?

drkrieger01, 2015-01-15:
Also, no touchscreen function. I know it uses the XPT2046 touch controller

notro, 2015-01-15:
**Console**
Move console 1 to framebuffer 1: con2fbmap 1 1
Permanent: https://github.com/notro/fbtft/wiki/Boot-console#enable-console

**Touch**
Example for a display with the same controller:
```
sudo modprobe ads7846_device model=7846 cs=1 speed=2000000 gpio_pendown=25 pressure_max=255 x_plate_ohms=60
```
https://github.com/notro/fbtft_tools/wiki/ads7846_device
https://github.com/notro/fbtft/wiki/Touchpanel

drkrieger01, 2015-01-15:
Great! Thank you very much for the help Notro and Sudobrew!

pykoon, 2015-01-21:
Thank you for your help, I have the same Waveshare 3.5 Spotpear screen and I have 2 questions : 
- How do you set the touch exactly with this display (3.5" 320x480)?

- How can I rotate my X11 session to 270°?

pykoon, 2015-01-24:
I really need your help. I can't rotate my screen and I don't know how do it. 
My /etc/modules is like this (same on the top af this thread) :
    
    flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3

    fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 gpios=reset:25,dc:24
    
When I try to change rotate to rotate=180 or 270 I can see that something change on my screen if width and height don't change but each time, X11 is load in the same direction. 

Please help me, please...

Notro, you made an raspian image for other screen. Do you have one for this screen ?

notro, 2015-01-24:
The flexfb driver doesn't have builtin rotation support. You have to change the init sequence as well as providing the rotation parameter (or just take care of swapping width/height yourself).
width=320  height=480 and rotate=90 is equivalent to width=480  height=320 and rotate=0

Register 0x36 controls rotation
```
init=[...],-1,0x36,0x28,[...]
```
I believe this is a MIPI compatible register, so this driver can be used as an example for the various bit fields: https://github.com/notro/fbtft/blob/master/fb_ili9341.c#L100
The values for this register can differ between panels.

You can try asking the seller for the values.

pykoon, 2015-01-25:
Notro, you are my god.

I wasn't easy, I tried a lot of value because it depend of width, height and rotate in fbtft_device.
I found the good value : register 0x36, value 0x48 with rotate = 0 in fbtft_device.

    flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x48,-1,0x11,-1,0x29,-3

    fbtft_device debug=3 rotate=0 name=flexfb speed=16000000 gpios=reset:25,dc:24

Now I have to configure touchscreen. drkrieger01 or sudobrew, could you explain how did you configure it?

pykoon, 2015-01-26:
To finish, here touchscreen config : 
A good link : http://cooljc.me.uk/?page_id=14

    sudo apt-get -y install xinput evtest libts-bin

Edit /etc/modules

    # Touchscreen
    ads7846_device model=7846 cs=1 gpio_pendown=17  keep_vref_on=1 swap_xy=0 pressure_max=255 x_plate_ohms=60 x_min=200  x_max=3900 y_min=200 y_max=3900

If you want a vertical mode, you can take this configuration :

pi@rasp ~ $ sudo nano /usr/share/X11/xorg.conf.d/99-calibration.conf 
    Section "InputClass"
	Identifier	"calibration"
	MatchProduct	"ADS7846 Touchscreen"
	Option	"Calibration"	"3869 178 3903 270"
    EndSection

If you need to recalibrate for other orientation, you can use xinput_calibrator : 

    wget http://adafruit-download.s3.amazonaws.com/xinput-calibrator_0.7.5-1_armhf.deb
    sudo dpkg -i -B xinput-calibrator_0.7.5-1_armhf.deb

Before to start it, you have to delete (if exist) /usr/share/X11/xorg.conf.d/99-calibration.conf. 

    sudo reboot.
    DISPLAY=:0.0 xinput_calibrator

After calibration, this tools give you informations usefull to create you own  /usr/share/X11/xorg.conf.d/99-calibration.conf witdh your calibration.

Enjoy,

twkrol, 2015-02-15:
Thanks for this thread! It works for Waveshare 4inch LCD on raspberry pi v2.
You saved my day :)

mikertr, 2015-02-17:
@twkrol can you speak to specifically which instructions you followed?  I'm also working on a Waveshare 4inch LCD with RPi 2.  I followed the instructions (twice) with a fresh install of Raspbian, and after reboot I can't hit the pi via SSH or get the HDMI display to show me anything.  So clearly, I'll be starting over again.  Ha!

twkrol, 2015-02-17:
@mikertr I've follwed instructions from http://www.circuitbasics.com/setup-lcd-touchscreen-raspberry-pi/#content.
But use flexfb device instead of waveshare32b. Keep the configuration of touch, but have to calibrate or rotate screen to keep it as is.

Add modules in /etc/modules as below:
 
    flexfb width=480 height=320 regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3 

    fbtft_device debug=3 rotate=0 name=flexfb speed=16000000 gpios=reset:25,dc:24

Changing screen rotation is a bit tricky but possible. You have to modify value of register 0x36 at the end of init string. In the example above there is 0x28, as well as rotate=0 below. This is for horizontal screen with power socket on upper side, touch inverted.
You can try these combinations:

Reg 0x36 value | rotate value | result
---|---|---
0x28|0|Horizontal (power socket on top)
0xF8|0|Horizontal (power socket on bottom side), touch works well
0x48|90|Vertical
0x84|90|Vertical (opposite)

Above values are found by testing, may not be the best settings but works :)

tony486, 2015-02-17:
@twkrol Thanks
And can you tell me what do you have in your /boot/cmdline.txt file? because when I modify mine, my rpi2 won't boot anymore

twkrol, 2015-02-17:
Here it is:
    
    dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbtft_device.custom fbtft_device.name=flexfb fbtft_device.gpios=dc:24,reset:25 fbtft_device.bgr=1 fbtft_device.speed=16000000 fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo dma.dmachans=0x7f35 console=tty1 consoleblank=0 fbtft_device.fps=50 fbtft_device.rotate=0

tony486, 2015-02-17:
@twkrol 
Thanks a lot :)

Just one more thing: can I have display on both HDMI & LCD screen at the same time?

mikertr, 2015-02-17:
@twkrol Thank you for the write up and quick reply.  I'm hoping to give this a shot in the next day or two.  Will report back.

tony486, 2015-02-17:
Thanks my screen works great but I don't know why, I can't run any program that uses X11... Emulationstation & kodi for exemple won't start on the touch screen... I can only use cmdline

pykoon, 2015-02-17:
Hi, 

Anybody knows how turn off backlight? I would like to turn off after 10 min. I tried some tips but nothing works.
Tx,

tony486, 2015-02-18:
@twkrol 

Whait is your touch screen calibration config ? Can u give your /etc/modules file plz?

twkrol, 2015-02-18:
Hi
Please find my /etc/modules entry for touch device

    ads7846_device model=7846 cs=1 gpio_pendown=17 speed=1000000 keep_vref_on=1 swap_xy=1 pressure_max=255 x_plate_ohms=60 x_min=200 x_max=3900 y_min=200 y_max=3900

thodoris85, 2015-02-18:
I followed @sudobrew's  steps/instructions and I also had the issue of "ERROR: could not insert 'fbtft_device': Invalid argument"

In case anyone else has the same issue, just to let you know that

>3.Enable SPI if needed
>
>'# remove or comment out the spi blacklist line
>sudo nano /etc/modprobe.d/raspi-blacklist.conf

is two different steps.
In order to enable the SPI, you will have to change it from "sudo raspi-config"

Thanks to @notro for his great work and to @sudobrew for summing it up!

tony486, 2015-02-19:
@twkrol
Thank you but it's strange ... Even with your etc/modules file my touch screen isn't calibrated...

notro, 2015-02-20:
@pykoon 
> Anybody knows how turn off backlight? I would like to turn off after 10 min. I tried some tips but nothing works.

For console the backlight turns off on the second blanking event, not the first.
There has been a change to the backlight core that isn't reflected in fbtft.
https://github.com/notro/fbtft/wiki/Boot-console#console-blanking

This is a workaround for X windows: https://github.com/notro/fbtft-spindle/wiki/Appendix#x-windows-backlight-script

pykoon, 2015-02-20:
@notro 
Tx for your response.

However, it doesn't work on my pi;
I followed your guide "X windows backlight script" and now, I have a pretty screen saver but blanking doesn't work.
I'm note sure but I have nothing in /sys/class/backlight so I don't understand how "echo 4 | sudo tee /sys/class/backlight/*/bl_power 1>/dev/null" will work.

Do you have an idea of my problem?

Tx,

notro, 2015-02-20:
Do you have anything in /sys/class/backlight ?
The 'led' gpio enables FBTFT backlight control.
I remember someone asking about a Waveshare display and backlight, maybe there isn't any backlight control on this display?

pykoon, 2015-02-20:
@notro 

I have nothing in /sys/class/backlight.
I already tried to contact waveshare but they don't respond. I will try again.

Tx for your analyze.

Sainthobson, 2015-02-21:
Thanks everyone for the effort in putting this together, I was able to successfully get my screen to display but as soon as I did this command:

sudo apt-get -y install xinput evtest libts-bin

and rebooted my screen no longer works it only just flashes, sometimes i see a faint image of my desktop but then it disappears.  Any ideas?  I was just trying to get the touch screen functionality to work and now I can't even return to the way it was before.

jharig23, 2015-02-22:
I had a lot of trouble getting the 'Waveshare Spotpear 3.5 A' touch screen to work.  The controller is XPT2046, which is the same as ads7843, not ads7846!  Change the device_model argument to `7843`, and it works!

> ads7846_device model=7843 irq=187 cs=1 gpio_pendown=17  keep_vref_on=1 swap_xy=0 pressure_max=255 x_plate_ohms=60 x_min=200  x_max=3900 y_min=200 y_max=3900

Sainthobson, 2015-02-22:
Hm I changed the device to 7843 but it didn't appear to make an ounce of difference, no matter what configuration I do from twkrol's post I always end up with one axis which is backwards.

I have the waveshare spotpear 3.5 no idea about a letter after it.

mikertr, 2015-02-25:
@twkrol I've attempted to follow the steps you took and seem to run off the rails on step one, switching fb0/fb1 in /usr/share/X11/xorg.conf.d/99-fbturbo.conf.  Upon reboot, I consistently get a kernel panic message (on my HDMI display, mind you.  Not the LCD).  Last line says "end Kernel panic - not syncing: Fatal exception in interrupt".

I detached the LCD and followed the rest of the steps, hoping to re-attach the LCD and reboot.  Upon doing so, my kernel panic message is "not syncing: VFS: Unable to mount root fs on unknown-block (179,6).

I'm about to dropkick this thing into a nearby lake.

mikertr, 2015-02-25:
@twkrol update - a silly error on my part in the cmdline file earned me the second kernel panic message (got some bad advice on another forum), so I feel like I've at least made some progress.  Upon reboot, I'm not seeing either kernel panic message.

When booting, I now receive:

Loading kernel module snd-bcm2835.
Loading kernel module flexfb.
Loading kernel module fbtft_device.
fbtft_device: SPI devices registered:
fbtft_device:   'fb' Platform devices registered:
fbtft_device:     bcm2708_fb id=-1 pdata? no
fbtft_device: spi_nusnum_to_master(0) returned NULL
fbtft_devicee: failed to register SPI device
ERROR:  could not insert 'fbtft_device': Invalid argument

I feel like I'm making some progress but missing something fairly obvious.  Thoughts?

twkrol, 2015-02-25:
@mikertr Let's try on fresh raspian install. Do you have enabled the SPI in raspian config?
If the problem presists, please post your /etc/modules for review.

mikertr, 2015-02-25:
Attempt on fresh raspbian install gave me the same results, though I fear I'm probably making the same mistakes.  Here's my /etc/modules

```
snd-bcm2835

flexfb width=480 height=320 regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3 

fbtft_device debug=3 rotate=0 name=flexfb speed=16000000 gpios=reset:25,dc:24
```

twkrol, 2015-02-26:
@mikertr 
Go add in your /etc/modules the following line:

    spi-bcm2708

just after 

    snd-bcm2835

mikertr, 2015-02-26:
I was able to boot up with this change, but the only thing working on the LCD is the backlight.  I scoured the boot log and couldn't see where anything went wrong.  For grins, here's my boot log (apologies to those of you cursing me while scrolling):

```
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.5-v7+ (pi@raspi2) (gcc version 4.8.3 20140106 (prer$
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cac$
[    0.000000] Machine model: Raspberry Pi 2 Model B
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 8085ca00, node_mem_map ba093000
[    0.000000]   Normal zone: 1888 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (8620->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 11 pages/cpu @ba05d000 s12864 r8192 d24000 u45056
[    0.000000] pcpu-alloc: s12864 r8192 d24000 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 2$
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1280 bcm27$
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 940460K/966656K available (5902K kernel code, 404K rwdata, 187$
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff000000   (1080 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x807a0b04   (7779 kB)
[    0.000000]       .init : 0x807a1000 - 0x80806000   ( 404 kB)
[    0.000000]       .data : 0x80806000 - 0x8086b06c   ( 405 kB)
[    0.000000]        .bss : 0x8086b06c - 0x8092f06c   ( 784 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:480
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (virt).
[    0.000015] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 3579139424$
[    0.000038] Switching to timer-based delay loop, resolution 52ns
[    0.000328] Console: colour dummy device 80x30
[    0.001956] console [tty1] enabled
[    0.002006] Calibrating delay loop (skipped), value calculated using timer frequen$
[    0.002088] pid_max: default: 32768 minimum: 301
[    0.002510] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002573] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003864] Initializing cgroup subsys memory
[    0.003952] Initializing cgroup subsys devices
[    0.004004] Initializing cgroup subsys freezer
[    0.004054] Initializing cgroup subsys net_cls
[    0.004118] Initializing cgroup subsys blkio
[    0.004238] CPU: Testing write buffer coherency: ok
[    0.004353] ftrace: allocating 20192 entries in 60 pages
[    0.054273] CPU0: update cpu_capacity 1024
[    0.054350] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.054389] [bcm2709_smp_prepare_cpus] enter
[    0.054555] Setting up static identity map for 0x552d78 - 0x552dd0
[    0.114246] [bcm2709_boot_secondary] cpu:1 started (0) 18
[    0.114555] CPU1: Booted secondary processor
[    0.114564] [bcm2709_secondary_init] enter cpu:1
[    0.114619] CPU1: update cpu_capacity 1024
[    0.114628] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.134183] [bcm2709_boot_secondary] cpu:2 started (0) 18
[    0.134425] CPU2: Booted secondary processor
[    0.134432] [bcm2709_secondary_init] enter cpu:2
[    0.134464] CPU2: update cpu_capacity 1024
[... 169 lines cut ...]
[    4.109534] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.126358] usb 1-1.4: Product: 802.11n WLAN Adapter
[    4.139599] usb 1-1.4: Manufacturer: Realtek
[    4.152174] usb 1-1.4: SerialNumber: 00e04c000001
[    4.260653] usb 1-1.5: new low-speed USB device number 6 using dwc_otg
[    4.421320] usb 1-1.5: New USB device found, idVendor=413c, idProduct=2003
[    4.436141] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    4.436153] usb 1-1.5: Product: DELL USB Keyboard
[    4.436162] usb 1-1.5: Manufacturer: DELL
[    4.489453] input: DELL DELL USB Keyboard as /devices/platform/bcm2708_usb/usb1/1-$
[    4.509664] hid-generic 0003:413C:2003.0002: input,hidraw1: USB HID v1.10 Keyboard$
[    4.534917] udevd[174]: starting version 175
[    4.562080] input: DELL DELL USB Keyboard as /devices/platform/bcm2708_usb/usb1/1-$
[    4.583247] hid-generic 0003:413C:2003.0003: input,hidraw2: USB HID v1.10 Device [$
[    5.096181] bcm2708_spi 3f204000.spi: DMA channel 2 at address 0xf3007200 with irq$
[... cut ...]
[    5.235756] bcm2708_spi 3f204000.spi: SPI Controller at 0x3f204000 (irq 80)
[    5.252774] bcm2708_spi 3f204000.spi: SPI Controller running in dma mode
[    5.303232] random: nonblocking pool is initialized
[    5.441409] usbcore: registered new interface driver rtl8192cu
[    7.084831] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    7.415586] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   12.605002] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   19.050961] Adding 102396k swap on /var/swap.  Priority:-1 extents:2 across:216264$
```

mikertr, 2015-03-01:
@twkrol Progress!!  Thank you for answering questions.  My display will inconsistently show boot messages for a short while, eventually start a slow flicker, and ultimately go to just a backlight.  I've read that some of the LCD issues can be power-related, but I'm using a 5 volt, 2 amp power supply.  Here's the end of my boot log:

```
[    4.681631] bcm2708_spi 3f204000.spi: DMA channel 2 at address 0xf3007200 wi$
[    4.707234] bcm2708_spi 3f204000.spi: DMA channel 4 at address 0xf3007400 wi$
[    4.795012] bcm2708_spi 3f204000.spi: SPI Controller at 0x3f204000 (irq 80)
[    4.806292] bcm2708_spi 3f204000.spi: SPI Controller running in dma mode
[    4.954272] random: nonblocking pool is initialized
[    5.194456] usbcore: registered new interface driver rtl8192cu
[    6.626059] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    7.021541] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    7.656646] fbtft_device:  SPI devices registered:
[    7.664539] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[    7.673889] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[    7.683234] fbtft_device:  'fb' Platform devices registered:
[    7.691823] fbtft_device:      bcm2708_fb id=-1 pdata? no
[    7.700047] fbtft_device: Deleting spi0.0
[    7.707968] flexfb spi0.0: fbtft_request_gpios: 'reset' = GPIO25
[    7.716935] flexfb spi0.0: fbtft_request_gpios: 'dc' = GPIO24
[    7.725579] flexfb spi0.0: flexfb_verify_gpios_dc()
[    7.733047] flexfb spi0.0: fbtft_init_display()
[    7.739985] flexfb spi0.0: fbtft_reset()
[    7.866317] flexfb spi0.0: init: write(0xB0) 0x00
[    7.873526] flexfb spi0.0: init: write(0x11)
[    7.880157] flexfb spi0.0: init: mdelay(250)
[    8.136609] flexfb spi0.0: init: write(0x3A) 0x55
[    8.143569] flexfb spi0.0: init: write(0xC2) 0x44
[    8.150435] flexfb spi0.0: init: write(0xC5) 0x00 0x00 0x00 0x00
[    8.158582] flexfb spi0.0: init: write(0xE0) 0x0F 0x1F 0x1C 0x0C 0x0F 0x08 0$
[    8.174224] flexfb spi0.0: init: write(0xE1) 0x0F 0x32 0x2E 0x0B 0x0D 0x05 0$
[    8.190239] flexfb spi0.0: init: write(0xE2) 0x0F 0x32 0x2E 0x0B 0x0D 0x05 0$
[    8.205796] flexfb spi0.0: init: write(0x36) 0x28
[    8.212997] flexfb spi0.0: init: write(0x11)
[    8.219651] flexfb spi0.0: init: write(0x29)
[    8.393497] flexfb spi0.0: Display update: 1793 kB/s (167.182 ms), fps=0 (0.$
[    8.403604] flexfb spi0.0: fbtft_register_backlight()
[    8.411193] flexfb spi0.0: fbtft_register_backlight(): led pin not set, exit$
[    8.519980] graphics fb1: flexfb frame buffer, 480x320, 300 KiB video memory$
[    8.535180] fbtft_device:  GPIOS used by 'flexfb':
[    8.541901] fbtft_device:    'reset' = GPIO25
[    8.548154] fbtft_device:    'dc' = GPIO24
[    8.554114] fbtft_device:  SPI devices registered:
[    8.560762] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[    8.568997] fbtft_device:      flexfb spi0.0 16000kHz 8 bits mode=0x00
[   12.394163] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   16.162228] Adding 102396k swap on /var/swap.  Priority:-1 extents:2 across:$
```

midnightradio, 2015-03-13:
I was spending a lot of time with a trouble on waveshare 3.5inch LCD by just following a wrong procedure (http://www.circuitbasics.com/setup-lcd-touchscreen-raspberry-pi/) assuming that it may be the same as "waveshare32b". This was my last shot before returning the troublesome LCD and finally the issue has been resolved. Thanks a lot!

twkrol, 2015-03-13:
Great!
It's worth noting, that waveshare32b DOESN'T work for SpotPear 3,5 or 4".

midnightradio, 2015-03-13:
It was little bit early to yell out a cheer as touch screen is not working.

Though I put following line in /dev/modules, my raspi seems not detecting the touch device.
```
ads7846_device model=7846 cs=1 gpio_pendown=17  keep_vref_on=1 swap_xy=0 pressure_max=255 x_plate_ohms=60 x_min=200  x_max=3900 y_min=200 y_max=3900
```
Also I tried with "model=7843" following what @jharig23 mentioned, but the same result.
```
pi@raspberrypi ~ $ sudo modprobe ads7846_device model=7846
FATAL: Module ads7846_device not found.
pi@raspberrypi ~ $ sudo modprobe ads7846_device model=7843
FATAL: Module ads7846_device not found.
pi@raspberrypi ~ $ lsmod
Module                  Size  Used by
ads7846                10632  0 
fbtft_device           34613  0 
flexfb                 14438  2 
fbtft                  34434  2 flexfb,fbtft_device
syscopyarea             3153  1 fbtft
sysfillrect             3584  1 fbtft
sysimgblt               2278  1 fbtft
fb_sys_fops             1565  1 fbtft
snd_bcm2835            21157  0 
snd_pcm                90778  1 snd_bcm2835
snd_seq                61097  0 
snd_seq_device          7209  1 snd_seq
snd_timer              23007  2 snd_pcm,snd_seq
snd                    66285  5 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device
spi_bcm2708             6018  0 
evdev                  11000  2 
uio_pdrv_genirq         3666  0 
uio                     9897  1 uio_pdrv_genirq
pi@raspberrypi ~ $ 
```

jharig23, 2015-03-13:
I must agree that this little screen is not so fun.  I spoke to soon with my earlier comment, @midnightradio.  Did you perhaps try 

```
sudo modprobe ads7846
sudo modprobe ads7846_device model=7846
```

Maybe you need to load the ads7846 module first?

I can see touch events with `evtest`, but I'm not able to get x11 to recognize it.

midnightradio, 2015-03-13:
Thanks @jharig23 , but...
 ```
pi@raspberrypi ~ $ sudo modprobe ads7846
pi@raspberrypi ~ $ sudo modprobe ads7846_device model=7846
FATAL: Module ads7846_device not found.
pi@raspberrypi ~ $ 
```

Even there's nothing about ads7846 in dmesg.
```
pi@raspberrypi ~ $ dmesg | grep ads7846
```

[92 more replies]

closed 2021-08-24 by github-actions[bot]