notro/fbtft #219: Raspberry Pi B and MI0283QT-11: display permanently white

sharpu, 2015-01-07:
Hi,
I have a Kali Linux 3.15.8 on a Raspberry Pi B, attached is a pre-assembled MI0283QT-11 from Watterott, on the circuit board the print says 8 Bit SPI.  I tried to use the 'mi0283qt-2' 8 Bit driver without any success, right after the start the display always turns completely white (using ssh with PuTTY on Windows 8), regardless what is specified in the /etc/modprobe.d/fb-watterott.conf file. 
I hope its not to blunt to list it all, some more details:

root@kali:~# lsmod
Module                  Size  Used by
ipv6                  325310  12
fb_hx8347d              4000  1
fbtft_device           29785  0
fb_watterott            5234  0
fbtft                  31177  3 fb_watterott,fb_hx8347d,fbtft_device
syscopyarea             2995  1 fbtft
sysfillrect             3302  1 fbtft
sysimgblt               2120  1 fbtft
fb_sys_fops             1431  1 fbtft
snd_soc_pcm512x_i2c     1909  0
snd_soc_wm8804          7840  0
snd_soc_pcm512x         8174  1 snd_soc_pcm512x_i2c
regmap_spi              1913  1 snd_soc_wm8804
regmap_i2c              1657  2 snd_soc_wm8804,snd_soc_pcm512x_i2c
snd_soc_bcm2708_i2s     5653  0
regmap_mmio             2982  1 snd_soc_bcm2708_i2s
snd_soc_core          131920  3 snd_soc_pcm512x,snd_soc_wm8804,snd_soc_bcm2708_i2s
snd_compress            8133  1 snd_soc_core
snd_pcm_dmaengine       5415  1 snd_soc_core
snd_pcm                85149  2 snd_soc_core,snd_pcm_dmaengine
snd_seq                52886  0
snd_seq_device          6533  1 snd_seq
snd_timer              20789  2 snd_pcm,snd_seq
leds_gpio               2123  0
led_class               4114  1 leds_gpio
snd                    61229  6 

snd_soc_core,snd_timer,snd_pcm,snd_seq,snd_seq_device,snd_compress
spi_bcm2708             8155  0
rpi_pwm                 6547  0
i2c_bcm2708             5484  0
root@kali:~#

from dmesg:
[    9.910907] fbtft_device: init
[    9.910958] fbtft_device:  SPI devices registered:
[    9.917595] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[    9.925821] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[    9.933983] fbtft_device:  'fb' Platform devices registered:
[    9.941416] fbtft_device:      bcm2708_fb id=-1 pdata? no
[    9.948526] fbtft_device:  I2C devices registered:
[    9.955093] fbtft_device:       i2c-0  addr 48896  IRQ 0
[    9.962097] fbtft_device:       i2c-1  addr 48896  IRQ 1
[    9.969016] fbtft_device:      wm8804 1-003b  addr 59  IRQ 0
[    9.976292] fbtft_device:      pcm5122 1-004c  addr 76  IRQ 0
[    9.983676] fbtft_device:  name='mi0283qt-2', busnum=0, cs=0
[    9.983720] fbtft_device: Deleting spi0.0 (spi0.0)
[    9.993196] fbtft_device:  GPIOS used by 'mi0283qt-2':
[   10.009315] fbtft_device:    'dc' = GPIO24
[   10.015040] fbtft_device:    'reset' = GPIO25
[   10.032654] fbtft_device:    'led' = GPIO18
[   10.038440] fbtft_device:  SPI devices registered:
[   10.044851] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   10.052800] fbtft_device:      fb_hx8347d spi0.0 32000kHz 8 bits mode=0x00
[   10.589668] graphics fb1: fb_hx8347d frame buffer, 240x320, 150 KiB video memory, 4 KiB 
DMA buffer memory, fps=20, spi0.0 at 32 MHz

root@kali:~# ls -la /dev/spi*
crw------- 1 root root 153, 1 Jan  1  1970 /dev/spidev0.1
root@kali:~#

I saw in the fbtft_device.c file that the spidev is having default for busnum=0 and chipset=0. I tried to recreate the missing spidev0.0, which was subsequently deleted after the start. Creating a spidev.conf file also did not help. 
Any hint for a solution would be appreciated.

Thanks and regards
sharpu

notro, 2015-01-08:
The mi0283qt-2 display had a different controller. Try name=rpi-display

About spidev: fbtft_device deletes spidev0.0 to make room for the display driver. Only one driver (device) per chip select.

sharpu, 2015-01-08:
I changed it. yes, it fits to the spec I found, the entries from the log look OK, but the behaviour does not change:
[   10.117207]
[   10.117207] fbtft_device: init
[   10.117261] fbtft_device:  SPI devices registered:
[   10.124190] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[   10.132470] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   10.140598] fbtft_device:  'fb' Platform devices registered:
[   10.147967] fbtft_device:      bcm2708_fb id=-1 pdata? no
[   10.155158] fbtft_device:  I2C devices registered:
[   10.161661] fbtft_device:       i2c-0  addr 48896  IRQ 0
[   10.168611] fbtft_device:       i2c-1  addr 48896  IRQ 1
[   10.175541] fbtft_device:      wm8804 1-003b  addr 59  IRQ 0
[   10.182827] fbtft_device:      pcm5122 1-004c  addr 76  IRQ 0
[   10.190197] fbtft_device:  name='rpi-display', busnum=0, cs=1
[   10.190245] fbtft_device: Deleting spi0.1 (spi0.1)
[   10.555644] graphics fb1: fb_ili9341 frame buffer, 320x240, 150 KiB video memory, 16 KiB 

DMA buffer memory, fps=20, spi0.1 at 32 MHz
[   10.571078] fbtft_device:  GPIOS used by 'rpi-display':
[   10.578343] fbtft_device:    'dc' = GPIO24
[   10.584503] fbtft_device:    'reset' = GPIO23
[   10.590917] fbtft_device:    'led' = GPIO18
[   10.597093] fbtft_device:  SPI devices registered:
[   10.604021] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[   10.612474] fbtft_device:      fb_ili9341 spi0.1 32000kHz 8 bits mode=0x00
[   17.710988] NET: Registered protocol family 10

notro, 2015-01-08:
Can you give me the link to the product page, I need to see which display this is.

sharpu, 2015-01-09:
Sure, bought it via amazon.de: 
http://www.amazon.de/gp/product/B00I7BGX5A?psc=1&redirect=true&ref_=oh_aui_detailpage_o07_s00
it mentions the url, which is also on the delivered package: https://github.com/watterott/RPi-Display

On the backside of the device it says MI0283QT-11, below it 2011.10, on front side Rev 1.0, but Multi-Inno 0283QT-8 (!)

notro, 2015-01-11:
Yes that's rpi-display. If it doesn't work with Kali, you can test the display with the SD image on the Watterott github page. It is set up specifically for this display. If that image is working, there has to be something with your distro/kernel.

sharpu, 2015-01-13:
I thought so myself and tried it. Similar to other users' reports about it, it works rightaway (except for the touchscreen functionality as my display has the TI TSC2046I on it), but I don't quite understand how it does. The lsmod is not really giving a hint to it:
 pi@raspberrypi ~ $ lsmod
Module                  Size  Used by
evdev                  10485  1
joydev                  9096  0
ads7846                 8513  0
ads7846_device          6125  0
snd_bcm2835            18169  0
snd_soc_bcm2708_i2s     5486  0
regmap_mmio             2818  1 snd_soc_bcm2708_i2s
snd_soc_core          146286  1 snd_soc_bcm2708_i2s
snd_compress           12419  1 snd_soc_core
regmap_i2c              1657  1 snd_soc_core
snd_pcm_dmaengine       5481  1 snd_soc_core
regmap_spi              1913  1 snd_soc_core
snd_pcm                81542  3 snd_bcm2835,snd_soc_core,snd_pcm_dmaengine
snd_page_alloc          5168  1 snd_pcm
snd_seq                54581  0
snd_seq_device          6485  1 snd_seq
snd_timer              20353  2 snd_pcm,snd_seq
leds_gpio               2055  0
led_class               4378  1 leds_gpio
snd                    61526  7 snd_bcm2835,snd_soc_core,snd_timer,snd_pcm,snd_seq,snd_seq_device,snd_compress
pi@raspberrypi ~ $

Only the dmesg is showing it:
pi@raspberrypi ~ $ dmesg | grep rpi-display
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0xe bcm2708.serial=0xe7743b79 smsc95xx.macaddr=B8:27:EB:74:3B:79 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbtft_device.name=rpi-display fbtft_device.rotate=270 fbtft_device.debug=0 fbtft_device.verbose=0 fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo

My understanding is the kernel command line parameters are determined by the /boot/cmdline.txt content:
pi@raspberrypi /lib/modprobe.d $ more /boot/cmdline*
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline roo
twait fbtft_device.name=rpi-display fbtft_device.rotate=270 fbtft_device.debug=0 fbtft_device.verbose=0 fbcon=map:
10 fbcon=font:ProFont6x11 logo.nologo
 but not all of the entries in the dmesg-messages are in this file.
Trying to use such a /boot/cmdline.txt content for the Kali linux is somehow ignored.
Generally speaking, is the assumption right that there should not be a difference in the way a driver would behave when it is envoked by kernel command line parameters compared to the modprobe command?

notro, 2015-01-13:
The kernel on that image has the FBTFT drivers built into the kernel proper (kernel.img).
The same kernel that this command would give:
```
sudo REPO_URI=https://github.com/notro/rpi-firmware BRANCH=builtin rpi-update
```
So lsmod won't list them as loadable modules, because they are builtin, always loaded.

BRANCH=builtin and the default (BRANCH=master) are identical except for the FBTFT drivers and spi_bcm2708 being builtin.

> Generally speaking, is the assumption right that there should not be a difference in the way a driver would behave when it is envoked by kernel command line parameters compared to the modprobe command?

A driver module isn't invoked by the kernel commandline. It's only module arguments there. If the driver is builtin it is always present, but if it's a loadable module, it's loaded automatically during boot when the root filesystem has been mounted and userspace is started if a matching device is present.
If that device is created later (by fbtft_device), Linux automatically loads the necessary driver (at least in this case).
modprobe will override arguments from the kernel commandline.

The start of this page explains a little about the 'device' I'm talking about: https://github.com/notro/fbtft/wiki/fbtft_device

closed 2015-03-29 by notro