notro/fbtft #220: Question about drive ic support ili9488

pawnies20, 2015-01-08:
Hello!
Can you tell me if ili9488 (320*480) ic driver will be support with notro.

Another question: some manufacturer have touchscreen without touch  spi ic (only four lines).

Is it possible to use them?

Last question: do you know NV30354 ic and GC 9312 ic?

Many thanx

pawnies20, 2015-01-14:
Hi notro.
Do you have an answer for me please?

I must buying screen (ili9488 or ili9486) but don't know if it will be compatible.
Thank you

notro, 2015-01-14:
Sorry I missed this.

> Can you tell me if ili9488 (320*480) ic driver will be support with notro.

From the datasheet (DBI mode) it looks like the flexfb driver will work, but I can't tell for sure: https://github.com/notro/fbtft/wiki/flexfb
At least the default window adressing of flexfb matches this controller.
But initially I though that the ili9486 could be used through the native SPI interface, but it turned out it was a command only interface and not for pixels.

> Another question: some manufacturer have touchscreen without touch spi ic (only four lines).
> Is it possible to use them?

Not directly, a touch controller is needed. Those lines are just connected to a resistive film.

> Last question: do you know ili9486, NV30354 ic and GC9312 ic and last ili9327 ic, hx8238D, R61581B?

I have only tried a ili9486 used in piscreen: https://www.tindie.com/products/ozzmaker/piscreen-35-tft-with-touch-for-the-raspberry-pi/
piscreen uses a SPI to parallel interface circuit, since native SPI is not supported by the controller.

PROJETFEG, 2015-01-14:
Many thanx notro!!!

Is existing a list of 
-ic for touch compatible
-ic for lcd compatible

Namy thanx again Notro and great job !

notro, 2015-01-14:
Linux touch drivers: http://lxr.free-electrons.com/source/drivers/input/touchscreen/Kconfig
The ADS7846 (TCS2046) touch controller is used by the majority of these small touch displays.

All files starting with **fb_** are LCD controller drivers: https://github.com/notro/fbtft
Here's the list of supported displays: https://github.com/notro/fbtft/blob/master/fbtft_device.c#L168
Any display with a supported controller can be used if you've got the initialization sequence.

closed 2015-03-29 by notro