notro/fbtft #222: Tontec 3.5 changed again (new hardware)

pieterk, 2015-01-08:
It seems that the Tontec 3.5 Pi screen has changed model again and only produces a white screen with the raspberry pi for example. At a very steep angle I can see a ever (very slight) shadow of the Pi logo but it is pretty much washed out and unreadable.

I am trying the ili9486 driver/config but this does not work.

The new model code is MZ61581 (dated 2014.12.28)

```(DMESG)

[   11.573707] fbtft_device:  SPI devices registered:
[   11.594326] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[   11.602391] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   11.612907] fbtft_device:  'fb' Platform devices registered:
[   11.621029] fbtft_device:      bcm2708_fb id=-1 pdata? no
[   11.632912] fbtft_device: Deleting spi0.0
[   11.643409] fbtft_device:  GPIOS used by 'tontec35_9486':
[   11.660088] fbtft_device:    'reset' = GPIO15
[   11.671975] fbtft_device:    'dc' = GPIO25
[   11.684312] fbtft_device:    'led_' = GPIO18
[   11.690047] fbtft_device:  SPI devices registered:
[   11.709900] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   11.720178] fbtft_device:      fb_ili9486 spi0.0 128000kHz 8 bits mode=0x03
[   12.898348] graphics fb1: fb_ili9486 frame buffer, 480x320, 300 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 128 MHz
```

```(LSMOD)

Module                  Size  Used by
fb_ili9486              3115  2 
fbtft_device           34686  0 
fbtft                  30834  2 fb_ili9486,fbtft_device
syscopyarea             2999  1 fbtft
sysfillrect             3294  1 fbtft
sysimgblt               2124  1 fbtft
fb_sys_fops             1435  1 fbtft
snd_bcm2835            19496  0 
snd_soc_tas5713         5573  0 
snd_soc_bcm2708_i2s     6186  0 
regmap_mmio             2818  1 snd_soc_bcm2708_i2s
snd_soc_core          146249  2 snd_soc_tas5713,snd_soc_bcm2708_i2s
snd_compress           12594  1 snd_soc_core
regmap_i2c              1661  2 snd_soc_core,snd_soc_tas5713
snd_pcm_dmaengine       5505  1 snd_soc_core
regmap_spi              1913  1 snd_soc_core
snd_pcm                83853  3 snd_bcm2835,snd_soc_core,snd_pcm_dmaengine
snd_page_alloc          5140  1 snd_pcm
snd_seq                55476  0 
snd_seq_device          6465  1 snd_seq
snd_timer              20981  2 snd_pcm,snd_seq
leds_gpio               2079  0 
led_class               4373  1 leds_gpio
snd                    62252  7 snd_bcm2835,snd_soc_core,snd_timer,snd_pcm,snd_seq,snd_seq_device,snd_compress
spi_bcm2708             8111  0 
i2c_bcm2708             4943  0 
```

JustAFatYeti, 2015-01-15:
Hey, I just got this same screen a few days ago for my Raspberry Pi B+, plug it in and nothing but a white screen. I've written rasbian onto the micro sd and can see it fine on a hdmi monitor, and from there have downloaded the driver from the following source.

https://s3.amazonaws.com/ttbox/35screen-img-9486.zip

When trying to extract the zip folder on the pi it crashed, however I formatted the sd and wrote the image at the above link onto the pi and still I only get a white screen.

It's probably evident that I'm fairly new to all this, and so I'm unable to gauge the scale of the issue, but I was wondering if you had any suggestions as to what I should try.

Many thanks.

ttmz, 2015-01-21:
/*
 * FB driver for Tontec Screen with R615819 LCD Controller 
 * for board on and after 2014.12.28
*/

#include <linux/module.h>
#include <linux/kernel.h>
#include <linux/init.h>

#include "fbtft.h"

#define DRVNAME		"fb_r61581"
#define WIDTH		320
#define HEIGHT		480

/* this init sequence matches Tontec Screen */
static int default_init_sequence[] = {
	-1,0xB0,0x00,
	-1,0x11,
	-2,250,
	-1,0xB3,0x02,0x00,0x00,0x00,
	-1,0xC0,0x13,0x3B,0x00,0x02,0x00,0x01,0x00,0x43,
	-1,0xC1,0x08,0x16,0x08,0x08,
	-1,0xC4,0x11,0x07,0x03,0x03,
	-1,0xC6,0x00,
	-1,0xC8,0x03,0x03,0x13,0x5C,0x03,0x07,0x14,0x08,0x00,0x21,0x08,0x14,0x07,0x53,0x0C,0x13,0x03,0x03,0x21,0x00,
	-1,0x35,0x00,
	-1,0x36,0xa0,
	-1,0x3A,0x55,
	-1,0x44,0x00,0x01,
	-1,0xD0,0x07,0x07,0x1D,0x03,
	-1,0xD1,0x03,0x30,0x10,
	-1,0xD2,0x03,0x14,0x04,
	-1,0x29,
	-1,0x2C,
	-3
};

static void set_addr_win(struct fbtft_par *par, int xs, int ys, int xe, int ye)
{
	fbtft_par_dbg(DEBUG_SET_ADDR_WIN, par,
		"%s(xs=%d, ys=%d, xe=%d, ye=%d)\n", __func__, xs, ys, xe, ye);

	/* Column address */
	write_reg(par, 0x2A, xs >> 8, xs & 0xFF, xe >> 8, xe & 0xFF);

	/* Row adress */
	write_reg(par, 0x2B, ys >> 8, ys & 0xFF, ye >> 8, ye & 0xFF);

	/* Memory write */
	write_reg(par, 0x2C);
}

static int set_var(struct fbtft_par *par)
{
	fbtft_par_dbg(DEBUG_INIT_DISPLAY, par, "%s()\n", __func__);

	switch (par->info->var.rotate) {
	case 0:
		write_reg(par, 0x36, 0x00 | (par->bgr << 3));
		break;
	case 90:
		write_reg(par, 0x36, 0x60 | (par->bgr << 3));
		break;
	case 180:
		write_reg(par, 0x36, 0xc0 | (par->bgr << 3));
		break;
	case 270:
		write_reg(par, 0x36, 0xA0 | (par->bgr << 3));
		break;
	default:
		break;
	}
	return 0;
}

static struct fbtft_display display = {
	.regwidth = 8,
	.width = WIDTH,
	.height = HEIGHT,
	.init_sequence = default_init_sequence,
	.fbtftops = {
		.set_addr_win = set_addr_win,
		.set_var = set_var,
	},
};
FBTFT_REGISTER_DRIVER(DRVNAME, "rense,r61581", &display);

MODULE_ALIAS("spi:" DRVNAME);
MODULE_ALIAS("platform:" DRVNAME);
MODULE_ALIAS("spi:r61581");
MODULE_ALIAS("platform:r61581");

MODULE_DESCRIPTION("FB driver for the R61581 LCD Controller");
MODULE_AUTHOR("Noralf Tronnes");
MODULE_LICENSE("GPL");

ttmz, 2015-01-21:
copy the above code to fb_r61581.c  to replace the ili9486.c

pieterk, 2015-01-21:
Thanks ttmz. That works perfectly. 
Had to fill in some blanks (github syntax quirks) but your paste help a lot.

closed 2015-01-21 by pieterk

skipmonkey, 2015-01-22:
Thanks for the code ttmz.  Been trying to work this out for couple of weeks with no success.  After reading through notros fbtft files figured it would be to do with 0x00 data, but no idea how to work out, not sure how you did either, but you did because it works.  Used notro fbtft development page to make changes to .c file on rpi, takes a while to do, but worth it.  Thanks to all sites have used to get this working - notro, ttmz, davestechmusings, cooljc.me.uk

ml1969, 2015-01-24:
Hi,

my name ist Michael and I am new here and also new with the raspberry.

I have a raspberry PI B+ and the tonTec 3,5 display.
I tried the manual and the driver (also the image) wich was listed on amazon -> screen35
Display was not working.
Then I tried to install th fbtft driver from notro, also no luck.

Then i contacted tontec, thy send me an new raspbian wheezy image called "mz61581".
Now I got the LCD working, but the touchscreen did not. Also some other issues with wifi, and 
if I do a apt-get update the driver won´t work anymore.

So.. now i knew that I also have the 61581-chipset ... fine
But I am not so expirienced in linux or building a driver or kernel.

Can someone help whre I should make the changes from above?
Do I have to build a custom kernel or driver or can I use the dowbload with some changes?

greetings
Michael (from germany :D)

pieterk, 2015-01-24:
For the touch screen part use step 6 here thanks to http://cooljc.me.uk/?p=64

This worked for the new Tontec 3.5 screen and presses seem to be accurate.

ml1969, 2015-01-24:
Hi pieterk
thanks for that .... 
for future it will be fine if I could reinstall the display with the original raspbian instead of the image from tontec, so I can do updates etc.

Michael

cooljc, 2015-01-31:
Hi Noralf,

I have created a patch based on comment 3 from ttmz. Any chance you can add it to the official master branch?

Cheers

Jon

```diff
diff --git a/Kconfig b/Kconfig
index 995a910..d650b3a 100644
--- a/Kconfig
+++ b/Kconfig
@@ -80,6 +80,12 @@ config FB_TFT_PCD8544
 	help
 	  Generic Framebuffer support for PCD8544
 
+config FB_TFT_R61581
+        tristate "FB driver for the R61581 LCD Controller"
+        depends on FB_TFT
+	help
+	  Generic Framebuffer support for R61581
+
 config FB_TFT_RA8875
         tristate "FB driver for the RA8875 LCD Controller"
         depends on FB_TFT
diff --git a/Makefile b/Makefile
index 71c755d..3636186 100644
--- a/Makefile
+++ b/Makefile
@@ -21,6 +21,7 @@ obj-$(CONFIG_FB_TFT_ILI9341)     += fb_ili9341.o
 obj-$(CONFIG_FB_TFT_ILI9481)     += fb_ili9481.o
 obj-$(CONFIG_FB_TFT_ILI9486)     += fb_ili9486.o
 obj-$(CONFIG_FB_TFT_PCD8544)     += fb_pcd8544.o
+obj-$(CONFIG_FB_TFT_R61581)      += fb_r61581.o
 obj-$(CONFIG_FB_TFT_RA8875)      += fb_ra8875.o
 obj-$(CONFIG_FB_TFT_S6D02A1)     += fb_s6d02a1.o
 obj-$(CONFIG_FB_TFT_S6D1121)     += fb_s6d1121.o
diff --git a/fb_r61581.c b/fb_r61581.c
new file mode 100644
index 0000000..ec48e79
--- /dev/null
+++ b/fb_r61581.c
@@ -0,0 +1,116 @@
+/*
+ * FB driver for the R61581 LCD Controller
+ *
+ * Copyright (C) 2014 Noralf Tronnes
+ *
+ * This program is free software; you can redistribute it and/or modify
+ * it under the terms of the GNU General Public License as published by
+ * the Free Software Foundation; either version 2 of the License, or
+ * (at your option) any later version.
+ *
+ * This program is distributed in the hope that it will be useful,
+ * but WITHOUT ANY WARRANTY; without even the implied warranty of
+ * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
+ * GNU General Public License for more details.
+ *
+ * You should have received a copy of the GNU General Public License
+ * along with this program; if not, write to the Free Software
+ * Foundation, Inc., 675 Mass Ave, Cambridge, MA 02139, USA.
+ */
+
+#include <linux/module.h>
+#include <linux/kernel.h>
+#include <linux/init.h>
+
+#include "fbtft.h"
+
+#define DRVNAME		"fb_r61581"
+#define WIDTH		320
+#define HEIGHT		480
+
+
+/* this init sequence matches R61581 */
+static int default_init_sequence[] = {
+	-1,0xB0, 0x00,
+	-1,0x11,
+	-2,250,
+	-1,0xB3, 0x02, 0x00, 0x00, 0x00,
+	-1,0xC0, 0x13, 0x3B, 0x00, 0x02, 0x00, 0x01, 0x00, 0x43,
+	-1,0xC1, 0x08, 0x16, 0x08, 0x08,
+	-1,0xC4, 0x11, 0x07, 0x03, 0x03,
+	-1,0xC6, 0x00,
+	-1,0xC8, 0x03, 0x03, 0x13, 0x5C, 0x03, 0x07, 0x14, 0x08,
+			 0x00, 0x21, 0x08, 0x14, 0x07, 0x53, 0x0C, 0x13,
+			 0x03, 0x03, 0x21, 0x00,
+	-1,0x35, 0x00,
+	-1,0x36, 0xa0,
+	-1,0x3A, 0x55,
+	-1,0x44, 0x00, 0x01,
+	-1,0xD0, 0x07, 0x07, 0x1D, 0x03,
+	-1,0xD1, 0x03, 0x30, 0x10,
+	-1,0xD2, 0x03, 0x14, 0x04,
+	-1,0x29,
+	-1,0x2C,
+	-3
+};
+
+static void set_addr_win(struct fbtft_par *par, int xs, int ys, int xe, int ye)
+{
+	fbtft_par_dbg(DEBUG_SET_ADDR_WIN, par,
+		"%s(xs=%d, ys=%d, xe=%d, ye=%d)\n", __func__, xs, ys, xe, ye);
+
+	/* Column address */
+	write_reg(par, 0x2A, xs >> 8, xs & 0xFF, xe >> 8, xe & 0xFF);
+
+	/* Row adress */
+	write_reg(par, 0x2B, ys >> 8, ys & 0xFF, ye >> 8, ye & 0xFF);
+
+	/* Memory write */
+	write_reg(par, 0x2C);
+}
+
+static int set_var(struct fbtft_par *par)
+{
+	fbtft_par_dbg(DEBUG_INIT_DISPLAY, par, "%s()\n", __func__);
+
+	switch (par->info->var.rotate) {
+	case 0:
+		write_reg(par, 0x36, 0x00 | (par->bgr << 3));
+		break;
+	case 90:
+		write_reg(par, 0x36, 0x60 | (par->bgr << 3));
+		break;
+	case 180:
+		write_reg(par, 0x36, 0xc0 | (par->bgr << 3));
+		break;
+	case 270:
+		write_reg(par, 0x36, 0xA0 | (par->bgr << 3));
+		break;
+	default:
+		break;
+	}
+
+	return 0;
+}
+
+
+static struct fbtft_display display = {
+	.regwidth = 8,
+	.width = WIDTH,
+	.height = HEIGHT,
+	.init_sequence = default_init_sequence,
+	.fbtftops = {
+		.set_addr_win = set_addr_win,
+		.set_var = set_var,
+	},
+};
+FBTFT_REGISTER_DRIVER(DRVNAME, "renesas,r61581", &display);
+
+MODULE_ALIAS("spi:" DRVNAME);
+MODULE_ALIAS("platform:" DRVNAME);
+MODULE_ALIAS("spi:r61581");
+MODULE_ALIAS("platform:r61581");
+
+MODULE_DESCRIPTION("FB driver for the R61581 LCD Controller");
+MODULE_AUTHOR("Noralf Tronnes");
+MODULE_LICENSE("GPL");
diff --git a/fbtft_device.c b/fbtft_device.c
index b9f4c30..c6af666 100644
--- a/fbtft_device.c
+++ b/fbtft_device.c
@@ -993,6 +993,26 @@ static struct fbtft_device_display displays[] = {
 			}
 		}
 	}, {
+		.name = "tontec35_61581", /* boards after 28 December 2014 */
+		.spi = &(struct spi_board_info) {
+			.modalias = "fb_r61581",
+			.max_speed_hz = 128000000,
+			.mode = SPI_MODE_3,
+			.platform_data = &(struct fbtft_platform_data) {
+				.display = {
+					.buswidth = 8,
+					.backlight = 1,
+				},
+				.bgr = true,
+				.gpios = (const struct fbtft_gpio []) {
+					{ "reset", 15 },
+					{ "dc", 25 },
+					{ "led_", 18 },
+					{},
+				},
+			}
+		}
+	}, {
 		.name = "upd161704",
 		.spi = &(struct spi_board_info) {
 			.modalias = "fb_upd161704",

```

notro, 2015-02-01:
Development has moved so I don't take any more patches to this repo.
Patches have to be sent like regular Linux kernel patches.
MAINTAINERS entry: https://git.kernel.org/cgit/linux/kernel/git/gregkh/staging.git/commit/MAINTAINERS?h=staging-testing&id=053e514f9172c10b55b1ef0706da57cecb6c85e3
Ref: https://github.com/notro/fbtft/wiki/Development

FBTFT will most likely appear in Linux 3.20.

Stevo1488, 2024-03-12:
can anyone help me get my tontec  MZ61581 touch display to work on raspberry pi 2b. I am new to code writing and it has been very frustrating trying to get it to work.... any help would definitely be appreciated and save my hair from being pulled out.