notro/fbtft #224: Sainsmart Waveshare 3.5 with fbcp

kurosov, 2015-01-11:
Following the information in another issue thread i managed to get this display working prefectly in Raspbian and even somewhat working in Raspbmc with decent performance.

However, in Raspbmc there seems to be a contrast issue in that anything on the display that is over a certain brightness threshold displays incorrectly and videos played on the display flicker every couple of seconds from looking perfectly normal to having this same issue.

If i open a transparent menu above the affected area (effectively bringing down the brightness of the area is displays fine so i'm fairly confident its an issue with contrast.

I've tried messing with gamma settings in /etc/modules but this seems to make no difference at all and doesn't even change the view in the Raspbian install. Do the displays that don't have dedicated support not work with gamma? if not is there anyway of forcing a reduction in contrast or lowering the brightness threshhold via FBTFT?

Reducing the speed value in /etc/modules makes the affected area greater and increasing it makes it less up till 24000000 when graininess and some artifacting appears. 22000000 oddly seems the sweet spot.

Here is a picture on imgur: http://imgur.com/tUDk7ls
 As you can see the music/system menu options appear fine while the highlighted programs option isn't. The bright parts of the background image in the centre are also affected.

Also, if anybody wants to know how i got the display somewhat working in Raspbmc feel free to ask.

notro, 2015-01-13:
> I've tried messing with gamma settings in /etc/modules but this seems to make no difference at all and doesn't even change the view in the Raspbian install. Do the displays that don't have dedicated support not work with gamma?

I don't know which driver this display uses, but only a few drivers support gamma. If it does, you should have the file /sys/class/graphics/fb1/gamma

A workaround is to supply a custom init= sequence (register settings) using a different gamma setting.
Look in the driver to get the current init sequence and then look here for the format used by fbtft_device: https://github.com/notro/fbtft/wiki/fbtft_device#parameters

IncredibleFruit, 2015-01-30:
I would love to know how you got this display working, in both Raspbian and Raspbmc. I have limited experience with linux and the pi, but love if you could give me some direction on how to get this display up and running.

closed 2015-05-23 by notro