notro/fbtft #225: need to get in touch with you to get gerber specs for add on board looking at 4.00 price range

canadametalsauction, 2015-01-12:
need specs to get gerber going for add on board, to get large run going, any skype userid? i speak french as well I'm in montreal.

looking at the board texy has made, the cost of 12 pounds is outrageous, all rasberry pi addons are way over priced and the new rasberry pi as well
http://harizanov.com/2013/02/1-8-tft-lcd-display-on-raspberry-pi/
http://harizanov.com/product/1-8-tft-display-for-raspberry-pi/

http://www.raspberrypi.org/forums/viewtopic.php?f=45&t=64993

Also anyone who can help with design of small terminal(case as well), would be appreciated, looks like latest version of banana pie can be bought at 25.00 in quantity.
http://www.aliexpress.com/store/contactinfo-32228220781-200526734/302756.html

Texy, 2015-01-12:
You think you'll get help after the pm you sent me? You have a lot to learn I life. 
No idea where you got £12 for either. 
Texy

notro, 2015-01-12:
You have displayed an attitude towards my friend that I do not tolerate. Way over line. You will not receive any help here.

closed 2015-01-12 by notro