notro/fbtft #227: fbtft_device:  spi_busnum_to_master(0) returned NULL

rdissertori, 2015-01-13:
Hello I've been searching the issues for this error but I haven't found any issues fitting my issue.
I'm on rpi archlinux and self compiled the kernel.

when I load the module via 
/etc/modules-load.d/raspberrypi.conf
>fbtft_device
I get the error message
fbtft_device:  spi_busnum_to_master(0) returned NULL

I've also added
/etc/modprobe.d/fbtft.conf
>options fbtft_device name=mi0283qt-v2 cs=0 rotate=90 gpios=reset:23

now the odd thing is that when I manually enter
modprobe fbtft_device name=mi0283qt-v2 cs=0 rotate=90 gpios=reset:23
the driver successfully loads and the devices get created

are there any hints how to do this on boot?

notro, 2015-01-13:
In your log fbtft_device is loaded before bcm2708_spi. So when fbtft_device is loaded it can't find the SPI bus, because it hasn't been created yet.
I don't know how to reorder that on Arch.

rdissertori, 2015-01-14:
Thanks a lot for the hint - actually it was so abvious that I didn't recognize it.
On Arch you can just put softdep into your /etc/modprobe.d/xxx.conf file
BUT it seems theres no dependencies defined in the fbtft modules to the spi modules so this won't work.

I got it finally working putting
>install fbtft_device /sbin/modprobe spi_bcm2708; /sbin/modprobe --ignore-install fbtft_device
options fbtft_device name=mi0283qt-v2 cs=0 rotate=90 gpios=reset:23

into /etc/modprobe.d/fbtft.conf

So yeah this has been resolved :+1:

closed 2015-01-14 by rdissertori