notro/fbtft #228: fbtft on bananapi

FishTest, 2015-01-16:
My friend send me a Banana PI (http://www.bananapi.com) and Banana pi 's GPIO is compatible with raspberry pi 
I want to use Adafruit 2.8 TFT Screen for raspberry pi on it
The screen works OK,but the touch shield can not be enable.
System:Bananian(http://www.bananian.org,kernel source code:http://github.com/bananian/linux-bananapi)
/etc/modules
fbtft_device name=pitft28 rotate=90
I got stmpe_device.c from fbtft_tools,then remove two #ifdef ARCH_BCM2708.
i compiled it and sucefully modprobe it use below paras:
stmpe_device cs=1 chip=stmpe610 blocks=gpio,ts irq-pullup irq-gpio=24 irq-base=330 sample-time=4 mod-12b=1 ref-sel=0 adc-freq=2 ave-ctrl=3 touch-det-delay=4 settling=2 fraction-z=7 i-drive=0

but in /dev/input,i can not see any ts device.
cat /proc/interrupts
and there's no anything about stmpe

could you give me some help about this?
Many thanks

FishTest, 2015-01-16:
change:
#ifdef CONFIG_ARCH_BCM2708
module_param(irq_pullup, bool, 0);
MODULE_PARM_DESC(irq_pullup, "Enable internal pull up resistor for irq");
#endif
to:
module_param(irq_pullup, bool, 0);
MODULE_PARM_DESC(irq_pullup, "Enable internal pull up resistor for irq");

change;
#ifdef CONFIG_ARCH_BCM2708
static void gpio_pull(unsigned pin, unsigned pud)
{
#define	GPIO_PUD     37
#define	GPIO_PUDCLK0 38
#define INP_GPIO(g) *(gpio+((g)/10)) &= ~(7<<(((g)%10)*3))
	u32 *gpio = ioremap(0x20200000, SZ_16K);

	if (verbose > 1)
		pr_info(DRVNAME": %s(%d, %d)\n", __func__, pin, pud);

	/* set as input */
	INP_GPIO(pin);
	/* set pull tri/down/up */
	*(gpio + GPIO_PUD) = pud & 3 ;		
	udelay(5);
	*(gpio + GPIO_PUDCLK0 + (pin>31 ? 1 : 0) ) = 1 << (pin & 31);
	udelay(5);
	*(gpio + GPIO_PUD) = 0;
	udelay(5);
	*(gpio + GPIO_PUDCLK0 + (pin>31 ? 1 : 0) ) = 0;
	udelay(5);

	iounmap(gpio);
#undef INP_GPIO
}
#else
static void gpio_pull(unsigned pin, unsigned pud)
{
}
#endif
to:
static void gpio_pull(unsigned pin, unsigned pud)
{
#define	GPIO_PUD     37
#define	GPIO_PUDCLK0 38
#define INP_GPIO(g) *(gpio+((g)/10)) &= ~(7<<(((g)%10)*3))
	u32 *gpio = ioremap(0x20200000, SZ_16K);

	if (verbose > 1)
		pr_info(DRVNAME": %s(%d, %d)\n", __func__, pin, pud);

	/* set as input */
	INP_GPIO(pin);
	/* set pull tri/down/up */
	*(gpio + GPIO_PUD) = pud & 3 ;		
	udelay(5);
	*(gpio + GPIO_PUDCLK0 + (pin>31 ? 1 : 0) ) = 1 << (pin & 31);
	udelay(5);
	*(gpio + GPIO_PUD) = 0;
	udelay(5);
	*(gpio + GPIO_PUDCLK0 + (pin>31 ? 1 : 0) ) = 0;
	udelay(5);

	iounmap(gpio);
#undef INP_GPIO
}

notro, 2015-01-17:
The pullup code is not needed on the latest displays. There's a pullup resistor on the pcb.

Have you enabled the driver?
```
$ zgrep STMPE /proc/config.gz
# CONFIG_KEYBOARD_STMPE is not set
CONFIG_TOUCHSCREEN_STMPE=y
CONFIG_GPIO_STMPE=y
CONFIG_MFD_STMPE=y
# STMicroelectronics STMPE Interface Drivers
CONFIG_STMPE_I2C=y
CONFIG_STMPE_SPI=y
```

FishTest, 2015-01-21:
root@bananapi ~ # zgrep STMPE /proc/config.gz
# CONFIG_KEYBOARD_STMPE is not set
CONFIG_TOUCHSCREEN_STMPE=m
CONFIG_GPIO_STMPE=y
CONFIG_MFD_STMPE=y
# STMPE Interface Drivers
CONFIG_STMPE_I2C=y
CONFIG_STMPE_SPI=y
root@bananapi ~ #

notro, 2015-01-21:
You should get this kernel messages when the core stmpe driver is probed:
```
stmpe-spi spi0.1: stmpe610 detected, chip id: 0x811
```
Then you get this when the stmpe touch driver is probed:
```
 input: stmpe-ts as /devices/virtual/input/inputXX
```

Can you give me the ouput of this:
```
$ sudo dmesg -C
$ sudo modprobe --first-time stmpe_device verbose=2 <the rest of your args>
$ dmesg
```

FishTest, 2015-01-22:
dmesg
[  185.514849] stmpe_device: stmpe_device_exit()
[  187.895420] stmpe_device: stmpe_device_init()
[  187.903892] stmpe_device: SPI devices registered:
[  187.914815] stmpe_device:    fb_ssd1306 spi0.0 16000kHz 8 bits mode=0x00
[  187.923806] stmpe_device:
[  187.929154] stmpe_device: Settings:
[  187.936224] stmpe_device:   chip = stmpe610
[  187.943969] stmpe_device:   irq_base = 330
[  187.951657] stmpe_device:   irq_trigger = 2
[  187.959278] stmpe_device:   irq_gpio = 24
[  187.967629] stmpe_device:   gpio->gpio_base = -1
[  187.976726] stmpe_device:   gpio->norequest_mask = 0
[  187.985768] stmpe_device:   ts->sample_time = 4
[  187.994012] stmpe_device:   ts->mod_12b = 1
[  188.001778] stmpe_device:   ts->ref_sel = 0
[  188.009643] stmpe_device:   ts->adc_freq = 2
[  188.017912] stmpe_device:   ts->ave_ctrl = 3
[  188.026462] stmpe_device:   ts->touch_det_delay = 4
[  188.035133] stmpe_device:   ts->settling = 2
[  188.043203] stmpe_device:   ts->fraction_z = 7
[  188.051367] stmpe_device:   ts->i_drive = 0
[  188.059159] stmpe_device: gpio_pull(24, 2)
[  188.069963] stmpe-spi spi0.1: failed to request IRQ GPIO: -16
[  188.081160] stmpe-spi: probe of spi0.1 failed with error -16
[  188.091146] stmpe_device: SPI devices registered:
[  188.102221] stmpe_device:    fb_ssd1306 spi0.0 16000kHz 8 bits mode=0x00
[  188.114998] stmpe_device:    stmpe610 spi0.1 500kHz 8 bits mode=0x00
[  188.123677] stmpe_device:

FishTest, 2015-01-22:
root@bananapi ~ # dmesg                                                    :(
[   58.529796] stmpe_device: stmpe_device_init()
[   58.538151] stmpe_device: SPI devices registered:
[   58.548435] stmpe_device:    spidev spi0.1 12000kHz 8 bits mode=0x03
[   58.560698] stmpe_device:    fb_ili9340 spi0.0 32000kHz 8 bits mode=0x00
[   58.577233] stmpe_device:
[   58.582412] stmpe_device: Settings:
[   58.589259] stmpe_device:   chip = stmpe610
[   58.597633] stmpe_device:   irq_base = 330
[   58.606845] stmpe_device:   irq_trigger = 2
[   58.615975] stmpe_device:   irq_gpio = 24
[   58.625545] stmpe_device:   gpio->gpio_base = -1
[   58.636057] stmpe_device:   gpio->norequest_mask = 0
[   58.646463] stmpe_device:   ts->sample_time = 4
[   58.656105] stmpe_device:   ts->mod_12b = 1
[   58.665517] stmpe_device:   ts->ref_sel = 0
[   58.675013] stmpe_device:   ts->adc_freq = 2
[   58.684629] stmpe_device:   ts->ave_ctrl = 3
[   58.694809] stmpe_device:   ts->touch_det_delay = 4
[   58.704993] stmpe_device:   ts->settling = 2
[   58.714782] stmpe_device:   ts->fraction_z = 7
[   58.724476] stmpe_device:   ts->i_drive = 0
[   58.733829] stmpe_device: gpio_pull(24, 2)
[   58.742909] stmpe_device: Deleting spi0.1
[   58.760323] stmpe-spi spi0.1: stmpe610 detected, chip id: 0x811
[   58.776114] Trying to set irq flags for IRQ330
[   58.786168] Trying to set irq flags for IRQ331
[   58.796238] Trying to set irq flags for IRQ332
[   58.806328] Trying to set irq flags for IRQ333
[   58.816453] Trying to set irq flags for IRQ334
[   58.826584] Trying to set irq flags for IRQ335
[   58.836684] Trying to set irq flags for IRQ336
[   58.846801] Trying to set irq flags for IRQ337
[   58.861579] Trying to set irq flags for IRQ339
[   58.872053] Trying to set irq flags for IRQ340
[   58.890274] Trying to set irq flags for IRQ341
[   58.900387] Trying to set irq flags for IRQ342
[   58.910510] Trying to set irq flags for IRQ343
[   58.920643] Trying to set irq flags for IRQ344
[   58.931902] stmpe-gpio stmpe-gpio.0: unable to get irq: -22
[   58.943168] Trying to set irq flags for IRQ339
[   58.953247] Trying to set irq flags for IRQ340
[   58.963333] Trying to set irq flags for IRQ341
[   58.973400] Trying to set irq flags for IRQ342
[   58.983501] Trying to set irq flags for IRQ343
[   58.993570] Trying to set irq flags for IRQ344
[   59.009182] stmpe-gpio: probe of stmpe-gpio.0 failed with error -22
[   59.022617] stmpe-ts stmpe-ts.0: Failed to request IRQ 331
[   59.035601] stmpe-ts: probe of stmpe-ts.0 failed with error -22
[   59.047609] stmpe_device: SPI devices registered:
[   59.060242] stmpe_device:    fb_ili9340 spi0.0 32000kHz 8 bits mode=0x00
[   59.074501] stmpe_device:    stmpe610 spi0.1 500kHz 8 bits mode=0x00
[   59.083856] stmpe_device:

notro, 2015-01-22:
This is probably wrong: ```irq-base=330```
It worked on the Raspberry Pi, because that was the first available IRQ.
Try grepping for ```NR_IRQS``` in the kernel log

Also try with only one of the blocks at a time to minimize log output: ```blocks=ts```

FishTest, 2015-01-23:
thanks!
with these paras i successfully load the module
stmpe_device cs=1 chip=stmpe610 blocks=gpio,ts irq-pullup irq-gpio=24 irq-base=150 sample-time=4 mod-12b=1 ref-sel=0 adc-freq=2 ave-ctrl=3 touch-det-delay=4 settling=2 fraction-z=7 i-drive=0
and i can see touchscreen in /dev/input/

root@bananapi ~ # ls /dev/input/
by-id/    event0  event2  mice    mouse1
by-path/  event1  event3  mouse0  touchscreen@

but when i execute evtest
the touchscreen can't get any data,but on raspberry pi it works perfect

root@bananapi ~ # evtest                                                   :(
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      sunxi-ir
/dev/input/event1:      axp20-supplyer
/dev/input/event2:      Logitech Unifying Device. Wireless PID:2012
/dev/input/event3:      stmpe-ts
Select the device event number [0-3]:

notro, 2015-01-23:
Check ```/proc/interrupts``` to see if any interrupts are fired. That is probably the problem, the driver is not receiving any interrupts.

FishTest, 2015-01-25:
Many Many Thanks,finally i got it to work!
the key:

stmpe_device cs=1 chip=stmpe610 blocks=gpio,ts irq-pullup irq-gpio=23 irq-base=150 sample-time=4 mod-12b=1 ref-sel=0 adc-freq=2 ave-ctrl=3 touch-det-delay=4 settling=2 fraction-z=7 i-drive=0

closed 2015-03-29 by notro