notro/fbtft #230: fbtft remains white even if there are no error messages in the debugging (fps=0 (0.000 ms))

rdissertori, 2015-01-17:
Hello again there,

I was expecting the display would show up on my adafruit28 (2.8") tft since there seemed to be no issues in "dmesg"

However the screen still remains white and I have _no_ idea what causes this.
I've posted the dmesg output, but it seems all ok to me, just the message fps=0 is weird

>[    6.217630] bcm2708_spi bcm2708_spi.0: DMA channel 2 at address 0xf2007200 with irq 77
[    6.268030] bcm2708_spi bcm2708_spi.0: DMA channel 4 at address 0xf2007400 with irq 20
[    6.312266] bcm2708_spi bcm2708_spi.0: SPI Controller at 0x20204000 (irq 80)
[    6.331931] bcm2708_spi bcm2708_spi.0: SPI Controller running in dma mode
[    6.422842] fbtft_device:  SPI devices registered:
[    6.438643] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[    6.468414] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[    6.468422] fbtft_device:  'fb' Platform devices registered:
[    6.468447] fbtft_device:      bcm2708_fb id=-1 pdata? no
[    6.468541] fbtft_device: Deleting spi0.0
[    6.483871] fbtft_device:  GPIOS used by 'adafruit28':
[    6.483880] fbtft_device:    'reset' = GPIO25
[    6.483884] fbtft_device:    'dc' = GPIO24
[    6.483888] fbtft_device:    'led' = GPIO18
[    6.483891] fbtft_device:  SPI devices registered:
[    6.483903] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[    6.483911] fbtft_device:      fb_ili9341 spi0.0 32000kHz 8 bits mode=0x00
[    6.778158] systemd[1]: Started Setup Virtual Console.
[    6.802084] systemd[1]: Started Create list of required static device nodes for the current kernel.
[    6.848178] systemd[1]: Started udev Coldplug all Devices.
[   10.677177] bcm2708_i2c_init_pinmode(1,2)
[   10.704658] fb_ili9341 spi0.0: fbtft_gamma_parse_str() str=
[   10.728072] bcm2708_i2c_init_pinmode(1,3)
[   10.756899] fb_ili9341 spi0.0: 1F 1A 18 0A 0F 06 45 87 32 0A 07 02 07 05 00
00 25 27 05 10 09 3A 78 4D 05 18 0D 38 3A 1F
[   10.778728] bcm2708_i2c bcm2708_i2c.1: BSC1 Controller at 0x20804000 (irq 79) (baudrate 100000)
[   10.909418] fb_ili9341 spi0.0: fbtft_request_gpios: 'reset' = GPIO25
[   10.984271] fb_ili9341 spi0.0: fbtft_request_gpios: 'dc' = GPIO24
[   11.039735] fb_ili9341 spi0.0: fbtft_request_gpios: 'led' = GPIO18
[   11.084219] fb_ili9341 spi0.0: fbtft_verify_gpios()
[   11.158095] fb_ili9341 spi0.0: init_display()
[   11.213938] fb_ili9341 spi0.0: fbtft_reset()
[   14.103970] fb_ili9341 spi0.0: set_var()
[   14.153075] fb_ili9341 spi0.0: fbtft_update_display(start_line=0, end_line=239)
[   14.201803] fb_ili9341 spi0.0: Display update: 3079 kB/s (48.715 ms), fps=0 (0.000 ms)
[   14.201816] fb_ili9341 spi0.0: set_gamma()
[   14.202052] fb_ili9341 spi0.0: fbtft_register_backlight()
[   14.347981] fb_ili9341 spi0.0: fbtft_update_display(start_line=0, end_line=239)

notro, 2015-01-17:
Can you post a link to the product page for that display?

rdissertori, 2015-01-18:
yes of course there we go:

http://www.adafruit.com/product/1601
(http://www.amazon.co.uk/gp/product/B00H9B1DTA)

notro, 2015-01-18:
I guess you use the FBTFT prebuilt kernel and not the Adafruit one. If so use ```name=pitft```

rdissertori, 2015-01-18:
none of it. I've just compiled the kernel myself and inserted your fbtft modules

and oh my god - it just WORKED and my weston (wayland) desktop just showed up by changing the name.
so it turns out you have to use pitft (fb_ili9340) for 2.2" AND 2.8"

What caused my confusion:
I was reading the document on https://github.com/notro/fbtft/wiki/LCD-Modules
and had a peek at the 2.2" adafruit driver display table
So I assumed that for 2.8" I'd need a different driver => adafruit28 (fb_ili9341)

closed 2015-01-18 by rdissertori