notro/fbtft #232: Can't load fbtft driver on Banana Pro

NATSOFT, 2015-01-23:
[info] Loading kernel module fbtft_device.
        <6>fbtft_device:  SPI devices registered:
        [    8.776576] fbtft_device:  SPI devices registered:
        <6>fbtft_device:  'fb' Platform devices registered:
        [    8.786037] fbtft_device:  'fb' Platform devices registered:
        <3>fbtft_device:  spi_busnum_to_master(0) returned NULL
        [    8.796725] fbtft_device:  spi_busnum_to_master(0) returned NULL
        <3>fbtft_device: failed to register SPI device
        [    8.806910] fbtft_device: failed to register SPI device
        ERROR: could not insert 'fbtft_device': Invalid argument
        [info] Loading kernel module ads7846_device.
        <3>ads7846_device: spi_busnum_to_master(0) returned NULL.
        [    8.870014] ads7846_device: spi_busnum_to_master(0) returned NULL.
        ERROR: could not insert 'ads7846_device': Invalid argument

When I tried to load fbtft driver on my Banana Pro I got the errors abrove. How could I fix this?

NATSOFT, 2015-01-23:
Ok, I successfuly load the device now. However, there are just some strange lines on the tft... I am using this screen: http://www.wvshare.com/product/3.5inch-RPi-LCD-A.htm and my /etc/modules are as follow: 

        # /etc/modules: kernel modules to load at boot time.
        #
        # This file contains the names of kernel modules that should be loaded
        # at boot time, one per line. Lines beginning with "#" are ignored.
        # Parameters can be specified after the module name.

        snd-bcm2835
        i2c-bcm2708
        i2c-dev
        spi-sun7i
        flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
        fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 gpios=reset:25,dc:24
        ads7846_device model=7846 cs=1 gpio_pendown=17  keep_vref_on=1 swap_xy=0 pressure_max=255 x_plate_ohms=60 x_min=200 x_max=3900 y_min=200 y_max=3900

FishTest, 2015-01-25:
bananapi pro's gpio is different from raspberry pi B+

NATSOFT, 2015-01-25:
> bananapi pro's gpio is different from raspberry pi B+

Thanks for answering,

So there are nothing I can do? Could I modify the Device Tree file of FBTFT and make it suitable for Banana Pro?

notro, 2015-01-25:
Based on the comment by @FishTest , you must find suitable gpio numbers for these:
gpios=reset:25,dc:24
gpio_pendown=17

NATSOFT, 2015-02-07:
The screen can display now.

[image]

However there are still something wrong... there were something moving in the small dark area while booting, I think that might be demsg... 

Is the parameters for flexfb wrong?

notro, 2015-02-12:
> Is the parameters for flexfb wrong?

Most likely, yes.

notro, 2015-05-23:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2015-05-23 by notro

rogercorrea, 2015-09-02:
Please,
I need to run the same screen in my Banana Banana Pi or M2. Someone successful?

rogercorrea, 2015-09-06:
I could run in Banana Pi LeMaker.
Set up steps 1 and 2 of this link by changing FB01 to FB02 both in /etc/X11 as /usr /share/X11.
https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian

My /etc/modules was exactly the same as yours. Strange that only worked in fact restarted after a few times.

The developer, I think this makes your library also Banana Pi support. =)