notro/fbtft #235: ili9481 3.2'' TFT - Wiring and SPI problem

metropt, 2015-02-05:
Hello,

I'm trying to connect [this LCD](http://www.ebay.com/itm/3-2-TFT-LCD-Module-Display-with-Touch-PCB-adapter-262K-Color-Screen-480-X-320-/261626227679?pt=AU_B_I_Electrical_Test_Equipment&hash=item3cea236bdf]this lcd) to RPi.

I followed this schematic:
[image]
but can't make it to work. I'm connecting LCD_DC pin to GPIO24 because it is what the source code said:
```
RESET - GPIO25
DC - GPIO24
LED - GPIO22
```

I installed fbtft using this command:
```
sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update
sudo reboot
```
Then run
```
sudo modprobe fbtft_device name=ili9481
FRAMEBUFFER=/dev/fb1 startx
```

I connect my logic analyzer to it and it doesn't look good.
[image]
We got the CSX after 16bit count and it sounds ok. RESX is high. But DC and WR doesn't change accordingly. I can see it change but I think that the fbtft change in a way that this circuit doesn't work anymore :s

Another example:
[image]

Below there is an image from the datasheet:
[image]

EDITED: I read from other issues that it dmesg | grep fbtft_request_gpios should give me if the pins was initialized but it doens't return nothing.

mwilliams03, 2015-02-05:
I created that schematic a long time ago, and it was based of Spritemods schematic and valdodov's drivers.

For Notro's FBTFT, you need to use this circuit;
http://github.com/notro/fbtft/wiki/SPI-interface-circuit

bogdanul2003, 2015-02-17:
Hi,

I know this is not my post but I have a quick question that is related to this subject and I don't want make a new topic.

I see that most interfacing circuits transform SPI to a 16 or 8 bit parallel interface.
For ILI9341 there is the option of directly connect the chip to a 3 or 4 wire SPI interface (http://www.buydisplay.com/download/interfacing/ER-TFT032-3_Interfacing.pdf). Does the fbtft driver support direct SPI connection with ILI9341? Is it 3 or 4 wires ?

notro, 2015-05-23:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2015-05-23 by notro

t0n1, 2018-05-09:
> sudo modprobe fbtft_device should load it.
> You'll need to create this` /etc/modprobe.d/fbtft_device.conf`
> `options fbtft_device name=adafruit28 rotate=90`
> That Adafruit screen uses the same ili9481 controller on spidev0.0

https://www.raspberrypi.org/forums/viewtopic.php?p=684957#p684957