notro/fbtft #236: raspberry pi 2 use win10 and SNAPPY UBUNTU CORE  with fbtft

DRAgon734465502, 2015-02-06:
Dear notro
              New raspberry Pi 2 with the 1G ram can run win10 and ubuntu. HOW can i use the LCD with ili9341 controler  on  win10 or ubuntu?

notro, 2015-02-06:
I'm working on FBTFT for Pi 2: https://github.com/notro/rpi-firmware/issues/16
I have never looked a snappy core, but if the official kernel works with it, the the FBTFT kernel should also.
I won't be doing any Windows 10 stuff.

DRAgon734465502, 2015-02-07:
> I'm working on FBTFT for Pi 2: notro/rpi-firmware#16

thanks

transmissions4less, 2015-03-23:
i have Ubuntu 14.04 LTS installed on rpi2, can i just use yours fbtft also with the dma spi driver with RA8875 on it? please advise

notro, 2015-03-24:
If you can use the kernel installed with regular rpi-update, then you can use the fbtft kernel also.
I haven't tried ubuntu on the Pi myself.

DRAgon715205, 2015-04-04:
Raspberry Pi 3.5 inch IPS touch screen 480 320 display LCD  can run Ubuntu  OS
http://www.ebay.com/itm/331518596018?ssPageName=STRK:MESELX:IT&_trksid=p3984.m1555.l2649

DRAgon715205, 2015-04-04:
the seller offer the image and diver to run the LCD。
But I don't know How I can change the official SNAPPY UBUNTU CORE to support the LCD , THE lcd is controled by ILI9486

I want to do it by myself

who have the ideal？

closed 2015-06-18 by notro