notro/fbtft #241: compile a linux kenal with fbtft can work at B+ But not work at RPi2

DRAgon734465502, 2015-02-12:
Dear notor
         I   compile linux kernel  with fbtft ，But the fbtft driver can just work at B+， NOT at RPi2. . After Compile the Linux Kernel ,I get the kenal.img fle. but the kenal.img fle can't work at RPi2.   Seem at RPi2 the kenal.img don't work. the work kenal is kenal7.img.  
      
        How can I compile linux kernel  to get the kenal7.img.  file??  I  try many ways , JUST can get the  kenal.img but not the kenal7.img.

notro, 2015-02-12:
You need to use the ```bcm2709_defconfig``` config.

This is how I tag and make kernel7.img in my build script
```
/home/pi/tools/mkimage/mkknlimg /home/pi/rpi-firmware/fbtft/workdir.bcm2709/linux/arch/arm/boot/zImage /home/pi/rpi-firmware/fbtft/workdir.bcm2709/out/kernel7.img
```

DRAgon734465502, 2015-02-12:
Sorry about your build script.

I can't find "mkimage/mkknlimg"  at my mkimage document. This is what I can find

mkdir             mkfs.bfs          mkfs.msdos        mk_modmap
mkdiskimage       mkfs.cramfs       mkfs.ntfs         mknod
mkdosfs           mkfs.ext2         mkfs.vfat         mkntfs
mke2fs            mkfs.ext3         mkhomedir_helper  mksmbpasswd
mkfifo            mkfs.ext4         mkinitramfs       mkswap
mkfontdir         mkfs.ext4dev      mkisofs           mktemp
mkfontscale       mkfs.fat          mklost+found      mkzftree
mkfs              mkfs.minix        mkmanifest 

my linux kenal : 3.18.6    .   Can you tell me it's why??

DRAgon734465502, 2015-02-12:
where is the "mkknlimg"

notro, 2015-02-12:
It's part of the compiler repo: https://github.com/raspberrypi/tools/blob/master/mkimage/mkknlimg
It tags the kernel so the bootloader recognizes it as DT capable.

closed 2015-05-23 by notro