notro/fbtft #245: Where should I put the Device Tree overlay I just made?

notro, 2015-02-14:
How do we share the Device Tree overlays that will be made for the multitude of displays supported by FBTFT?

Discussion releated to this page: https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays

notro, 2015-02-14:
One solution is to create a github repo and use the wiki.
The wiki would have one page for each display.
Each page would have one section for each platform, since FBTFT now is used on many platforms, and the &spi and &gpio labels will differ, and not many supports overlays like RPi and BBB.

The Device Tree fragments could then be put in the [github markdown code tags](https://help.github.com/articles/github-flavored-markdown/#fenced-code-blocks).

notro, 2015-02-14:
Another approach would be if someone volunteered to maintain a github repo containing these overlays.
Platforms that don't have fragements could be DT include files (.dtsi).

acidjazz, 2015-02-26:
I've written an overlay for the JBTEK 3.5" LCD [here](https://github.com/acidjazz/jbtekoverlay)

I've had somewhat success on it working, but i'm still getting crashes and would love some help, ive been writing this by referencing older drivers that specify the following in `/etc/modules`

```config
flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
fbtft_device debug=3 rotate=90 name=flexfb speed=16 000 000 gpios=reset:25,dc:24
ads7846_device model=7846 cs=1 gpio_pendown=17  keep_vref_on=1 swap_xy=1 pressure_max=255 x_plate_ohms=60 x_min=200 x_max=3900 y_min=200 y_max=3900

```

swkim01, 2015-04-17:
For waveshare 3.2"/3.5" inch LCD, the device trees is <a href="https://github.com/swkim01/waveshare-dtoverlays">here</a>
The 3.5" overlay is almost same with the JBTek's. Thanks acidjazz.

notro, 2015-06-14:
I have added a section for overlays: https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays#overlays

closed 2015-06-14 by notro