notro/fbtft #246: 3.5inch RPi LCD (A) V3 issues with Raspberry PI 2 and Raspbian OS

ccadic, 2015-02-15:
Dear sir
I bought a  3.5inch RPi LCD (A) V3 from a HK supplier last week.
I've been trying to make it work on a Raspberry PI2 - Raspbian OS machine.
I have been fighting for 3 days now with no sucess

The hardware is detected as a: fb_ili9341 frame buffer, 320x240, 150 KiB video memory, 16 KiB DMA buffer memory, fps=33, spi0.0 at 32 MHz

root@raspberrypi:~# dmesg | grep graph

[    5.153178] graphics fb1: fb_ili9341 frame buffer, 320x240, 150 KiB video memory, 16 KiB DMA buffer memory, fps=33, spi0.0 at 32 MHz

The SPI mode shows

 dmesg | grep spi

[    4.599651] ads7846 spi0.1: touchscreen, irq 473
[    4.614311] input: ADS7846 Touchscreen as /devices/soc/3f204000.spi/spi_maste                              r/spi0/spi0.1/input/input2
[    5.153178] graphics fb1: fb_ili9341 frame buffer, 320x240, 150 KiB video mem                              ory, 16 KiB DMA buffer memory, fps=33, spi0.0 at 32 MHz
[    5.171669] bcm2708_spi 3f204000.spi: SPI Controller at 0x3f204000 (irq 80)
[    5.185442] bcm2708_spi 3f204000.spi: SPI Controller running in dma mode
[    7.723645] fbtft_device:      ads7846 spi0.1 2000kHz 8 bits mode=0x00
[    7.736609] fbtft_device:      ili9341 spi0.0 32000kHz 8 bits mode=0x00
[    7.773380] fbtft_device: Deleting spi0.0
[    7.785121] fb_ili9341 spi0.0: fbtft_request_gpios: gpio_request_one('reset'=                              25) failed with -16
[    7.800375] fb_ili9341: probe of spi0.0 failed with error -16
[    7.864452] fbtft_device:      ads7846 spi0.1 2000kHz 8 bits mode=0x00
[    7.876762] fbtft_device:      fb_ili9341 spi0.0 32000kHz 8 bits mode=0x00
[    7.937488] ads7846_device: Deleting spi0.1
[    8.051009] ads7846 spi0.1: touchscreen, irq 471
[    8.062577] input: ADS7846 Touchscreen as /devices/soc/3f204000.spi/spi_maste  

The lsmod says

root@raspberrypi:~# lsmod
Module                  Size  Used by
fuse                   82238  1
ads7846_device          3262  0
fbtft_device           27669  0
snd_bcm2835            18850  0
snd_pcm                76338  1 snd_bcm2835
snd_seq                56241  0
snd_seq_device          5632  1 snd_seq
snd_timer              17940  2 snd_pcm,snd_seq
snd                    52233  5 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device
8192cu                528469  0
fb_ili9341              3895  0
fbtft                  28286  2 fb_ili9341,fbtft_device
syscopyarea             2869  1 fbtft
sysfillrect             3305  1 fbtft
sysimgblt               1829  1 fbtft
fb_sys_fops             1141  1 fbtft
ads7846                10888  0
evdev                  10014  3
joydev                  8903  0
spi_bcm2708             7690  0
uio_pdrv_genirq         2958  0
uio                     8115  1 uio_pdrv_genirq

I am unable to undestand.
The Screen is supposed to do more than 320x240 (is this normal the dmesg says 320x240 only ? Did I buy a counterfiet product ?) (the seller and the WaveShare SpotPear 3.5inch RPi LCD (A) V3 says: "Raspberry Pi 3.5inch B/B+ HD 320*480 LCD Touch Screen V3 with Touch Pen"

The driver seems to be:  FBTFT and fb_ili9341
I'm with RASBERRY PI 2 (new model) - The system image you give on a DVD does not boot on raspberry 2
I'm unable to find the correct INIT string . I get error -16 or similar when I try to test various init strings

Any suggestion ?
Did someone managed to run this screen with PI 2?

notro, 2015-02-15:
The FBTFT drivers can't detect which hardware it's running on. The SPI bus doesn't have enumuration capabilities.
It has to be told explicitly what is connected.
This is usually done using the fbtft_device module, either from /boot/cmdline.txt or /etc/modules

> [ 7.785121] fb_ili9341 spi0.0: fbtft_request_gpios: gpio_request_one('reset'= 25) failed with -16

This error states that the driver couldn't load returning -EBUSY. This indicates that the gpio is already claimed by som other driver.

I suggest you contact the seller about this.

ccadic, 2015-02-18:
thank you . The seller is not reactive. I put those 2 screens in the bin, swapped with a SainSmart 3.2. Now fighting with a  fb_ssd1289.0: Missing info about 'dc' gpio. Aborting . 
Shoudld find in a minute

closed 2015-05-23 by notro