notro/fbtft #247: ILI9341 interface circuit

bogdanul2003, 2015-02-17:
Hi,

A short question regarding ILI9341 interface.

I see that most interfacing circuits transform SPI to a 16 or 8 bit parallel interface.
For ILI9341 there is the option of directly connect the chip to a 3 or 4 wire SPI interface (http://www.buydisplay.com/download/interfacing/ER-TFT032-3_Interfacing.pdf). Does the fbtft driver support direct SPI connection with ILI9341? Is it 3 or 4 wires ?

ladyada, 2015-02-20:
FBTFT ILI9341 supports 4-wire SPI only at this time

bogdanul2003, 2015-02-20:
Hi,

You are right, 4 wire SPI. I managed to make it work on a openwrt router based on AR9331 chip using a software emulated SPI.
The thing is that my SPI does not exceed 1 MHz using the standard spi-gpio driver. This makes refresh rate kind of slow.
Do you have any suggestions that may speed things up a bit ?

notro, 2015-02-20:
> Do you have any suggestions that may speed things up a bit ?

```c
/*
 * Because the overhead of going through four GPIO procedure calls
 * per transferred bit can make performance a problem, this code
 * is set up so that you can use it in either of two ways:
 *
 *   - The slow generic way:  set up platform_data to hold the GPIO
 *     numbers used for MISO/MOSI/SCK, and issue procedure calls for
 *     each of them.  This driver can handle several such busses.
 *
 *   - The quicker inlined way:  only helps with platform GPIO code
 *     that inlines operations for constant GPIOs.  This can give
 *     you tight (fast!) inner loops, but each such bus needs a
 *     new driver.  You'll define a new C file, with Makefile and
 *     Kconfig support; the C code can be a total of six lines:
 *
 *              #define DRIVER_NAME     "myboard_spi2"
 *              #define SPI_MISO_GPIO   119
 *              #define SPI_MOSI_GPIO   120
 *              #define SPI_SCK_GPIO    121
 *              #define SPI_N_CHIPSEL   4
 *              #include "spi-gpio.c"
 */
```
http://lxr.free-electrons.com/source/drivers/spi/spi-gpio.c#L56
I don't know if this applies to your platform though.

bogdanul2003, 2015-02-20:
Can you explain a bit?
I also came across this but didn't quite get it.
From the comment above I understand that only when doing "#define SPI_MOSI_GPIO   120" will make a function to be inlined ? I see that the functions that interact with GPIO's are all defined as "static inline" (static inline void setsck(const struct spi_device *spi, int is_on)). The function looses it's inline property in some conditions ?

Thanks

notro, 2015-02-20:
I haven't studied this, but I guess it has to do with how [gpio_set_value_cansleep](http://lxr.free-electrons.com/ident?i=gpio_set_value_cansleep) is defined on your platform. If it is a macro/inline.

Another solution is to make a copy of spi-gpio and then in set{sck,mosi} write directly to the gpio hardware register instead of using gpio_set_value_cansleep. This should give some boost.

You would have to look at the asm output to see if you get it inlined. I don't have any knowledge about compiler optimization, execpt that it usually "does the right thing" :-)

One of these gives asm output, don't remember which:
```text
make help
[...]
  dir/file.[oisS] - Build specified target only
```
https://kernel.org/doc/makehelp.txt

bogdanul2003, 2015-02-20:
I'll take a look and do some speed tests with what you suggested later.
At the moment I've put a logic analyzer on the SPI and I saw that fbtft driver sends the entire video memory once every 0.163 seconds. Is this ok? I did a print screen of the output ( http://wikisend.com/download/658076/salea.png ). 
I thought that data will be sent to ILI9341 IC only when I write something in the video memory. At the moment of the capture there was no application writing to the framebuffer device /dev/fb0.
Is it possible that deferred IO is not working ?

notro, 2015-02-20:
> At the moment I've put a logic analyzer on the SPI and I saw that fbtft driver sends the entire video memory once every 0.163 seconds. Is this ok?

If the console is mapped to the framebuffer it would send a partial update (cursor blinking) roughly around every 1/20 seconds. con2fbmap shows mappings.

If nothing uses the framebuffer, you shouldn't get any output.

If you set debug=7 you will get info about everything FBTFT does in the kernel log.
You can compare that to what you get on the pins.

closed 2015-05-23 by notro