notro/fbtft #248: SainSmart 3.2 - It is killing me not to find :=)

ccadic, 2015-02-18:
This week I bought 3 different displays for my new Raspberry PI 2  (one 3.5 inch LCD A V3 Waveshare SpotPear which I discovered to be impossible to configure with the PI2, the 2 4inch I also discovered to be a brother of the 3.5 inch LCD A V3 Waveshare SpotPear , so impossible to configure ).
Now I alm on the SainSmart 3.2 . I'm still stuck on a white blank screen on my PI2 + Rasbian  updated

The driver seems to work at bootup

dmesg sounds ok

[    7.617700] fbtft_device:  SPI devices registered:
[    7.628596] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[    7.640889] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[    7.652998] fbtft_device:  'fb' Platform devices registered:
[    7.664364] fbtft_device:      bcm2708_fb id=-1 pdata? no
[    7.675450] fbtft_device: Deleting spi0.0
[    7.685641] fbtft_device:  GPIOS used by 'sainsmart32_spi':
[    7.697721] fbtft_device:    'reset' = GPIO25
[    7.708366] fbtft_device:    'dc' = GPIO24
[    7.719660] fbtft_device:  SPI devices registered:
[    7.731001] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[    7.743112] fbtft_device:      fb_ssd1289 spi0.0 16000kHz 8 bits mode=0x00
[    7.944884] graphics fb1: fb_ssd1289 frame buffer, 320x240, 150 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 16 MHz

/boot/config.txt is

dtparam=spi=on
dtoverlay=sainsmart32_spi

The blacklist file is empty

The /etc/modules has only 2 lines:
snd-bcm2835
fbtft_device name=sainsmart32_spi rotate=90

Screen stays desperately white.
Do I need to send a kind of init string ? How ?

The con2fbmap 1 1 does nothing (in a ssh terminal, not in local)

Does any one know if there is a step by step for the PI2 and this screen
I do not want to buy all the collection of screens from different brands of the market :) Just need a small system for my holidays :)

ccadic, 2015-02-18:
Also tried to remove everything and  do this
 
joe /boot/config.txt
dtparam=spi=on

Then from the cmd line

modprobe fbtft_device name=flexfb speed=16000000 gpios=reset:25,dc:24
$ modprobe flexfb debug=3 width=240 height=320 rotate=0 regwidth=16 setaddrwin=2 init=-1,0x00,0x0001,-1,0x03,0xa2a4,-1,0x0c,0x0004,-1,0x0d,0x0308,-1,0x0e,0x3000,-1,0x1e,0x00ea,-1,0x01,0x2b3f,-1,0x02,0x0600,-1,0x10,0x0000,-1,0x11,0x6030,-1,0x07,0x0233,-1,0x0b,0x0039,-1,0x0f,0x0000,-1,0x03,0x0707,-1,0x31,0x0204,-1,0x32,0x0204,-1,0x33,0x0502,-1,0x34,0x0507,-1,0x35,0x0204,-1,0x36,0x0204,-1,0x37,0x0502,-1,0x3a,0x0302,-1,0x3b,0x0302,-1,0x23,0x0000,-1,0x24,0x0000,-1,0x25,0xa000,-1,0x48,0x0000,-1,0x49,319,-1,0x4a,0x0000,-1,0x4b,0x0000,-1,0x41,0x0000,-1,0x42,0x0000,-1,0x44,0xef00,-1,0x45,0x0000,-1,0x46,319,-3

I get a 

[   75.811985] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[   75.812002] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   75.812099] fbtft_device: Deleting spi0.0
[   75.812778] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   75.812792] fbtft_device:      flexfb spi0.0 16000kHz 8 bits mode=0x00
[   85.224762] graphics fb1: flexfb frame buffer, 240x320, 150 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 16 MHz

But the con2fbmap 1 1 does not work either

I'm completely lost . Should I understand thazt the PI2  might be not compatible with LCD ?

notro, 2015-02-20:
The Sainsmart 3.2 display has a parallel bus. Do you have an interface circuit in front of it?
It's needed if you connect to the SPI bus: https://github.com/notro/fbtft/wiki/SPI-interface-circuit

You can connect the display using the 16-bit databus directly.
This is the command I used to test it on a regular Model B over a year ago:
```
sudo modprobe fbtft_device name=sainsmart32 gpios= reset:29,dc:2,wr:31,db00:11,db01:9,db02:10,db03:22,db04:27,db05:17,db06:4,db07:3,db08:7,db09:8,db10:25,db11:24,db12:23,db13:18,db14:14,db15:15
```

Spektre99, 2015-04-16:
Hello.  Great project.

Is there a reason why we cannot use 16 GPIO pins as a databus instead of using what must be a much slower shift register method?  If there isn't a compelling reason would you be able to point me to where in the code I might be able to modify to use it instead of the SPI bus?

notro, 2015-04-17:
If you read the previous comment you'll see how to do that.

Spektre99, 2015-05-14:
Thanks Notro, I'm going to try this out!

Since the driver code is now built into the kernel, I should be able to run the latest version of Raspbian and just type in that command (with the proper wiring) and see video?

I'm assuming the mappings listed in the command you have given can be changed to any suitable GPIO pins yes?

notro, 2015-05-15:
You should be able to use the latest Raspian, and you can use any of the available gpios.

Spektre99, 2015-06-28:
Notro.

Looking up the touchscreen capabilities for this leads me to a page referencing the ADS7846.  This display seems to use the ADS7843.  Same commands to get either to work?

notro, 2015-06-29:
ADS7843 uses the same driver, but pressure calculation is different for it: http://lxr.free-electrons.com/source/drivers/input/touchscreen/ads7846.c#L774

You need this comatible string: "ti,ads7843"

notro, 2015-09-06:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2015-09-06 by notro

11ishan, 2016-08-01:
Hey notro, 
I'm facing a similar problem as this guy.
http://raspberrypi.stackexchange.com/questions/37611/tft-kernel-module-not-loading-on-startup

Please could you help me out?
Thank you