notro/fbtft #249: Module stmpe_device not found

edmandiesamonte, 2015-02-19:
Hi,

I'm having trouble setting up the touchscreen panel of PiTFT on ArchLinuxARM. I was able to make the display work by following the instructions found here: https://github.com/notro/fbtft/issues/116. However, I can't seem to run the mentioned `modprobe stmpe_device` . I'm getting 'Module stmpe_device not found'.

I'm using the Capacitive version of the PiTFT. 

Thanks.

ladyada, 2015-02-20:
The capacitive version of PiTFT does not use STMPE610, it uses the FT6x06!
see https://github.com/adafruit/adafruit-rpi-fbtft/commits/master for some commits on adding FT6x06 support

edmandiesamonte, 2015-02-20:
Hello ladyada,

Thanks for the response. I've checked the link that you've provided. It seems that ft6x06 is already supported by adafruit's fbtft. Is it already merged with notro's fbtft? Or should I build it inside ArchLinux?

Apologies but I'm fairly new to linux kernels and such. I use the rpi-update tutorial by notro in order to setup the display using PiTFT.

Thanks again.

ladyada, 2015-02-20:
we're working on integrating it into 3.18 and device tree overlay. the code that's up is compatible with at least 3.6, 3.12 and 3.15. no eta on DTO support!

edmandiesamonte, 2015-02-20:
Thanks again for the response.

Is it possible that I could use [Adafruit's rpi-firmware](https://github.com/adafruit/rpi-firmware) instead of notro's to make it work with my ArchLinux ARM installation? Or I really need to build the linux kernel with the patch from scratch?

ladyada, 2015-02-20:
in theory you should be able to run the linux kernel on any linux derivative...but Adafruit has no support, tutorials, or ETA on Arch support - you are on your own!

edmandiesamonte, 2015-02-20:
Thanks ladyada. At least I have something to start with.

edmandiesamonte, 2015-02-20:
I was able to try rpi-update using Adafruit's firmware. I found adafruitct28 as the appropriate name for the fbtft_device. However, I'm getting this weird message (this might not be weird at all. I'm probably just new to this):
```
fbtft_device: init
[  880.331079] fbtft_device:  SPI devices registered:
[  880.343171] fbtft_device:  'fb' Platform devices registered:
[  880.353701] fbtft_device:      bcm2708_fb id=-1 pdata? no
[  880.363354] fbtft_device:  I2C devices registered:
[  880.372121] fbtft_device:       i2c-0  addr 48911  IRQ 0
[  880.381425] fbtft_device:       i2c-1  addr 48911  IRQ 1
[  880.390701] fbtft_device:      wm8804 1-003b  addr 59  IRQ 0
[  880.401104] fbtft_device:      pcm5122 1-004c  addr 76  IRQ 0
[  880.411654] fbtft_device:      ft6x06_ts 1-0038  addr 56  IRQ 0
[  880.422360] fbtft_device:  name='adafruitct28', busnum=0, cs=0
[  880.422400] fbtft_device: Looking at item 0
[  880.422412] fbtft_device: Setting pin 24 to 2
[  880.506617] fbtft_device:    i2c_new_device() returned NULL
[  880.519406] fbtft_device: failed to register I2C device
```
As far as I'm aware, ft6x06 uses i2c and was successfully registered base on the message. But the confusing part is the `i2c_new_device() returned NULL`. I wonder why is ft6x06 is registered even if the message says `failed to register I2C device`?

I already run modprobes for spi_bcm2708 and i2c_bcm2708.

FishTest, 2015-02-25:
I'm using pitft(r) too.i can compile fbtft and make it successfully display,but i can't get the touchscreen working..
I'm trying to compile fbtft_device from https://github.com/notro/fbtft_tools with 3.18.7 kernel
but it give me some errors.
can u help me to solve this problem?

even if the touchscreen working ,but when i want to compile other device drivers ,i still have no good method to get the source code math the exist kernel......

notro, 2015-05-23:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2015-05-23 by notro

ladyada, 2015-05-25:
woops forgot to catch up on this thread, but we did get all of this working very well thanks to help from @notro and adafruit team :)
https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch has the tutorial