notro/fbtft #250: how to un-invert either x or y axis in ads7846 touchscreen?

thebestofall007, 2015-02-19:
Good to see this go into staging! The screen model is a waveshare32b. I was able to get buildroot to work fabulously in buildroot through peter seiderer's patch in this mailing list http://lists.busybox.net/pipermail/buildroot/2015-January/116511.html  but I am having an issue with the axes on the ads7846 touchscreen. I was able to swap the axes with the swap_xy=1 option, but I am stuck with either the x or y axis being inverted. I am using the options in an /etc/init.d script that starts up at boot, and the options for the driver being loaded at boot. For example, I can move my finger on the x axis and it works just fine, but the y axis is inverted. I tried to rotate the screen 90 degrees instead of 270 (flipping it over horizontally), and then the x axis is inverted. What is the issue here, and how can I get both axes working. I am not using x11, but instead directfb and fbdev, as those are a lot faster than xorg for my application (a diy touchscreen car stereo headunit).

Here is my code if you would like to take a look:

#!/bin/sh
#
# Start waveshare32b (fbtft_device, ads7846_device)
#

case "$1" in
  start)
        echo -n "Starting waveshare32b: "
        modprobe fbtft_device name=waveshare32b rotate=270
        # from https://github.com/notro/fbtft/issues/181
        modprobe ads7846_device model=7846 cs=1 gpio_pendown=17 speed=1000000 keep_vref_on=1 swap_xy=1 pressure_max=255 x_plate_ohms=60 x_min=200 x_max=3900 y_min=200 y_max=3900
        # from http://www.raspberrypi.org/forums/viewtopic.php?f=63&t=95231
        # modprobe ads7846_device model=7846 cs=1 gpio_pendown=17 speed=1000000 keep_vref_on=1 swap_xy=0
        echo "OK"
        ;;
  stop)
        echo -n "Stopping waveshare32b: "
        rmmod fbtft_device
        rmmod fb_ili9340
        rmmod ads7846_device
        rmmod ads7846
        echo "OK"
        ;;
  restart|reload)
        ;;
  *)
        echo "Usage: $0 {start|stop|restart}"
        exit 1
esac

exit $?

notro, 2015-02-20:
Can you use tslib with directfb?
https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian#tslib-sdl-pygame

thebestofall007, 2015-02-20:
@notro, Yes. I read the link you gave me, and my buildroot system already has those ts_calibrate and ts_test commands located at /usr/bin, which I believe are enabled thru tslib. I also looked at the buildroot configuration menu and there is a package that enables directfb tslib support (it's called BR2_PACKAGE_DIRECTFB_TSLIB), which is enabled in my setup. I tested the touchscreen with df_bltload, a directfb application that draws a black screen and an arrow for testing purposes.

notro, 2015-02-20:
So, which part of the chain doesn't work?

thebestofall007, 2015-02-20:
@notro, not sure yet.

thebestofall007, 2015-02-21:
I can run ts_calibrate, but the interface doesn't respond to touch, despite exporting 
with "export TSLIB_FBDEVICE=/dev/fb1" and "export TSLIB_TSDEVICE=/dev/input/event2" then running ts_calibrate. If I try to create the udev rules to make a symbolic link to /dev/touchscreen or whatever, I lose touchscreen altogether, so only part of these instructions on the link are applicable to my situation, as I'm not running raspbian.

notro, 2015-02-21:
Do you run ts_calibrate as a root user or through sudo? If through sudo does the en vars pass through: sudo env | grep TSLIB ?

> Why doesn't the driver itself have that parameter??

Well I guess it's because no one needed it enough to send a patch to mainline.
You can also just patch the file yourself and invert one of the axis.

thebestofall007, 2015-02-21:
@notro I'm running it as root, not sudo.

closed 2015-05-23 by notro