notro/fbtft #251: MD070SD shifted row and column lines

friek2k, 2015-02-22:
Hi,
after testing the Sainsmart SSD1963 with fbtft, I got always an tearing effect with X11, so i ordered the new version of the 7" display with an MD070SD controller. The display does not need all this painful initialization code, so I just set the back light. 

sudo modprobe --first-time fbtft_device name=flexpfb debug=3 rotate=0 verbose=3  gpios=reset:2,dc:3,led:21,wr:4,db00:17,db01:27,db02:22,db03:6,db04:13,db05:19,db06:26,db07:20,db08:14,db09:15,db10:18,db11:23,db12:24,db13:25,db14:12,db15:16
sudo modprobe --first-time flexfb width=800 height=480 buswidth=16 init=-2,1,-1,0x01,16,-3

Generally the display seems to work with fbtft but I think the row lines and columns are shifted. I think this is because of the automatic line/column address calculation and increment of the controller. I tried to disable this feature, but there is no way doing this at with initialization code.

I double checked the wiring, and tested with fbtest. Photos are attached.

You can find the documentation at:
http://www.sainsmart.com/sainsmart-7-7-inch-tft-lcd-display-cpld-sdram-800x480-for-arduino-uno-mega2560.html

There is nearly the same display board with the same documentation from Iteadstudio. 
But the Sainsmart one has the additional feature to power it with 5V 500mA and 3.3V 300mA, which I think is perfect for the Pi B+ header.

http://imall.iteadstudio.com/md070sd-7-16-bit-parallel-bus-tft-module.html
ftp://imall.iteadstudio.com/TFT%20LCM/IM130820001/DS_IM130820001.pdf

What do you think about the problem, is there a way to get this working with the driver?       
Thanks in advance and for everything you've done here with this great driver!

[image]
[image]

notro, 2015-02-22:
Slowing down I/O might solve this.
The current FBTFT kernel has optimized parallel I/O: https://github.com/notro/fbtft/blob/master/fbtft-io.c#L187
The next release will not, because FBTFT has moved into mainline and I will use that codebase: https://github.com/torvalds/linux/commit/e60088575bcbaf8d71a0e9f1d44fbf136f95e2cd

But I'm really at loss with this problem and I don't have equipment to analyze what's going on with the bus signal waveforms.
Here's some discussions I've participated in regarding this topic:
http://www.raspberrypi.org/forums/viewtopic.php?f=41&t=92967
http://www.raspberrypi.org/forums/viewtopic.php?f=44&t=85219

younatan, 2015-02-28:
Hi friek2k may I ask your wiring? Because my LCD won't turn on

friek2k, 2015-03-01:
@notro: thx for your short answer. I'll try the not optimized codebase with the SSD1963 controller. But the problem described relates to a different controller, the MD070SD. 

@younatan: I have mentioned two different controllers.

<b>Sainsmart 7" with SSD1963 controller</b>
3.3V  - 3.3V    - P01
RST   - GPIO2   - P03
RS/DC  - GPIO3    - P05
WR       - GPIO4    - P07
CS        - GND       - P09
DB00    - GPIO17  - P11
DB01    - GPIO27  - P13
DB02    - GPIO22  - P15
RD        - 3.3V       - P17
DB03    - GPIO6    - P310                         
DB04    - GPIO13  - P33
DB05    - GPIO19  - P35
DB06    - GPIO26  - P37
DB08    - GPIO14  - P08
DB09    - GPIO15  - P10
DB10    - GPIO18  - P12
DB11    - GPIO23  - P16
DB12    - GPIO24  - P18
DB13    - GPIO25  - P22
DB14    - GPIO12  - P32
DB15    - GPIO16  - P36
DB07    - GPIO20  - P38
LED_A  - GPIO21  - P40 

With this wiring the default SPI Pins are free for the touch.
For me the display also works with 5V instead of 3.3V directly from the Rpi B+ header. 
Unlilke the MD070SD, do not wire 5V and 3.3V!!! 
Wiring LED_A is necessary because by default the backlight of the display is off. 
As reported this display works well except for X11.
```
modprobe --first-time fbtft_device name=flexpfb debug=3 rotate=0 verbose=0 fps=25 gpios=reset:2,dc:3,wr:4,led:21,db00:17,db01:27,db02:22,db03:6,db04:13,db05:19,db06:26,db07:20,db08:14,db09:15,db10:18,db11:23,db12:24,db13:25,db14:12,db15:16

modprobe --first-time flexfb width=800 height=480 buswidth=16 init=-1,0xE2,0x23,0x02,0x04,-1,0xE0,0x01,-2,10,-1,0xE0,0x03,-2,10,-1,0x01,-2,100,-1,0xE6,0x04,0x93,0xE0,-1,0xB0,0x00,0x00,0x03,0x1f,0x01,0xdf,0x00,-1,0xB4,0x03,0xA0,0x00,0x2E,0x30,0x00,0x0f,0x00,-1,0xB6,0x02,0x0D,0x00,0x10,0x10,0x00,0x08,-1,0xBA,0x05,-1,0xB8,0x07,0x01,-1,0x36,0x00,-1,0xf0,0x03,-2,10,-1,0x29,-1,0xBE,0x06,0xf0,0x00,0x00,-1,0xd0,0x0d,-1,0x2C,-3
```

<b>Sainsmart 7" with MD070SD  controller</b>

The wiring is nearly the same. Just wire 3.3V to P01 <u>and</u> 5V to P02 or use the extra USB-Power port. LED_A is optional, because per default the backlight jumper is bridged.
You'll not need additional initialization code but as reported in my first post, rows and columns are shifted on every redraw.
```
sudo modprobe --first-time fbtft_device name=flexpfb debug=3 rotate=0 verbose=3 gpios=reset:2,dc:3,led:21,wr:4,db00:17,db01:27,db02:22,db03:6,db04:13,db05:19,db06:26,db07:20,db08:14,db09:15,db10:18,db11:23,db12:24,db13:25,db14:12,db15:16

sudo modprobe --first-time flexfb width=800 height=480 buswidth=16 init=-2,1,-1,0x01,16,-3
```

notro, 2015-03-01:
The latest foundation kernel has fbtft support now (plain rpi-update), and the optimization code is removed.

Akex2, 2015-03-01:
Hi freak2k 
Sorry i dont speak english very well ...
I want a lcd touch screen for my rpi b+
I see there are 2 ci ssd1963 and md050sd ( for 5inch display )
Can tou say me for you the best of them ?
And do you have a tactil touch working on your display ?

friek2k, 2015-03-08:
Hi Notro,
the SSD1963 works like a charm now with X11. But the MD070SD controller has still the same problem. I think this is because the controller is calculating the new row and column addresses himself and/or also adds line and column initialization/wrap code himself. So there is always a double line/column shift. There is no way to disable this controller feature.

A video showing the problem: 
http://thinkzone.de/temp/MD070SD/shifted.mp4

The automatic line and column wrap is also documented at ftp://imall.iteadstudio.com/TFT%20LCM/IM130820001/DS_IM130820001.pdf

> 4.2 Writing of row and column addresses
> The corresponding RAM address of row and column addresses is obtained by internal arithmetic of the control board , thus users do not need to calculate the correspondence between the row and column addresses and the RAM address, and they can directly enter coordinates addresses.

> 4.3 Row and column address increment direction
> Row and column address increment direction can be achieved by configuring "address increment direction register (0x0D)". The address is automatically incremented by 1 when writing continuously. The control board can be set as the row direction or the column direction address auto increment. When it comes to the end of the row, it will circulate to the beginning of the row.

@Akex2: For several reasons I think the Sainsmart MD050SD fits better for the Raspberry PI. Just the FBTFT driver is not yet working.  The Sainsmart SSD1963 works fine with FBTFT if using the latest firmware, like Notro said. I have not yet tested the touch.

notro, 2015-03-10:
Sorry, but I haven't got time to look into this.

clouetb, 2015-03-14:
Hi guys,
I think I'm close to get an MD070SD working, using the wiring and the parameters you gave above, friek2k.
```
sudo modprobe --first-time fbtft_device name=flexpfb debug=3 rotate=0 verbose=3 gpios=reset:2,dc:3,led:21,wr:4,db00:17,db01:27,db02:22,db03:6,db04:13,db05:19,db06:26,db07:20,db08:14,db09:15,db10:18,db11:23,db12:24,db13:25,db14:12,db15:16

sudo modprobe --first-time flexfb width=800 height=480 buswidth=16 init=-2,1,-1,0x01,16,-3
````
See the output on the screen below, I don't know if this is the kind of shift you mentioned.
[image]
The question is now how to offset the display so that the screen is correctly rendered.

I also noticed something strange, I don't know if it's related though:
```
pi@pi2 ~ $ fbset

mode "720x480"
    geometry 720 480 720 480 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/16
endmode
````
I tried to force the correct framebuffer size with 
```
fbset -xres 800 -yres 480 -depth 16
```
but got no luck that way.

I also noticed an error in the Xorg.0.log
```
(EE) FBDEV(0): FBIOBLANK: Invalid argument
```

I'll keep you posted if I get something interesting.

younatan, 2015-03-17:
Hi notro thank you it works. I am trying to use SPI with md070sd. Is there any problem with touchscreen if I used SPI for interface?

notro, 2015-03-17:
> I am trying to use SPI with md070sd. Is there any problem with touchscreen if I used SPI for interface?

At least in theory it should work. I have never tried using a SPI interface circuit together with a SPI touch controller myself.

Akex2, 2015-03-18:
Hi younatan
I have a same screen, and don t work
Md050sd
What do you do? For work it please
Thanks
Alex

friek2k, 2015-03-21:
Hi clouetb, 
yes its the same issue on my screen. The shift is better visible at the console ("con2fbmap 1 1"). Have you tried that?

younatan, 2015-05-04:
Hi akex2 I used the same code. What is your problem? maybe the supply is not enough?

younatan, 2015-05-04:
Hi notro can mdo70sd use regwidth16? because mine cannot on when I use SPI

younatan, 2015-05-06:
I found the problem for md070sd. the addressing modecommand from notro driver different with md070sd.

Akex2, 2015-05-07:
Hy younatan
I have a same probleme of you
Do you can resolve this problem ?
Thx alex

notro, 2015-09-06:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2015-09-06 by notro

maxstrauch, 2020-09-27:
@notro @friek2k I've gotten my hands on a `md050sd` display (identical to the 7'' versions, but only 5'') and I have the same issue with shifted lines. Is there a solution for this problem? I tried all commands here in the issue but nothing helped! It would be nice if you could give me some hints how to resolve it. 😄