notro/fbtft #252: TFT35" Touch Shield For RPi SKU: AC-TFT35V-2.1

lcwii, 2015-02-27:
Anyone seen this one? Looks similar to the Waveshare. Acquired from eBay.
[image]

notro, 2015-02-28:
Looking at the ic's, it uses this interface circuit: https://github.com/notro/fbtft/wiki/SPI-interface-circuit
That makes me think that it could be a PiScreen clone.
If so, the latest Pi foundation kernel (rpi-update) and /boot/config.txt: dtoverlay=piscreen, should work.
Check here to see if the init sequence matches the one you have: https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/boot/dts/piscreen-overlay.dts

lcwii, 2015-03-04:
Got in a Pi B+ today and looked at some of the files.

lsmod reports
Module                  Size  Used by
joydev                  9257  0
ads7846                 8720  0
ads7846_device          6125  0
fbtft_device           34686  0
flexfb                 14012  2
fbtft                  30834  2 flexfb,fbtft_device
syscopyarea             2999  1 fbtft
sysfillrect             3294  1 fbtft
sysimgblt               2124  1 fbtft
fb_sys_fops             1435  1 fbtft
i2c_dev                 7215  0
snd_bcm2835            19496  0
snd_soc_tas5713         5573  0
snd_soc_bcm2708_i2s     6186  0
regmap_mmio             2818  1 snd_soc_bcm2708_i2s
snd_soc_core          146249  2 snd_soc_tas5713,snd_soc_bcm2708_i2s
snd_compress           12594  1 snd_soc_core
regmap_i2c              1661  2 snd_soc_core,snd_soc_tas5713
snd_pcm_dmaengine       5505  1 snd_soc_core
regmap_spi              1913  1 snd_soc_core
snd_pcm                83853  3 snd_bcm2835,snd_soc_core,snd_pcm_dmaengine
snd_page_alloc          5140  1 snd_pcm
snd_seq                55476  0
snd_seq_device          6465  1 snd_seq
snd_timer              20981  2 snd_pcm,snd_seq
evdev                  10713  8
leds_gpio               2079  0
led_class               4373  1 leds_gpio
snd                    62252  7 snd_bcm2835,snd_soc_core,snd_timer,snd_pcm,snd_seq,snd_seq_device,snd_compress
spi_bcm2708             8111  0
i2c_bcm2708             4943  0

I tried to use this as a guide: http://www.circuitbasics.com/setup-lcd-touchscreen-raspberry-pi/ using the contents of the files found on the Pi B+. It did not work. I did see the device name is flexfb and the touch device is an ads7846. The touch driver does not seem to load, although the fbtft driver do seem to load. I am new to Linux and the Pi. I don't know how to check the init sequence to see if it matches. I also tried to modify /boot/config.txt with the addition of the line dtoverlay=piscreen.

lcwii

notro, 2015-03-04:
If you add this to /boot/cmdline.txt you'll get the applied init sequence output in the kernel log. Shown with dmesg.
```
fbtft_device.debug=3
```

lcwii, 2015-03-04:
Ok I put the fbtft_device.debug=3 line in /boot/config.txt. 
/boot/config.txt now looks like this:

dtparam=spi=on
dtparam=i2c_arm=on
dtoverlay=piscreen
fbtft_device.debug=3

The only part of dmesg that seems to be pertinent is the following, it 
appears there is an error on init

[   18.044018] i2c /dev entries driver
[   18.122712] flexfb: `-1' invalid for parameter `init'
[   18.206812] fbtft_device:  SPI devices registered:
[   18.206834] fbtft_device:  'fb' Platform devices registered:
[   18.206859] fbtft_device:      bcm2708_fb id=-1 pdata? no
[   18.206891] fbtft_device:  spi_busnum_to_master(0) returned NULL
[   18.207007] fbtft_device: failed to register SPI device
[   21.512559] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote 
wakeup
[   23.204597] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, 
lpa 0x41E 1
[   24.645373] Bluetooth: Core ver 2.19

I am also not seeing the ads7648_device anywhere. Here is my /etc/modules

# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
# Parameters can be specified after the module name.

snd-bcm2835
i2c-bcm2708
i2c-dev

flexfb  width=320  height=480  regwidth=16 
init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00$
fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 
gpios=reset:25,dc:24
ads7846_device model=7846 cs=1 gpio_pendown=17  keep_vref_on=1 swap_xy=1 
pressure_max=255 x_plate_ohms=60 x_min=200 x_max=3$

Looking at this again, it looks like the fbtft_device was already set to 
a debug level of 3... I make that assumption not knowing how 
/etc/modules work versus /boot/config.txt

lcwii

lcwii, 2015-03-04:
Notro,

     I also just noticed that in dmesg the SPI fails to register the 
device...

lcwii

notro, 2015-03-04:
You should put debug it in /boot/cmdline.txt not config.txt
But only if you use fbtft_device to load the device.

If you use Device Tree to load the device, do it like this in /boot/config.txt
dtoverlay=piscreen,debug=3

> [   18.206891] fbtft_device:  spi_busnum_to_master(0) returned NULL

This means that spi-bcm2708 is not loaded.

Which kernel are you using?

lcwii, 2015-03-04:
Notro,

     I can't read I guess, sorry about getting the files confused... 
3.18.6-v7i-aufs

lcwii

lcwii, 2015-03-04:
Notro,

     Could this be my problem?

http://raspberrypi.stackexchange.com/questions/27073/firmware-3-18-x-breaks-i2c-spi-audio-lirc-1-wire-e-g-dev-i2c-1-no-such-f

     I have turned on Device tree using raspi-config and according to 
this webpage it breaks I2C, I2S & SPI.

lcwii

notro, 2015-03-04:
> 3.18.6-v7i-aufs

This kernel is unknown to me.

If you have the folder ```/proc/device-tree``` with contents, you have a Device Tree enabled kernel.
/boot/config.txt options if you have DT:
- dtparam=spi=on  
  This enables SPI so you can use fbtft_device
- dtoverlay=piscreen  
  This enables SPI and adds a PiScreen device for use with FBTFT

lcwii, 2015-03-04:
Notro,

     I have /proc/device_tree with contents, my /boot/config.txt is as 
follows:

dtparam=spi=on
dtparam=i2c_arm=on
dtoverlay=piscreen,debug=3

     When I rebooted I noticed several ERROR messages.

[info] Loading kernel module snd-bcm2835.
[     0.107655] flexfb: '-1' invalid for parameter 'init'
[     0.193113] fbtft_device:  spi_busnum_to_master(0) returned NULL
[     0.193236] fbtft_device: failed to register SPI device
[info] Loading kernel module i2c-bcm2708
[info] Loading kernel module i2c-dev
[info] Loading kernel module spi-bcm2708
ERROR: could not insert 'spi-bcm2708' : No such device
[info] Loading kernel module flexfb.
ERROR: could not insert 'flexfb': invalid argument
[info] Loading kernel module fbtft_device.
ERROR: could not insert 'fbtft_device': invalid argument
[info] Loading kernel module ads7846_device.
FATAL: Module ads7846_device not found.
[ok] Cleaning up temporary files... /tmp.

lcwii

krylm, 2015-03-04:
I bought same TFT35" Touch Shiled For RPi. SKU: AC-TFT35-V2.1 from dx.com `http://eud.dx.com/product/844363295`
After 2-3 hours of googling and trying I am finally done!

That worked:

I've made raspi-config to enable *SPI and I2C*
I've installed *FBTFT drivers*:
`sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update`

I've modified */etc/modules*
`fbtft_device name=piscreen rotate=90 speed=16000000`

And I've added into */boot/config.txt*
`dtoverlay=piscreen
fbtft_device.debug=3`

After reboot is TFT screen finally black (instead of white) and after typing 
`con2fbmap 1 1`
I can see login information.

Thanks a lot!

lcwii, 2015-03-04:
Milan,

     Was this on a Pi B, B+ or a Pi 2?

Lester...

krylm, 2015-03-04:
Pi2

lcwii, 2015-03-04:
Milan,

     Ok, I have done a lot of things that are not required then, what 
version of kernel are you running?

Lester...

krylm, 2015-03-04:
I have latest Raspbian installation from http://www.raspberrypi.org/downloads/

Linux 3.18.5-v7+ #1 SMP PREEMPT Fri Feb 6 23:06:57 CET 2015 armv7l

lcwii, 2015-03-04:
Milan,

     I installed using Berryboot and my linux is showing 3.18.6-v7i-aufs 
for some reason. Notro indicated that he was not familiar with that version.

     Well, it didn't work...

     I think I am going to start over and use Noobs first and try to go 
from there. That way I know there isn't anything else that has been 
loaded onto the system.

Lester...

lcwii, 2015-03-05:
Milan,

Thank you! I have the display working now. Now I just need to calibrate 
the touch screen.

   Your last step 'sudo con2fbmap 1 1', did not seem to work for me.

     So I have it running in graphical mode. I used this from 
http://www.circuitbasics.com/setup-lcd-touchscreen-raspberry-pi/ :

Next we need to configure the /fbturbo video driver/ to output the 
display of the Raspberry Pi to the SPI bus instead of the HDMI bus. At 
the command prompt, enter:

sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf

In this file, find the line that says: Option    “fbdev” “/dev/fb0″ and 
change the fb0 to fb1:

I had screwed up my /boot/config.txt file.

Lester...

//||//||

krylm, 2015-03-05:
Yes, that's where I ended yesterday. With running text mode on PiScreen. 

I'll try to continue with other steps today or tomorrow. If I'll be succcessful I let you know here.

Some resources:
https://github.com/notro/fbtft/wiki/Touchpanel
https://github.com/notro/fbtft_tools/wiki/ads7846_device

notro, 2015-03-05:
@krylm you only need one of fbtft_device or dtoverlay, not both.
They both add the piscreen device. I will guess that fbtft_device is the one that wins by overwriting the device added by dtoverlay.
And if you use dtoverlay, you don't have to enable SPI in raspi-config. The overlay enables it.
Source: https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/boot/dts/piscreen-overlay.dts

Regarding touch, here's a guide I wrote: https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian

krylm, 2015-03-05:
@notro Thanks for your help! Finally I understand it a bit. Default was dtoverlay. I wasn't able to make it work only with fbtft_device so I am using SPI & dtoverlay with 90deg rotation.

I've calibrated Touchpanel, didn't understand it that I need to touch the red crosses for the first time and had swapped axes. But now after proper calibration it's ok. So I don't need missing  /etc/X11/xorg.conf.d/ directory. 

Just one non related question. Not initialized TFT is shining white, after init it's ok in black. But when I shutdown raspberry it's white again. I have to remove power cord to make it black.

lcwii, 2015-03-05:
Ok, how do you incorporate piscreen_overlay.dts ? I tried 
dtoverlay=piscreen, piscreen_overlay but that did not work. I am 
assuming that piscreen_overlay is incorporated into fbtxt_device or am I 
wrong?

lcwii

krylm, 2015-03-05:
I just removed line from /etc/modules
`fbtft_device name=piscreen rotate=90 speed=16000000`
And line from /boot/config.txt
`fbtft_device.debug=3`

Changed line
`dtoverlay=piscreen`
to
`dtoverlay=piscreen,rotate=90`

and rebooted

after everything working, I've added into /boot/cmdline.txt
`fbcon=map:10`
before rootwait

from my description before
https://github.com/notro/fbtft/issues/252#issuecomment-77264581

lcwii, 2015-03-05:
> #252 (comment)

Milan,

     Are you running in console mode? I am trying to run in GUI mode 
because a friend also got the same set up and is not a console person 
and I am trying to help him with his Pi2. Getting the touch calibration 
to work with the GUI has to be different than in console mode.

     One thing I know is different is I have to modify
/usr/share/X11/xorg.conf.d/99-fbturbo.conf
and change
Option          "fbdev" "/dev/fb0"
to
Option          "fbdev" "/dev/fb1"
to get the LCD working in GUI mode.

     I also tried to get both console and GUI mode to work at the same 
time and locked up the Pi2. Had to power down everything and start over 
with a fresh start install.

Lester...

notro, 2015-03-05:
> Not initialized TFT is shining white, after init it's ok in black. But when I shutdown raspberry it's white again. I have to remove power cord to make it black.

The Pi gpios are set as inputs on poweron. The display has the backlight control tied high, so a floating backlight pin turns on backlight.
When the FBTFT driver loads, it sets it's pins as output, initializes the display and fills it with zeros: Black display
When the Pi shuts down, Linux sets the pins back to a floating input, including the reset pin: Display loses initialization which results in a white display.

leoneo, 2015-03-12:
Hello,
I am new on RPi2, I have the SKU: AC-TFT35-V2.1 from dx.com, I have install rasbian with berryboot. 

How do you do to use this TFT ? Can you please explan step by step ?

Thanks you so much

Sorry for my bad english ;-)

lcwii, 2015-03-12:
I would if I could, but I still don't have the touch screen working 
correctly.

Lester...

GuntherGamma, 2015-03-15:
Hi all,
i got this thing working as far, that i can view pictures with fbi on the touchscreen.
Does anyone know how to enable the console on the touchscreen ?

leoneo, 2015-03-16:
I lost a lot of time , I tried to install it several times but every time I have a hdmi black screen with a dash flashing _ ... _   ... _  ... _ ...

Please help

GuntherGamma, 2015-03-16:
It stoped working after i tried to calibrate the touchscreen and now it doesn't want to work again.
If it is working again, i'll write what i've done.

deckardstp, 2015-03-24:
I already spend a few hours on getting this lcd working, but I only have the white screen. According to syslog and startup messages it looks quite good, nonetheless it doesn't work. If anyone has a working image I would really appreciate any help!

lcwii, 2015-03-25:
Pi 2?

Lester...

deckardstp, 2015-03-25:
Jep Pi2. I currently install it on a Pi1. Hope this works out.

deckardstp, 2015-03-25:
Even on Pi1 same result. I tried also another Kernel, since I read it could be bug in the SPI communication in one of the last builds. I tried this yesterday on my Pi2 and produced kernel panic. I haven't worked on the Pi1 either. Last shot is an image from dx.com. As soon as I know more I post it here.

deckardstp, 2015-03-25:
Same result. dx.com Image was from 2014/10/31. I give up at that point. Spend to much time on this piece of crap.

lcwii, 2015-03-25:
For console or desktop?

Lester...

deckardstp, 2015-03-26:
The image? It didn't run on both Pi versions.

overheder, 2015-03-29:
Just chiming in here unfortunately. I have also spent quite a lot of time trying to get the display to work - it's white regardless what I do :-( Same problem on my RPI 1 incidentally.

Anyway, I have one question: How connected is the screen and the touch? I see pin 11 high all the time, even if I have a pin placed constantly on the resistive pad.... come to think of it, this may be expected behaviour, but I am clutching for straws.

These are the relevant files from my RPI 2 - I think! let me know if you need more

# /etc/modprobe.d/raspi-blacklist.conf
 
 
#blacklist spi and i2c by default (many users don't need them)
 
#blacklist spi-bcm2708
#blacklist i2c-bcm2708
 
# pi@raspberrypi ~ $ ls /proc/device-tree
#address-cells  clocks      interrupt-parent  name           sound
aliases         compatible  memory            __overrides__  __symbols__
axi             cpus        memreserve        #size-cells    system
chosen          display     model             soc            timer
pi@raspberrypi ~ $
 
 
# cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fbtft_device.debug=3 rootwait config.txt
 
#for more options see http://elinux.org/RPi_config.txt
#dtparam=spi=on
dtparam=spi=on
dtparam=i2c_arm=on
dtoverlay=piscreen,fbtct_device.debug=3
 
# no 99-fbturbo.conf file
 
 
# /etc/modules: kernel modules to load at boot time.
#
#This file contains the names of kernel modules that should be loaded
#at boot time, one per line. Lines beginning with "#" are ignored.
#Parameters can be specified after the module name.
 
#snd-bcm2835
#spi-bcm2708
#i2c-BCM2708
#w1-gpio
#w1-therm
 
 
#tft
 
 
#fbtft_device name=piscreen rotate=90 speed=16000000
 
#flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0x36,0x28,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x0,0x0,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x11,-1,0X29,-3
 
#fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 gpios=reset:25,dc:24,led:22
#ads7846_device model=7843 gpio_pendown=17 verbose=3 x_plate_ohms=100 pressure_max=255  swap_xy=1
 
 
# redacted DMESG:
 
[    3.045390] usb 1-1.5: Manufacturer: Microsoft
[    3.069450] input: Microsoft Wired Keyboard 600 as /devices/platform/bcm2708_usb/usb1/1-1/1-1.5/1-1.5:1.0/0003:045E:0750.0002/input/input1
[    3.075540] hid-generic 0003:045E:0750.0002: input,hidraw1: USB HID v1.11 Keyboard [Microsoft Wired Keyboard 600] on usb-bcm2708_usb-1.5/input0
[    3.106538] input: Microsoft Wired Keyboard 600 as /devices/platform/bcm2708_usb/usb1/1-1/1-1.5/1-1.5:1.1/0003:045E:0750.0003/input/input2
[    3.113946] hid-generic 0003:045E:0750.0003: input,hidraw2: USB HID v1.11 Device [Microsoft Wired Keyboard 600] on usb-bcm2708_usb-1.5/input1
[    3.203769] udevd[175]: starting version 175
[    3.725183] bcm2708_spi 3f204000.spi: DMA channel 2 at address 0xf3007200 with irq 77
[    3.735038] bcm2708_spi 3f204000.spi: DMA channel 4 at address 0xf3007400 with irq 20
[    3.783186] ads7846 spi0.1: touchscreen, irq 465
[    3.798501] input: ADS7846 Touchscreen as /devices/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input3
[    3.913238] fbtft_of_value: regwidth = 16
[    3.929328] fbtft_of_value: buswidth = 8
[    3.975809] fbtft_of_value: debug = 0
[    3.984895] fbtft_of_value: rotate = 270
[    3.992610] fbtft_of_value: fps = 30
[    4.367633] graphics fb1: fb_ili9486 frame buffer, 480x320, 300 KiB video memory, 4 KiB DMA buffer memory, fps=33, spi0.0 at 32 MHz
[    4.374195] bcm2708_spi 3f204000.spi: SPI Controller at 0x3f204000 (irq 80)
[    4.380601] bcm2708_spi 3f204000.spi: SPI Controller running in dma mode
[    4.387222] bcm2708_i2c_init_pinmode(1,2)
[    4.394521] bcm2708_i2c_init_pinmode(1,3)
[    4.402708] bcm2708_i2c 3f804000.i2c: BSC1 Controller at 0x3f804000 (irq 79) (baudrate 100000)
[    5.441402] random: nonblocking pool is initialized
[    5.964066] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    6.222879] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    8.985455] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   10.382315] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   14.380080] Adding 102396k swap on /var/swap.  Priority:-1 extents:1 across:102396k SSFS
[  223.214745] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1

As you can see from the dmesg, the display is recognized, and set as fb1
sudo fbi -noverbose -T 1 -a -d /dev/fb1 image-test.gif
yields that fb1 is write protected :-(

overheder, 2015-03-30:
I forgot to mention, that there is no input from input3, when I test that.

exzile, 2015-04-12:
I am working on installing this into my pi also. I am in the step of downloading the image and I am using a pi b. Do you guys know where I can find the pin layout? I want to be able to free up pins, especially the communication pins 'I2C, SPI'.

lcwii, 2015-04-14:
Here is the schematic I got from the seller I got mine from. Pi 
connector on the left.

Lester...

Texy, 2015-04-14:
No picture or attachment ?

lcwii, 2015-04-15:
I guess github tosses attachments...

Lester...

lcwii, 2015-04-15:
Ok, attachments don't work, clipboard paste doesn't seem to work either. 
Try this link http://imgur.com/iSHliDt

Lester...

ShinyPC, 2015-04-17:
I have this touch screen installed on a pi 2 with latest kernel 3.18.11-v7. When I used Notro's firmware for kernel 3.18.10 everything worked correctly, including the touch screen. When I upgraded to latest raspbian firmware, the touch screen no longer works. Dmesg error ads7846: probe of spi0.1 failed with error 22. I am using device tree with piscreen and ads7846 overlays. How do I fix the ads7846 error to get touch working again?

ShinyPC, 2015-04-17:
For those interested with latest raspbian on pi 2 B, the install for only the lcd display using device tree. This orientation puts the power cord on top.

#LCD Screen
#url https://github.com/notro/fbtft/issues/252
#if you use dtoverlay, you don't have to enable SPI in raspi-config. The overlay enables it.

#install lcd and touch screen
	sudo nano /boot/config.txt
	add
		dtoverlay=piscreen,speed=32000000,rotate=90,debug=3 # fps=,debug
		dtoverlay=ads7846,speed=32000000,rotate=90,swapxy,xohms=60	#xohms=60to100

#move display to lcd
	sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf
		change option “fbdev” /dev/fb0 to /dev/fb1

#move console to lcd
	sudo /boot/cmdline.txt
	add to end of line
		fbcon=map:10 fbcon=rotate:2 fbcon=font:VGA8x8

cnc4less, 2015-05-02:
ShinyPC, I followed what u wrote and got it working but without touch screen any ideas why is doing that ?

futuneral, 2015-05-19:
Hi. I'm pretty new to Linux/RPi, but with your help guys I was able to make TFT and touchscreen work on RPi2. However, one weird wrinkle. If I skip the step of installing FBTFT kernel with SPI DMA, everything is perfect (albeit slow). But if I do use it, I get some sort of corruption in the frame buffer - some colors are wrong, random white dots on the screen etc. What could be the cause for this? Any parameters to adjust? If I do raspi-update I go back to normal colors.
Here is my full sequence in case it helps someone:
```
1. sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update

2. sudo nano /boot/config.txt
add
dtoverlay=piscreen,speed=32000000,rotate=90
dtparam=i2c_arm=on

3. sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf
change option “fbdev” /dev/fb0 to /dev/fb1

4. sudo nano /boot/cmdline.txt
add to end of line
fbcon=map:10 fbcon=rotate:2 fbcon=font:VGA8x8

5. cd /tmp
wget http://tronnes.org/downloads/xinput-calibrator_0.7.5-1_armhf.deb
sudo dpkg -i -B xinput-calibrator_0.7.5-1_armhf.deb
rm xinput-calibrator_0.7.5-1_armhf.deb

6. sudo wget -O /etc/X11/Xsession.d/xinput_calibrator_pointercal https://raw.github.com/tias/xinput_calibrator/master/scripts/xinput_calibrator_pointercal.sh
echo "sudo /bin/sh /etc/X11/Xsession.d/xinput_calibrator_pointercal" | sudo tee -a /etc/xdg/lxsession/LXDE-pi/autostart

7. cd /etc/X11
sudo mkdir -p xorg.conf.d
sudo nano /etc/X11/xorg.conf.d/99-ads7846-cal.conf

Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "SwapAxes"      "1"
EndSection
```

mwilliams03, 2015-05-19:
Reduce the speed below to 24000000 or 16000000. See if that makes a difference

futuneral, 2015-05-19:
mwilliams03, thanks! 24000000 made slight improvements, but 16000000 fixed it completely.

overheder, 2015-05-19:
Updates from here

I finally got myfbturbo file to edit - by installing fbturbo, duh!
sudo apt-get update && sudo apt-get install xserver-xorg-video-fbturbo

@futuneral thanks! I updated to match your settings, and 
sudo fbi -noverbose -T 1 -a -d /dev/fb1 image-test.gif
no longer yields that fb1 is write protected :-)

Do you have dtparam=spi=on in tour cmdline.txt?

When I try startx I get "module g2d_23 not found", so FBTURBO is installed - or I did not notice it before :-/, anyway it should be benign

I still have a display that's just white, but I'll keep hacking away - it's probably due to having an old image that's been upgraded over the years

futuneral, 2015-05-20:
overheder
spi=on doesn't seem to matter with overlays. I had it both on and off and it worked the same.
I had to re-image my sdcard multiple times before I got this working. Sometime something just screws things up and it gets stuck at white screen even if you try to revert.
So I'd suggest to get the newest version of raspbian, then do step #2 and step #3. This actually seems to be the minimum that gets display working. After this you can proceed to other steps.

chickey, 2015-05-25:
Has anyone been able to get emulationstation working on this tft display.  I can get the console working well as well as x windows really well but by default emulationstsation won't display on the tft.  I have been able to get it to display using fbcp but it's rather slow and not playable :-(

FRALEU, 2015-06-17:
Hi, 
thanks everybody for all your comments.
With this touchscreen, by following the tuto from @notro https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian and the procedure described by @futuneral posted in this forum one month ago, my TFT35" Touch Shield (SKU: AC-TFT35V-2.1) works fine (calibration ok, touch control properly handled in X windows).
But now, I have an issue with kodi: kodi is properly displayed on my TFT screen by using fbcp but the touchcontrol doesn't work (I have also tried with kodi 're-touch' skin designed to be used on touch screen devices)
I'm stuck: I don't know what's wrong (kodi, xinput, ...) with my config.
Any suggestion ?
Thanks;

plivive, 2015-08-02:
Hi,
I followed the steps described  by "futuneral" and it worked hover my touch interface is still in portrait mode
I did the screen calibration but does not help.
tried with different screen rotation. 90°,180°,270°
any suggestions on how to rotate the touch interface with the screen.

have a Raspberry pi 2 B

notro, 2015-10-18:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2015-10-18 by notro

lfelipee, 2015-11-16:
Hello friends,
I would like to help those who were able to run the display and the touch sensor, but as I was, are annoyed to have to redo the calibration at each startup:

1- I disabled the calibration at start like this:
sudo nano / etc / xdg / lxsession / LXDE-pi / autostart
I inserted comment in the last line, but you can delete, whatever. The last killed string / is commented: sudo /bin/sh /etc/x11/Xsession.d/xinput_calibrator_pointcal

2- Run:
cd /etc/X11
sudo mkdir -p xorg.conf.d
sudo nano /etc/X11/xorg.conf.d/99-ads7846-cal.conf

Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "SwapAxes"      "0"
EndSection

or simply change the option to "0"

3- Restart the Raspberry Pi 2

4- Run in terminal:
bin/sh -c "xinput_calibrator; cat"

Do the touch screen calibration and follow the instructions that will guide copying the calibration result and save the file:
sudo nano /etc/x11/xorg.conf.d/99-calibration.conf

5- Ready! Reinice to test!

Grades:
I'm using the Raspbian Wheezy version, Release date: 05/05/2015

I followed the information from our friend futuneral shortly after instação system.

I did not realize the system upgrade (step 1) as was already the latest version of Raspbian, funciou well, so I left anyway!

Follows playback futuneral Tutorial, by changing paso 7 since it would be necessary to change later (step 4 of the above tutorial):

1. sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update

2. sudo nano /boot/config.txt
add
dtoverlay=piscreen,speed=32000000,rotate=90
dtparam=i2c_arm=on

3. sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf
change option “fbdev” /dev/fb0 to /dev/fb1

4. sudo nano /boot/cmdline.txt
add to end of line
fbcon=map:10 fbcon=rotate:2 fbcon=font:VGA8x8

5. cd /tmp
wget http://tronnes.org/downloads/xinput-calibrator_0.7.5-1_armhf.deb
sudo dpkg -i -B xinput-calibrator_0.7.5-1_armhf.deb
rm xinput-calibrator_0.7.5-1_armhf.deb

6. sudo wget -O /etc/X11/Xsession.d/xinput_calibrator_pointercal https://raw.github.com/tias/xinput_calibrator/master/scripts/xinput_calibrator_pointercal.sh
echo "sudo /bin/sh /etc/X11/Xsession.d/xinput_calibrator_pointercal" | sudo tee -a /etc/xdg/lxsession/LXDE-pi/autostart

7. cd /etc/X11
sudo mkdir -p xorg.conf.d
sudo nano /etc/X11/xorg.conf.d/99-ads7846-cal.conf

Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "SwapAxes"      "0"
EndSection

alexeyzimarev, 2016-06-29:
I have got the same screen and I have RPi3. I followed the [instructions](http://ozzmaker.com/piscreen-driver-install-instructions-2/) for setting up pi-screen and it just worked.

kjartani, 2017-05-01:
The same experience as [alexeyzimarev](https://github.com/alexeyzimarev). I have got the same screen and but I have RPi. I followed the instructions for setting up pi-screen and it just worked.