notro/fbtft #253: Support for WaveShare 4"

rcdilorenzo, 2015-02-27:
Is there already support for the WaveShare 4" LCD?

notro, 2015-02-28:
Is this the same display?
http://raspberrypi.stackexchange.com/questions/27714/how-to-install-waveshare-spotpear-4-inch-lcd-in-raspberrypi-2

rcdilorenzo, 2015-02-28:
Yes, but I tried those exact instructions and couldn't get it to work. I'm actually planning to return mine and get an HDMI display for my RPi.

arthur-esob, 2015-03-29:
Hello,
I managed to have the waveshare 4in display on my Pi2 but NOT the touchscreen. I've tried a lot of stuff and look around the internet without any sucess.
the line i'm using in /etc/modules is the following :
ads7846_device model=7846 cs=1 gpio_pendown=17 speed=1000000 keep_vref_on=1 swap_xy=1 pressure_max=255 x_plate_ohms=60 x_min=200 x_max=3900 y_min=200 y_max=3900
which is, i assume, the line in charge of driving the touchscreen but doesn't work.
first, how can i check if the pi gets any signal at all from the touchscreen ?
pierre

arthur-esob, 2015-03-29:
Hello,
I managed to have the waveshare 4in display on my Pi2 but NOT the touchscreen. I've tried a lot of stuff and look around the internet without any sucess.
the line i'm using in /etc/modules is the following :
ads7846_device model=7846 cs=1 gpio_pendown=17 speed=1000000 keep_vref_on=1 swap_xy=1 pressure_max=255 x_plate_ohms=60 x_min=200 x_max=3900 y_min=200 y_max=3900
which is, i assume, the line in charge of driving the touchscreen but doesn't work.

closed 2015-06-18 by notro