notro/fbtft #254: Support for tft 3.6" ili9325

ont, 2015-03-03:
I recently bought 3.6 " lcd screen for my raspberry b+ but I can't figure out how to use it with one of your driver. 
[image]
[image]

I successfully install custom image with precompiled kernel from [win32_screem.rar](https://drive.google.com/uc?export=download&confirm=FO_M&id=0BxpI8ASCfaeBbUF6SEJzSm9zU2M). Custom image is raspbian with kernel:
```
root@raspberrypi:/# uname -a
Linux raspberrypi 3.6.11 #5 PREEMPT Sun Nov 9 19:00:34 CST 2014 armv6l GNU/Linux
```
And some magical driver which is compiled into the kernel (no **.ko** files on device).
By searching in **/sys** dir I found this references
```
root@raspberrypi:/# find -name '*932*'
./sys/bus/platform/devices/ili9325.0
./sys/bus/platform/drivers/ili9325
./sys/bus/platform/drivers/ili9325/ili9325.0
./sys/devices/platform/ili9325.0

root@raspberrypi:/sys/bus/platform/devices/ili9325.0/graphics/fb0# ls -la
total 0
drwxr-xr-x 3 root root    0 Nov  9 08:37 .
drwxr-xr-x 3 root root    0 Nov  9 08:37 ..
-rw-r--r-- 1 root root 4096 Nov  9 10:20 bits_per_pixel
-rw-r--r-- 1 root root 4096 Nov  9 10:20 blank
-rw-r--r-- 1 root root 4096 Nov  9 10:20 console
-rw-r--r-- 1 root root 4096 Nov  9 10:20 cursor
-r--r--r-- 1 root root 4096 Nov  9 10:20 dev
lrwxrwxrwx 1 root root    0 Nov  9 10:20 device -> ../../../ili9325.0
-rw-r--r-- 1 root root 4096 Nov  9 10:20 mode
-rw-r--r-- 1 root root 4096 Nov  9 10:20 modes
-r--r--r-- 1 root root 4096 Nov  9 10:20 name
-rw-r--r-- 1 root root 4096 Nov  9 10:20 pan
drwxr-xr-x 2 root root    0 Nov  9 08:37 power
-rw-r--r-- 1 root root 4096 Nov  9 10:20 rotate
-rw-r--r-- 1 root root 4096 Nov  9 10:20 state
-r--r--r-- 1 root root 4096 Nov  9 10:20 stride
lrwxrwxrwx 1 root root    0 Nov  9 10:20 subsystem -> ../../../../../class/graphics
-rw-r--r-- 1 root root 4096 Nov  9 10:20 uevent
-rw-r--r-- 1 root root 4096 Nov  9 10:20 virtual_size
```
Some info from this frame buffer:
```
name --> ILI9325
bits_per_pixel -->  16
modes --> U:400x240p-0
virtual_size --> 400,240
```

```
root@raspberrypi:/# fbset -i -fb /dev/fb0 

mode "400x240"
    geometry 400 240 400 240 16
    timings 0 0 0 0 0 0 0
    rgba 5/11,6/5,5/0,0/0
endmode

Frame buffer device information:
    Name        : ILI9325
    Address     : 0xdc804000
    Size        : 192512
    Type        : PACKED PIXELS
    Visual      : TRUECOLOR
    XPanStep    : 0
    YPanStep    : 0
    YWrapStep   : 0
    LineLength  : 800
    Accelerator : No
```

The problem is that I have no success with your fb_ili9325 driver. It results with errors in dmesg and white screen. I try this commands:
```
modprobe fbtft_device name=ili9325
modprobe fbtft_device name=fb_ili9325
```

I figure out wich pins is used by this lcd (# is used by tft screen):
```
   3.3v  1 # . 2
         3 . . 4
         5 . # 6   ground
 GPCLK   7 # . 8
         9 . . 10
 GPIO0  11 # # 12  PCM_CLK / PCM0
        13 . . 14
 GPIO3  15 # # 16  GPIO4
        17 . # 18  GPIO5
  MOSI  19 # . 20
  MISO  21 # # 22  GPIO6
  SCLK  23 # # 24  CS0
        25 . # 26  CS1
```
Absence of one of this pins results (in most cases) in white flashing screen. 
How to detect purposes of each of this pins?

notro, 2015-03-03:
They obviously use the fb_ili9325 driver, but with a custom init sequence.
Add this to /boot/cmdline.txt and ```dmesg``` will give you more info:
```
fbtft_device.debug=7
```

ont, 2015-03-04:
I read https://github.com/notro/fbtft/blob/master/fbtft_device.c more carefully. As I understand a correct command for my display should be one of this variants:
```
modprobe fbtft_device name=hy28b debug=7
modprobe fbtft_device name=itdb28 debug=7
modprobe fbtft_device name=itdb28_spi debug=7
```
For each of this commands module is successfully loaded (no error "ERROR: could not insert 'fbtft_device': Invalid argument") but lcd screen is still white.

Dmesg contains no errors for *hy28b*. For *itdb28* there are
```
[  150.773288] fb_ili9325 fb_ili9325.0: fbtft_verify_gpios()
[  150.773312] fb_ili9325 fb_ili9325.0: Missing info about 'dc' gpio. Aborting.
[  150.773415] fb_ili9325: probe of fb_ili9325.0 failed with error -22
```

For *itdb28_spi* there is message about led pin:
```
[   77.014483] fb_ili9325 spi0.0: fbtft_register_backlight()
[   77.014511] fb_ili9325 spi0.0: fbtft_register_backlight(): led pin not set, exiting.
[   77.015526] graphics fb1: fb_ili9325 frame buffer, 240x320, 150 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 32 MHz
```

For each command I do full device reboot. SPI is enabled in raspi-config. And in /boot/config.txt line is present:
```
dtparam=spi=on
```

notro, 2015-03-04:
You have to enable debug output on the custom image to see the setup they have used.
I said level 7, but that isn't good if the console is set to use the tft, so use 3 instead:
fbtft_device.debug=3

If that doesn't give you anything, they might be using Device Tree. If so try this (it should work but it's a long time since I tried this argument to fbtft):
fbtft.debug=3

ont, 2015-03-05:
Still no luck...  Nothing appeared in dmesg output. My current **cmdline.txt**:
```
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbtft_device.debug=3 bcm2708_fb.debug=3 bcm2708.debug=3 fbtft.debug=3
```
Output of **/proc/cmdline** is more intresting and contains references to **bcm2708_fb**:
```
dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0x10 bcm2708.serial=0xc1db745d smsc95xx.macaddr=B8:27:EB:DB:74:5D bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbtft_device.debug=3 bcm2708_fb.debug=3 bcm2708.debug=3 fbtft.debug=3
```
Full list of kernel modules (lsmod):
```
Module                  Size  Used by
snd_bcm2835            16304  0 
snd_pcm                77560  1 snd_bcm2835
snd_page_alloc          5145  1 snd_pcm
snd_seq                53329  0 
snd_seq_device          6438  1 snd_seq
snd_timer              19998  2 snd_pcm,snd_seq
snd                    58503  5 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device
leds_gpio               2235  0 
led_class               3562  1 leds_gpio
```

And some blacklisted modules (/etc/modprobe.d/raspi-blacklist.conf):
```
blacklist spi-bcm2708
blacklist i2c-bcm2708
blacklist snd-soc-pcm512x
blacklist snd-soc-wm8804
```

Grepping for bcm2708_fb gives this output:
```
./sys/devices/platform/bcm2708_fb/modalias
./sys/devices/platform/bcm2708_fb/uevent
./boot/cmdline.txt  -- my bcm2708_fb.debug=3
./boot/start.elf
./boot/start_cd.elf
./boot/start_x.elf
./lib/modules/3.12.28+/modules.builtin
./var/log/dmesg
./var/log/Xorg.0.log.old
./var/log/Xorg.0.log
./var/log/syslog
./var/log/kern.log
./var/log/messages
./var/log/dmesg.0
```

```
root@raspberrypi:/sys/devices/platform/bcm2708_fb/subsystem/drivers# ls
alarmtimer    bcm2708_sdhci  bcm2835_AUD2  bcm2835_AUD6     ili9325
arm-pmu       bcm2708_vcio   bcm2835_AUD3  bcm2835_AUD7     leds-gpio
bcm2708_dma   bcm2835_AUD0   bcm2835_AUD4  bcm2835_thermal
bcm2708_gpio  bcm2835_AUD1   bcm2835_AUD5  dwc_otg
```

```
root@raspberrypi:/lib/modules/3.12.28+# cat modules.builtin
....
kernel/drivers/video/bcm2708_fb.ko
....
```

notro, 2015-03-05:
I suggest you contact the seller to get the correct setup.

keerttttt, 2015-03-23:
I want to buy the same screen. What is the framerate, is it okay?

ont, 2015-03-24:
Frame rate is good. Visually there is no flickering when I run glxheads mesa-utils test program. It is hard to test with real videos because decoding of usual video codecs eats all cpu of raspberry pi and frame rate is near zero.

keerttttt, 2015-03-24:
Why does it eat 100% CPU? It has gpu video decoding right?

ont, 2015-03-25:
I don't research this question yet. Fast googling reveal this link https://wiki.matthiasbock.net/index.php/Hardware-accelerated_video_playback_on_the_Raspberry_Pi   I simply try to open *.mp4 file with vlc player. Seems that vlc doesn't use gpu out of the box.

aparaske, 2015-03-28:
Did you have any luck using this with NRF24L01 (SPI device). I can see that there's traces coming from both CS0 and CS1. Does that mean that the display uses both? The display would be useless to me if I can't use another SPI device.

ont, 2015-04-02:
Yes, this display uses both CS0 and CS1 pins. Also I have problems with control of some non-SPI GPIO pins when I use custom raspbian image. Possibly you need to build SPI interface from free GPIO pins manually.

aparaske, 2015-04-02:
Do you know if it has been done before? I've looked and haven't found anything.

keerttttt, 2015-04-28:
I got this datasheet for this device from the manufacturer, with the wiring that is used. With this information (by the way, i think it is a ILI9327 device), is it possible to get this thing working with fbtft?

See the pdf:

ranked.nl/pi-36-raspberry.pdf

ont, 2015-04-28:
@keerttttt very intresting! But this schema lack information about pins on J2 (44-9327-08). Are you sure that this device is ILI9327? Here is datasheet for it http://www.datasheet-pdf.com/datasheet/ILITECHNOLOGY/837831/ILI9327.pdf.html  (www.mcufriend.com is broken at this moment). Also, if you intrested, I can provide timeseries data recorded by digital analyzer (http://www.aliexpress.com/item/1sets-New-Arrival-USB-Logic-Analyze-24M-8CH-MCU-ARM-FPGA-DSP-debug-tool/1960280884.html). Yeah it has only 8 channels... So I intend to record data in two stages.

ont, 2015-04-28:
Oh, my mistake... I missed small rst, wr, rs, d1, d2, d3, ... labels at PI-26 scheme block.

nopnop2002, 2015-05-16:
http://www.ranked.nl/pi-36-raspberry.pdf

This page can't be indicated.
I'd like to see a data seat, too.

ont, 2015-05-17:
@nopnop2002 if i understand correctly what is drawn in pdf:
```
              3.3v   1 # . 2
                     3 . . 4
                     5 . # 6   ground
              [cs]   7 # . 8
                     9 . . 10
              [wr]  11 # # 12  [rs]
                    13 . . 14
              [d1]  15 # # 16  [d2]
                    17 . # 18  [d3]
              [d4]  19 # . 20
              [d6]  21 # # 22  [d5]
              [d7]  23 # # 24  [d8]
                    25 . # 26  [rst]
```
Original pdf: https://yadi.sk/i/fsk5Cddsggfh9
Strange order of d5 and d6...  And seems that [rs] is actually command/data switch.

V1R11, 2015-05-20:
I also have one of this screen and found a comment on a chinese forum that this will work on this screen  http://imagewriteriij.blogspot.jp/2014/01/raspberry-pi-9-lcd-1.html since my knowledge is limited on this I hope you guys could get something out that I will get benefit of later.

keerttttt, 2015-05-22:
Okay, i've used the sheet in the pdf: https://yadi.sk/i/fsk5Cddsggfh9
And it works! so i can confirm that this is the correct pinout!

But, ofcourse only with the specified kernel or image... So, Notro, would it be possible to make it work with fbtft? It should be possible right? It's parallel driven...

I'm not good enough with low level programming and C to do such a thing... :(

notro, 2015-05-23:
> So, Notro, would it be possible to make it work with fbtft? It should be possible right? It's parallel driven...

If it uses a WR signal to latch in the data, I guess it should work.

keerttttt, 2015-05-24:
I'm trying this standard kernel driver for now, but i can't seem to find any way to flip the screen or rotate it 180 degrees. Is there any way to do so in Raspbian? (I know it's bit of topic, but still)...

Xrandr is not working:
Rotation: [i]xrandr: output default cannot use rotation "inverted" reflection "none"[/i]

Or orientation: [i]X Error of failed request:  BadMatch (invalid parameter attributes)
  Major opcode of failed request:  149 (RANDR)
  Minor opcode of failed request:  2 (RRSetScreenConfig)
  Serial number of failed request:  14
  Current serial number in output stream:  14
[/i]

notro, 2015-05-24:
I have never tried randr, but many of the fbtft drivers support hardware rotation. Which driver are you using and are you using a DT overlay or fbtft_device?

MDM63, 2015-05-29:
Hello. I have the same display. keerttttt could you be so kind and tell us how exactly you got your display to work with the standard kernel image? What fbtft_device name you used and/or how did you connect the display? This would be very much appreciated.

keerttttt, 2015-06-01:
I did not get it to work with the standard kernel. Better yet, i didn't even try it. I'm using the kernel from the chinese image, but i used my own sd card raspbian installation for the rest. So i changed only the kernel.

sfemping, 2015-06-13:
I also have a couple of these displays and am stuck with the precompiled kernel and no way to make support for Pi2 work, or any other interesting developments.

notro, I know you're trying to get away from end device support, but please do you have any more pointers as to how we can identify how mcufriend (website now gone, no support) are initialising what appears to be a ili9325 driver?

I've had a go with the fbtft_device.debug=7 and fbtft_device.debug=3 but got nothing that I could find that was relevant in dmesg output.

Also tried fbtft.debug=3  (I'm assuming output should appear in dmesg log?...)

There must be quite a few people with these displays... I wonder if this is why they're going cheap! ;-)

Thanks again for all your efforts and development.

notro, 2015-06-14:
If debug=7 and the like doesn't work, they must have disabled it.
I see 2 ways to get to the init sequence:
1. Attach a SPI protocol analyzer to the SPI bus and capture the output
2. Disassemble the driver
```objdump -Dr fb_ili9325.ko```
Look at the init_display function to get the register values.
I suggest doing this on a regular fb_ili9325 first so you can compare with the source.
I have never done this myself.

ont, 2015-06-14:
My small addition:
1. It is really great idea! And I hardly try to do that. But at current point I have no luck with my 8 channel 24 Mhz logic analyzer. After some research it is clear that this display transfer data at greater frequencies. Now I am waiting for delivery of 16 channel 200 Mhz logic analyzer. It will arrive within a month.
2. The most terrible thing about this custom driver is that it is compiled into custom kernel image. I could not find any *.ko files related to this driver. And linux kernel image is very huge.

notro, 2015-06-14:
The win32_screem.rar kernel is not using a fbtft driver:
```
$ binwalk -Me kernel.img

Scan Time:     2015-06-14 17:02:20
Target File:   /home/pi/kernel.img
MD5 Checksum:  1daa4dddff863c8bd7ce8e0bd3720b5f
Signatures:    332

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
50604         0xC5AC          gzip compressed data, maximum compression, from Unix, NULL date (1970-01-01 00:00:00)

Scan Time:     2015-06-14 17:02:45
Target File:   /home/pi/_kernel.img.extracted/C5AC
MD5 Checksum:  3c9a394321437299f02610b71e5c75cb
Signatures:    332

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
2025993       0x1EEA09        Certificate in DER format (x509 v3), header length: 4, sequence length: 1284
2026113       0x1EEA81        Certificate in DER format (x509 v3), header length: 4, sequence length: 1288
2027193       0x1EEEB9        Certificate in DER format (x509 v3), header length: 4, sequence length: 1284
2499693       0x26246D        Certificate in DER format (x509 v3), header length: 4, sequence length: 4355
3776636       0x39A07C        Linux kernel version "3.6.11 (lgh@lgh-ThinkPad-T420) (gcc version 4.7.1 20120402 (prerelease) (crosstool-NG 1.15.2) ) #5 PREEMPT Sun Nov 9 19:00:34 CS"
3784712       0x39C008        gzip compressed data, maximum compression, from Unix, NULL date (1970-01-01 00:00:00)
4586124       0x45FA8C        Unix path: /proc/sys/kernel/hung_task_timeout_secs" disables this message.
4678783       0x47647F        Unix path: /misc/vc04_services/interface/vchiq_arm/vchiq_core.c
4687723       0x47876B        Unix path: /misc/vc04_services/interface/vchiq_arm/vchiq_arm.c
4690251       0x47914B        Unix path: /misc/vc04_services/interface/vchiq_arm/vchiq_2835_arm.c
4690607       0x4792AF        Unix path: /misc/vc04_services/interface/vchiq_arm/vchiq_shim.c
4690691       0x479303        Unix path: /misc/vc04_services/interface/vchiq_arm/vchiq_util.c
4745015       0x486737        Unix path: /S70/S75/505V/F505/F707/F717/P8
4759705       0x48A099        Copyright string: "Copyright(c) Pierre Ossman"
4775786       0x48DF6A        Unix path: /char/broadcom/vc_cma/vc_cma.c
5275648       0x508000        CRC32 polynomial table, little endian

Scan Time:     2015-06-14 17:04:01
Target File:   /home/pi/_kernel.img.extracted/_C5AC.extracted/39C008
MD5 Checksum:  6d756fd2dde0e49ae137abb6a902c4f4
Signatures:    332

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
26080         0x65E0          MPFS filesystem, Microchop, version 61.121, 17162 file entries
26098         0x65F2          MPFS filesystem, Microchop, version 95.77, 21839 file entries
69785         0x11099         MPFS filesystem, Microchop, version 61.121, 17162 file entries
69800         0x110A8         MPFS filesystem, Microchop, version 95.80, 21327 file entries
69825         0x110C1         MPFS filesystem, Microchop, version 95.88, 21569 file entries

$ strings _kernel.img.extracted/C5AC | grep 9325
ili9325_probe
ili9325_video_alloc
ili9325_pages_alloc
ili9325_init
li9325_
li9325
li9325_
li9325_
li9325_i
li9325_c
li9325_
li9325_w
li9325_
li9325_
li9325_
li9325_v{
ili9325_
dbcm2708_ili9325
li9325
li9325_
li9325_d
ili9325
%s: unable to kzalloc for ili9325
%s: unable to ili9325_video_alloc
%s: unable to kmalloc for ili9325_page
%s: unable to kmalloc for ili9325_page oldbuffer
%s: unable to ili9325_pages_init
                                                                                                                                                          ILI9325
```
fb_ili9325 does not have these strings.

sfemping, 2015-06-14:
Thanks notro for taking a look into it, appreciate your expertise!

What a shame, seems like they built support, locked it away and then disappeared off the face of the planet...

Nice little 3.6" displays in themselves, but bit of a dead-end if there's no way to get it working beyond their one precompiled support. :-(

notro, 2015-06-14:
You're welcome. It was an interesting challenge.

I forgot to google those strings. This seems to be the driver they used as their base: https://searchcode.com/codesearch/view/64781754/
But it doesn't help with the init sequence.
I guess @ont will solve this when he gets his analyzer.

sfemping, 2015-06-14:
Ah, good call with the google-fu.

"The Solomon Systech ILI9325 chip drive TFT screen up to 240x320."

From the datasheets it seems more likely that this display is driven from a ILI9327, as it as 400x240 I think. Look forward to @ont getting his analyzer too!

"Hello World!" seems like an appropriate starting message if we ever get these working!  ;-)

webmonger, 2015-06-14:
I've been watching this discussion with great interest (though little understanding) because I also own one of these displays too, with this new info about the ILI9327 I thought I would do a little googling. 

I found this github repo that's a fork of another driver but it references mcufriend quite a lot and the ILI9327 would it be possible to get the init sequence from this code?

https://github.com/dgolda/UTFT

I know the drivers seem to focus on Arduino but there might be something of use in there?

Hope it helps

StoneAgeSkillz, 2015-06-17:
With my programing skills of a troglodyte, which are at same level as my english, i rely on you (and google on translation help). I hope someone cracks the code. But until then, could someone please post pinout for this display? Like: RPi pin -> LCD pin.  I'm using RPi B+, and as i understood, connection does not fit A or B version. But why, if B+ only has more pins, but at same place? Thank you in advance.

hermanhermitage, 2015-06-18:
Possibly this one: http://www.52pi.net/forum.php?mod=viewthread&tid=36
See static void __init ili9325_setup(struct ili9325 *item)

notro, 2015-06-18:
> Possibly this one: http://www.52pi.net/forum.php?mod=viewthread&tid=36

That one uses an 8-bit parallel bus, the OP has a SPI bus display.
The ili9325 controller supports both parallel and SPI, so if the display panels are the same, the init sequence  is the same, unless some of them are interface registers.

ont, 2015-06-18:
> Possibly this one: http://www.52pi.net/forum.php?mod=viewthread&tid=36

@hermanhermitage thanks for the info!

If I understand correctly, this patch use this pinout (I use article http://www.pieter-jan.com/node/15 for information about numbering of pins):
```
#define LCD_NCS 4                       1 . . 2
#define LCD_NWR 17                      3 . # 4
//nrd is tied to 3.3V in hardware.      5 . . 6
#define LCD_D10 22                      7 # # 8
#define LCD_D13 10                      9 # # 10
#define LCD_D15 9                      11 # . 12
#define LCD_D16 11                     13 . . 14
                                       15 . . 16
#define LCD_RS 18                      17 # # 18
#define LCD_D11 23                     19 . . 20
#define LCD_D12 24                     21 . # 22
#define LCD_D14 25                     23 # # 24
#define LCD_D17 8                      25 # . 26
#define LCD_NRST 7
```
Which is different from my variant (detected experimentally):
```
   3.3v  1 # . 2
         3 . . 4
         5 . # 6   ground
 GPCLK   7 # . 8
         9 . . 10
 GPIO0  11 # # 12  PCM_CLK / PCM0
        13 . . 14
 GPIO3  15 # # 16  GPIO4
        17 . # 18  GPIO5
  MOSI  19 # . 20
  MISO  21 # # 22  GPIO6
  SCLK  23 # # 24  CS0
        25 . # 26  CS1
```
@StoneAgeSkillz this is your pinout (look top down at your raspberrypi when pins are located at upper-right corner of board, # is used for communication). Also read this http://pi4j.com/pins/model-2b-rev1.html

And this part is wrong too:
```
static struct fb_var_screeninfo ili9325_var __initdata = {
   .xres       = 320,
   .yres       = 240,
   .xres_virtual   = 320,
   .yres_virtual   = 240,
   .width      = 320,
   .height     = 240,
...
```

smartdriver, 2015-06-25:
I have also been following this thread over the last week as I have a couple of these screens too. After some digging through the Wayback Machine, I found this link for the screen: http://detail.1688.com/offer/43147400868.html?spm=a2615.2177701.0.0.NARbt5

It says it's a ili9327

Hope this of some help

ont, 2015-06-29:
I got some progress at understanding how this device works. I convert some code from UTFT library for arduino. Here is my prototype program on python which draws yellow rectangle 100x100 pixels (some other garbage at screen coming from protection film attached to the screen) :
[image]
Note that Init sequence is still not right. I don't know correct values for voltages so screen is flickering. But this pinout is absolutely correct:
```
                             3.3v   1 # . 2
                                    3 . . 4
                                    5 . # 6   ground
    session write trigger    [cs]   7 # . 8
                                    9 . . 10
       byte write trigger    [wr]  11 # # 12  [rs]   command / data switch
                                   13 . . 14
                             [d1]  15 # # 16  [d2]   d1-d8: bits of byte for command / data
                                   17 . # 18  [d3]
                             [d4]  19 # . 20
                             [d6]  21 # # 22  [d5]
                             [d7]  23 # # 24  [d8]
                                   25 . # 26  [rst]  hard reset trigger
```
Here is prototype code: https://gist.github.com/ont/52c91e4ff090ee805e27
I still waiting my logic analyzer for getting correct values for init sequence...

sfemping, 2015-07-01:
Great progress on confirming pinouts and actually getting something to display!

I suppose with the closed kernel, the logic analyzer is going to be the only way to get the init strings for the controller by sniffing them off the bus...? Would the init be going over the parallel d1-d8?

I get the impression there are quite a few people wanting to break away from this limitation, these screens were going at a good price... (although I still see them advertised for top dollar too!)

ont, 2015-07-02:
And finally I got init sequence for this display with my new logic analyzer. It is worth noting that init sequence is hardly obfuscated with random inserted commands: 0x2a (set column address), 0x2b (set page address) and 0x2c (write memory start). Without command 0x29 (set display on) they are doing nothing, so i remove them and got this variant (not very different from UTFT variant):
```
> 0xe9    ???
  0x20    ???
> 0x11    Exit sleep mode
> 0xd1    VCOM Control  (voltages)
  0x0        SELVCM  0x00 = Register D1h for VCM setting
  0x71       VCM     0x71 = VREG1OUT x 0.944
  0x19       VDV     0x19 = VREG1OUT x 1.20
> 0xd0    Power setting (voltages)
  0x7        VC      0x07 = V_ci1 = 1.0 x V_ci
  0x1        BT      0x01 ->  DDVDH = V_ci1 x 2, VCL = -V_ci1, VGH = V_ci1 x 6, VGL = -V_ci1 x 4
  0x8        VRH     0x08 ->  VCIRE = 0 (external reference voltage VciLVL),  V_reg1out = VciLVL x 1.800
> 0x36    Set address mode
  0x28       0x28 = 0b00101000
                0 -> page address order (0=top to bottom)
                0 -> column address order (0=left to right)
                1 -> page/column order (1=reverse mode)
                0 -> line address order (0=LCD refresh top to bottom)
                1 -> RGB/BGR order (1=pixel sent in BGR order)
                0 -> .. skip ..
                0 -> horizontal flip (0=normal display, non flipped)
                0 -> vertical flip (0=normal display, non flipped)
> 0x3a    Set pixel format
  0x5        0x05 = 0b0000101
                          |-|----> DBI pixel definition (16bit/pixel, 65,536 colors)
> 0xc1    Display timing setting for normal/partial mode
  0x10       BC + DIV:   0x10 (default) = 0b10000; BC0 = 1 (line inversion waveform is selected); DIV0[1] = 0, DIV[0] = 0 (division ratio = 1/1)
  0x10       RTN:        0x10 (default) line period is equal to 16 clocks
  0x2        BP:         0x02 (default) front porch period (blank period before the beginning of display) equals to 2 lines
  0x2        FP:         0x02 (default) back porch period (bland period before the beginning of display) equals to 2 lines
> 0xc0    Panel driving setting
  0x0        SS+BGR+GS+SM+REV:  0x00 = 0b00000 (default) 
                0 -> gate scan order (default)
                0 -> gate scan order (default)
                0 -> grayscale inversion of image (0=no inversion)
                0 -> display data pixel order (0=in RGB order)
                0 -> shifting direction (0=default)
  0x35       NL:  0x35 (default)  number of lines to drive LCD  8 * (0x35+1) = 432 lines
  0x0        SCN: 0x00 (default)  starting gate for scan  (G[1 + 0x00*4] = G[1])
  0x0        PTG: 0x00 (default)  scan mode in non-display area (0=normal scan)
  0x1        PTG+ISC: 0x01 (default)  scan cyle in non-display area drive period (0x01= 3 frames, 50ms), 
  0x2        DIVE: 0x02 (default)  clock frequency for DPI interfaces (0x02=1/4 division ratio)
> 0xc5    Frame rate control
  0x0        FRA   0x00  frame rate of display (0=96 Hz (maximal))
> 0xd2    Power setting for normal mode
  0x1        AP0         0x01 (default)  amplifier equal to 1.00  (in no-display period, recommended set to 0x00 for energy consumption)
  0x44       DC00+DC10   0x44 (default)  0x44 = 0b1000100  charge pump frequencies (0x04 and 0x04 = F_osc / 16 and F_osc / 256)
> 0xc8    Gamma setting (non-default settings!)
  0x4        \
  0x67        |
  0x35        | 
  0x4         |
  0x8         |
  0x6         |
  0x24        |
  0x1          >  bla-bla-bla...  KP0-KP3, RP0-RP1, VRP0-VRP1, KN0-KN5, RN0-RN1, VRN0-VRN1, VREP0-VREP1, VREN0-VREN2 params
  0x37        |
  0x40        |
  0x3         |
  0x10        |
  0x8         |
  0x80        |
  0x0        /
> 0x29    Set display on
```

Aaand... @notro can you point us what to do next?  How to write correct device-tree or where to start for creating new driver?

notro, 2015-07-05:
0x2a, 0x2b and 0x2c are used to set which part of the display that will be updated. In FBTFT this is done in the set_addr_win() function on each update.
Look through all the FBTFT drivers and find the ones that use those registers in set_addr_win(). Then look at those and see if you find one that sets 0x36 = 0x28. This controls rotation among other things.
If you find one, use that and see if it works. If it doesn't work, use one that doesn't have set_var() implemented as that will mess up your settings.
Example overlay that has a custom init sequence: https://github.com/raspberrypi/linux/blob/rpi-4.0.y/arch/arm/boot/dts/overlays/mz61581-overlay.dts
In your case:
```
init = <0x10000e9 0x20
        0x1000011
...
```

The flexfb driver does not support Device Tree: ```modprobe flexfb width=XX height=XX init=-1,0xe9,0x20,-1,0x11,-1,...,-3```

NedScott, 2015-07-09:
I have one of these guys too, and right now it's not useful to me at all if I can only use their image. If you guys need anything more destructive done to find out information, let me know and I will be glad to cut, slice, burn, or whatever.

I wouldn't mind even sending it to someone who's more knowledgeable about reverse engineering this kind of thing. The only bad thing is that it seems these aren't being made anymore, as the ebay prices for them have gone from about $9 USD to about $40 USD, so it might really be a "dead end" product.

berryelectronics, 2015-07-09:
@NedScott I dont know from where you got that price, but on aliexpress it seems to stay at 8$.

NedScott, 2015-07-09:
@berryelectronics so it is, excellent! I guess those on ebay are just being cheap right now, lol. Thank you for the heads up.

samuraijap, 2015-07-10:
$ sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf
Option “fbdev” “/dev/fb1″
$ sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update
$ sudo reboot
$ sudo nano /etc/modules
fbtft_device name=flexpfb rotate=180 fps=60
gpios=dc:18,reset:7,wr:17,cs:4,db00:22,db01:23,db02:24,db03:10,db04:25,db05:9,db06:11,db07:8
flexfb width=240 height=400 buswidth=8 init=-1,0xE9,0x20,-1,0x11,-2,100,-1,0x3A,0x55,-1,0xD1,0x00,0x6B,0x19,-1,0xD0,0x07,0x07,0x80,-1,0x36,0x48,-1,0xC1,0x10,0x10,0x02,0x02,-1,0xC0,0x00,0x35,0x00,0x00,0x01,0x02,-1,0xC5,0x01,-1,0xD2,0x01,0x22,-1,0xC8,0x01,0x52,0x37,0x10,0x0D,0x01,0x04,0x51,0x77,0x01,0x01,0x0d,0x08,0x80,0x00,-1,0xEA,0x80,-1,0x29,-3
*landscape mode  "width = 400 height = 240",  "-1,0x36,0x48 to -1,0x36,0x28"
Enable the Boot Splash Screen.
$ sudo nano /boot/cmdline.txt
fbcon=map:10 fbcon=font:VGA8x8 fbcon=rotate:3

berryelectronics, 2015-07-10:
@samuraijap  confirmed to work with newest kernel? Would be reaaly nice :+1:

webmonger, 2015-07-10:
@berryelectronics I have just followed the instructions to the best of my ability and I don't see anything on the screen. The only thing I missed out was the line "Enable the Boot Splash Screen." as I assumed that that was a title rather than something for the configuration.

Also I added the "fbcon=map:10 fbcon=font:VGA8x8 fbcon=rotate:3" just before the rootwait command as I assumed that it should be the last command on bootup.

@samuraijap Have I followed the instructions incorrectly?

berryelectronics, 2015-07-10:
@webmonger After a reboot it atleast cleared my screen. Thats something I guess. I have no clue why he uses fb1. could just find fb0 in the dev folder

samuraijap, 2015-07-11:
input the In a row  "fbtft_device"  from "db07:8"  to   add line break.
 input the In a row  "flexfb" from "-3 " to  add line break.

webmonger, 2015-07-11:
@samuraijap Did you apply this on to the known working image or was this on a clean install of raspbian?

ont, 2015-07-11:
Wow! So much progress!
@samuraijap you possibly have mistake in init command (240x432 instead of 240x400). To anyone who want to try manually:

* rpi-update with REPO_URI is not necessary as latest fbtft drivers is already added in kernel tree. Just do rpi-update
* call command:
```
modprobe fbtft_device name=flexpfb rotate=180 fps=60 gpios=dc:18,reset:7,wr:17,cs:4,db00:22,db01:23,db02:24,db03:10,db04:25,db05:9,db06:11,db07:8
```
* call command:
```
modprobe flexfb width=240 height=432 buswidth=8 init=-1,0xE9,0x20,-1,0x11,-2,100,-1,0x3A,0x55,-1,0xD1,0x00,0x6B,0x19,-1,0xD0,0x07,0x07,0x80,-1,0x36,0x48,-1,0xC1,0x10,0x10,0x02,0x02,-1,0xC0,0x00,0x35,0x00,0x00,0x01,0x02,-1,0xC5,0x01,-1,0xD 2,0x01,0x22,-1,0xC8,0x01,0x52,0x37,0x10,0x0D,0x01,0x04,0x51,0x77,0x01,0x01,0x0d,0x08,0x80,0x00,-1,0xEA,0x80,-1,0x29,-3
```
* after steps 2 and 3 you should see two devices
/dev/fb0
/dev/fb1
* I don't know at this moment what is **/dev/fb0**, but **/dev/fb1** seems to work properly. To test just run
```cat /dev/urandom > /dev/fb1```

At current moment I try to patch fbtft_device.c and add custom fb_ili9327.c then compile kernel module. But it seems to be harder than I expected.

samuraijap, 2015-07-11:
The resolution of this panel is 400x240.
ILI9327 driver chip is  up to 432×240 support.

ont, 2015-07-11:
@samuraijap yes, my mistake. I counted pixels in photo of my screen and it is equal to 400...

samuraijap, 2015-07-11:
I attached a picture together that time.
[image]
[image]

berryelectronics, 2015-07-11:
@samuraijap Now it works! That is great! :+1: 

But somehow it seems like it needs longer to refresh the screen. Did you managed to get the console on the screen? "con2fbmap 1 1" didnt helped me.

keerttttt, 2015-07-11:
Ok, so the framerate/display speed is slower then with the normal included kernel? Or not?

Great to see it is working! :)

berryelectronics, 2015-07-11:
@keertttt i was using the original image With update upgrade and rpi-update. Frameratr felt like 1-2fps. Desktop took around 5 -7 seconds to built up. 
But atleast it works now With the New kernel ^^

keerttttt, 2015-07-11:
Hm, if we could improve this, it would be great. What are the possibilities
in this?

ont, 2015-07-11:
My observations:
1) **/dev/fb0** from original driver works 1.5-2.0x faster, but not too fast (i can see refreshing process when I fill it with random data)
2) system with 3.6.11 kernel with original driver works faster (it is only my feelings).
3) glxgears and glxheads works very fast on original kernel

My idea: they are using caching for drawing area and updating only changed pixels. Also authors of original driver may do some low-level system optimization.

@notro, @samuraijap what are you thinking about this?

keerttttt, 2015-07-11:
On the original driver, full image change took about 1 sec (top to bottom).
Faster than this should be possible right?

>
> My observations:
> 1) /dev/fb0 from original driver works 1.5-2.0x faster, but not too fast
(i can see refreshing process when I fill it with random data)
> 2) system with 3.6.11 kernel with original driver works faster (it is
only my feelings).
> 3) glxgears and glxheads works very fast on original kernel
>
> My idea: they are using caching for drawing area and updating only
changed pixels. Also authors of original driver may do some low-level
system optimization.

[24 more replies]

closed 2021-08-24 by github-actions[bot]