notro/fbtft #255: ADS7846 in rpi-display-overlay not working on recent 3.18.y kernel

cnvogelg, 2015-03-05:
Hi,

I have a RPi Model B with a Watterott rpi-display (8 Bit Mode enabled). It was running fine on a custom image with your fbtft drivers, the ads7846_device, and kernel 3.12.y without DT.

Now I upgraded to a recent Raspbian image and I am using the DT with the staging fbtft drivers.
I added the dpi-display-overlay to enable the display and the touch controller.
While the display runs fine with this overlay, the touch controller does not respond at all.

    > uname -a
    Linux ernie 3.18.8+ #764 PREEMPT Tue Mar 3 15:17:20 GMT 2015 armv6l GNU/Linux

It gets detected and registered as an input device:

    > dmesg
    ...
    [    6.428853] bcm2708_spi 20204000.spi: master is unqueued, this is deprecated
    [    6.799227] ads7846 spi0.1: touchscreen, irq 419
    [    6.953567] input: ADS7846 Touchscreen as    /devices/soc/20204000.spi/spi_master/spi0/spi0.1/input/input2

Remarkable here is: In older kernels it got ``irq 195`` now it uses ``irq 419``...

Running ```evtest /dev/input/input2``` gives no reaction on any movements on the screen...
```
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "ADS7846 Touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max     4095
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max     4095
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max      255
Properties:
Testing ... (interrupt to exit)
```
Looking at the interrupts I don't see any incoming calls:

    $ cat /proc/interrupts | grep ads
    419:          0  pinctrl-bcm2835  25  ads7846

Looks to me that something is wrong with the GPIO irq...

Anyone has any ideas?
or has anyone a working setup with the new kernel and the overlay?

notro, 2015-03-05:
I have only tested that kernel with a Pi1 B+ and a Pi2 B+.
I have a Model B somewhere so I'll test it when I'm done with an issue I'm currently working on.

cnvogelg, 2015-03-05:
TIA!

I also have the newer ones here and just ran the same test on a similar installation.
I can confirm that the this kernel with the shipped overlay works both on the P1B+ and the P2B+.

So it seems to be a regression with the old model...

notro, 2015-03-05:
Touch is working for me on my Model B (pcb 2.0).
```
$ uname -a
Linux raspberrypi 3.18.8+ #765 PREEMPT Thu Mar 5 15:41:59 GMT 2015 armv6l GNU/Linux

$ dmesg | grep model
[    0.000000] Machine model: Raspberry Pi Model B

$ cat /proc/cpuinfo | grep Revision
Revision        : 000e

$ cat /proc/interrupts | grep ads
419:        230  pinctrl-bcm2835  25  ads7846

$ grep dtoverlay /boot/config.txt
dtoverlay=rpi-display
```

Which revision do you have?

http://elinux.org/RPi_HardwareHistory#Board_Revision_History

cnvogelg, 2015-03-06:
I have two similar boards here:

```
$ uname -a
Linux bert 3.18.8+ #765 PREEMPT Thu Mar 5 15:41:59 GMT 2015 armv6l GNU/Linux

$ dmesg | grep Model
[    0.000000] Machine model: Raspberry Pi Model B

$ cat /proc/cpuinfo | grep Rev
Revision        : 000e

$ cat /proc/interrupts | grep ads
419:          0  pinctrl-bcm2835  25  ads7846

$ grep dtoverlay /boot/config.txt
dtoverlay=rpi-display
```

Both have the same parameters but no incoming ADS IRQs...

My /boot/config.txt has activated:
```
disable_overscan=1
hdmi_group=2
hdmi_mode=9
gpu_mem=64
dtoverlay=rpi-display
```

My /boot/cmdline.txt contains:
```
wc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
```

The rpi-display boards are from an early revision because I had to manually switch to 8 bit mode by changing the solder bridges (namely IM0, IM1 and 1024 RS). (As described on the watterott page)
But those are only relevant for the display controller and should not affect the ADS7846...

notro, 2015-03-06:
I have one of those 9-bit displays also and I can confirm that touch doesn't work.
When I look at the state of the irq pin, it's low, when it should be high:
```
$ gpio -g read 25
0
```
Enable pullup:
```
$ gpio -g mode 25 up
$ gpio -g read 25
1
```
Now it works.

It's strange that this worked previously, because ads7846_device doesn't pull up the irq gpio.

@awatterott can you confirm that the irq is floating on these displays?
I can enable pullup in the DT overlay.

(gpio is from wiringpi.com)

cnvogelg, 2015-03-06:
I can confirm that setting the pull-up fixes my problem here, too! Nice find :)

I just revisted the old image and found out that there this pin was configured as an **OUTPUT**!?!
So the value reads 1 and the ADS7840 was still able to drive it low....

```
$ gpio -g readall
+----------+-Rev2-+------+--------+------+-------+
| wiringPi | GPIO | Phys | Name   | Mode | Value |
+----------+------+------+--------+------+-------+
...
|      6   |  25  |  22  | GPIO 6 | OUT  | High  |
```

Looking at the [schematic](https://github.com/watterott/RPi-Display/blob/master/pcb/RPi-Display_v10.pdf) there is a pull-up placed (R8 100k) that should do the job

I am now checking the board if the early revision had a soldering problem at R8...

cnvogelg, 2015-03-06:
No, soldering is fine. I first stumbled across the fact that I can't measure the Pull-Up on the RPi 3V3 Rail (Pin 1 EXT Connector) because the Pull-Up is connected to the local rpi-display 3V3 Rail. But there 100k is connected.

I wonder if the R8 value was changed in later revisions?
Looking at the B+ schematic of the display the value was reduced to 47k (now R7)

notro, 2015-03-06:
I have now tested the 9-bit display on the Pi2 and touch didn't work there either.
This patch solves it:
```diff
diff --git a/arch/arm/boot/dts/rpi-display-overlay.dts b/arch/arm/boot/dts/rpi-display-overlay.dts
index 0578810..a8fa974 100644
--- a/arch/arm/boot/dts/rpi-display-overlay.dts
+++ b/arch/arm/boot/dts/rpi-display-overlay.dts
@@ -30,6 +30,7 @@
                        rpi_display_pins: rpi_display_pins {
                                brcm,pins = <18 23 24 25>;
                                brcm,function = <1 1 1 0>; /* out out out in */
+                               brcm,pull = <0 0 0 2>; /* - - - up */
                        };
                };
        };
```
And the B+ version (40-pin) of the display still works (not suprising).
I'll send it in along some other overlay patches I'm working on.

cnvogelg, 2015-03-07:
Thanks, this patch solves this issue.

I'll close this ticket, but I am still curious about the floating gpio input in the first batch of the model b rpi-display. Do you also have a later revision of the rpi-display for model b? Maybe you could spot a difference?

notro, 2015-03-07:
I have stashed away those old rpi-displays now.
I was rather hoping Andreas Watterott could settle this.
I know he is very busy, maybe he chimes in when he's back on work monday.

notro, 2015-03-07:
The change has been merged and is present in latest rpi-update release.
https://github.com/raspberrypi/linux/pull/874
https://github.com/Hexxeh/rpi-firmware/commit/b6201ffde8f845cb59a0c6ce47ce2ee8fa9e5a44

cnvogelg, 2015-03-09:
I just emailed with Andreas and he confirmed that newer models of the model b rpi display also use 47k pull ups. So it seems 47k pull up is working without internal pull up and 100k does not.

Nevertheless, the patched overlay works for both types of pull ups. So lets stick with it.

closed 2015-03-09 by cnvogelg