notro/fbtft #258: 2015-03-27 update : ads7846 DT overlay doesnt seem to work

mwilliams03, 2015-04-01:
Hi Notro
I am trying out the recent updates you have made.
And when trying to use ads7846 overlay, ads7846 fails to load.

If /boot/config.txt has
'''
dtoverlay=piscreen,speed=32000000,rotate=90
'''
The two drivers for the display and touchscreen load.

however, I would like to swap X & Y axis, so I try
'''
dtoverlay=piscreen,speed=32000000,rotate=90
dtoverylay=ads7846,swapxy=1
'''
However, the ads7846 driver fails to load.

/var/log/kernel shows
'''ads7846: probe of spi0.1 failed with error -22'''
I think -22 indicate invalid parameters.

If I try it without the swapxy;
'''
dtoverlay=piscreen,speed=32000000,rotate=90
dtoverylay=ads7846
'''
The driver still fails to load and the log still shows -90.

Am i doing something wrong?

notro, 2015-04-01:
The overlays was never meant to be used together. That will give 2 devices and one of them will fail to load.
I added the ads7846 overlay to be used together with fbtft_device.

Something like this would give swapxy overlay property in the piscreen overlay:
```diff
-			piscreen-ts@1 {
+			piscreen_ts: piscreen-ts@1 {

	__overrides__ {
+		swapxy = <&piscreen_ts>,"ti,swap-xy?";
```

Ref
https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/boot/dts/piscreen-overlay.dts
https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/boot/dts/ads7846-overlay.dts
https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays

mwilliams03, 2015-04-02:
okay, understood.

Then this would mean to override the XY axis, I would just need to use the below line?

```dtoverlay=piscreen,speed=32000000,rotate=90,swapxy=1```

I assume this still needs to be updated in the repo?

notro, 2015-04-02:
Actually I don't know if swapxy=1 works. It's supposed to be used alone like this: swapxy
I might work, just haven't tried it. swapxy=0 will not work.

> I assume this still needs to be updated in the repo?

Yes. I have left it to the makers of the displays to take care of any changes to the overlays.
I need to free up time to do fun programming :-)
There's lots of guides on how to make Pull Requests.
It can even be done directly from the github webgui now.

The overlays are documented in [/boot/overlays/README](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README)

mwilliams03, 2015-04-02:
programming fbtft isn't fun?  :-)

swapxy by itself doesn't work, i just tested it.

I will see if I can implement the change myself.  I hope I don't break
anything.

notro, 2015-04-02:
> programming fbtft isn't fun?  :-)

Programming fbtft is fun. It's all the other surrounding stuff that has taken to much time.
Writing code is fun, making it available for easy use (prebuilt kernel) not so fun, answering questions not so fun. fbtft has become so popular that I have had problem finding time to write new code. The core of fbtft has hardly seen any change since it was made 2 years ago.
It's after all a hobby and when the Pi2 was released with it's new CPU that didn't work with the FBTFT kernel, I started to wonder if I was in an unpaid job and had lost my hobby :-)
So now I don't get that much involved in support requests. And hopefully there's no Pi3 around the corner :-)

> swapxy by itself doesn't work, i just tested it.

It worked for me on a PiScreen display :-) As did swapxy=1. I touched the the same spot on the display.
```
dtoverlay=ads7846,penirq=17,xohms=100,pmax=255

Event: time 1427974230.515996, type 3 (EV_ABS), code 0 (ABS_X), value 314
Event: time 1427974230.515996, type 3 (EV_ABS), code 1 (ABS_Y), value 3157

dtoverlay=ads7846,penirq=17,xohms=100,pmax=255,swapxy

Event: time 1427974282.102672, type 3 (EV_ABS), code 0 (ABS_X), value 3300
Event: time 1427974282.102672, type 3 (EV_ABS), code 1 (ABS_Y), value 336

dtoverlay=ads7846,penirq=17,xohms=100,pmax=255,swapxy=1

Event: time 1427974346.482833, type 3 (EV_ABS), code 0 (ABS_X), value 2480
Event: time 1427974346.482833, type 3 (EV_ABS), code 1 (ABS_Y), value 342

```

mwilliams03, 2015-04-02:
I do see you doing a lot of support requests, so it must eat up a lot of
your time.

Well... now I am confused.  (i'll admit that im not up to full speed with
the use of overlays)

For the test you just did with evtest, was fbtft loaded?  or just ads7846?

I got the impression that I would need to use an override like this;
dtoverlay=piscreen,speed=32000000,rotate=90,swapxy

notro, 2015-04-02:
No, I just used the ads7846 overlay alone to see if swapxy could be used with and without an assignment.

mwilliams03, 2015-04-02:
understood.

well.. i submitted a pull.. I hope this works
https://github.com/raspberrypi/linux/pull/917

Texy, 2015-04-02:
Hi,
I am using fbtft_device with my HY28B displays, together with the ads7846 overlay enabled :
dtoverlay=ads7846, penirq=17, pmax=255, ymin=190, ymax=3850, xmax=3850, xmin=230, xohms=100, swapxy=1, verbose=3

This works for em, hpowever, changing to swapxy=0, or removing it altogether makes no difference, so I conclude that it is not being actioned.
Texy

notro, 2015-04-02:
It's the presence of the property *ti,swap-xy* in the DT that matters. The value doesn't matter.
These are my results:
```
dtoverlay=ads7846,penirq=17,xohms=100,pmax=255

$ sudo vcdbg log msg 2>&1 | grep dtparam
001405.350: dtparam: penirq=17
001409.311: dtparam: xohms=100
001410.922: dtparam: pmax=255

$ ls -l "/proc/device-tree/soc/spi@7e204000/ads7846@1/ti,swap-xy"
ls: cannot access /proc/device-tree/soc/spi@7e204000/ads7846@1/ti,swap-xy: No such file or directory

---

dtoverlay=ads7846,penirq=17,xohms=100,pmax=255,swapxy

$ sudo vcdbg log msg 2>&1 | grep dtparam
001405.068: dtparam: penirq=17
001409.030: dtparam: xohms=100
001410.643: dtparam: pmax=255
001412.261: dtparam: swapxy=true

$ ls -l "/proc/device-tree/soc/spi@7e204000/ads7846@1/ti,swap-xy"
-r--r--r-- 1 root root 0 Apr  2 16:45 /proc/device-tree/soc/spi@7e204000/ads7846@1/ti,swap-xy

---

dtoverlay=ads7846,penirq=17,xohms=100,pmax=255,swapxy=0

$ sudo vcdbg log msg 2>&1 | grep dtparam
001404.847: dtparam: penirq=17
001408.807: dtparam: xohms=100
001410.419: dtparam: pmax=255
001412.037: dtparam: swapxy=1

$ ls -l "/proc/device-tree/soc/spi@7e204000/ads7846@1/ti,swap-xy"
-r--r--r-- 1 root root 0 Apr  2 16:41 /proc/device-tree/soc/spi@7e204000/ads7846@1/ti,swap-xy

---

dtoverlay=ads7846,penirq=17,xohms=100,pmax=255,swapxy=1

$ sudo vcdbg log msg 2>&1 | grep dtparam
001406.057: dtparam: penirq=17
001410.016: dtparam: xohms=100
001411.630: dtparam: pmax=255
001413.248: dtparam: swapxy=1

$ ls -l "/proc/device-tree/soc/spi@7e204000/ads7846@1/ti,swap-xy"
-r--r--r-- 1 root root 0 Apr  2 16:44 /proc/device-tree/soc/spi@7e204000/ads7846@1/ti,swap-xy
```

mwilliams03, 2015-04-03:
I also tested with ads7846 overlay and I was able to swap XY.

I was also did some testing with making modifications to a local .dts to
see what the effects would be.  using;
sudo dtc -@ -I dts -O dtb -o /boot/overlays/piscreen-overlay.dtb
 piscreen-overlay.dts

I was able to get XY swapped when using piscreen overlay.
I have requested another pull to get this updated within the raspberrypi
Repo.

Notro, thanks for you help and guidance with getting this resolved.

Mark

Texy, 2015-04-04:
OK, so to confirm, just swapxy needs to be included in the overlay parameters, not swapxy=1 (or =0).
I see that Marks pull request has been accepted and the issue closed. Does that mean if I now
sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update
the change will be incorporated? Or will it take a while to filter through?
Texy

notro, 2015-04-04:
Marks change only affects PiScreen, and will be available when Dom Cobley does a build in a few days.
If you want the same on the hy28b overlay, you can make a PR to get the same change. Or add a property like in the ads7846 overlay.
I see know that I should have added swaxp to all the display overlays since the displays are rotated by default.

Texy, 2015-04-04:
OK, understood. I have NO idea how to get this implemented via GIT for the hy28a and hy28b overlays, as Mark has done. Also I do not understand why Marks addition doesn't change the __overrides__ section of the code? I'm thinking that the section that has changed is the default values, but the overrides section allows them to be changed.to other values.
Texy

notro, 2015-04-05:
Mark just changed the default value without providing an overlay property.
This should get you what you want (not tested):
```diff
 		fps = 		<&hy28b>,"fps:0";
 		debug = 	<&hy28b>,"debug:0";
 		xohms =		<&hy28b_ts>,"ti,x-plate-ohms;0";
+		swapxy =	<&hy28b_ts>,"ti,swap-xy?";
 		resetgpio =	<&hy28b>,"reset-gpios:4",
 				<&hy28b_pins>, "brcm,pins:1";
 		ledgpio =	<&hy28b>,"led-gpios:4",
```
When you have tested that this works:
Go to https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/boot/dts/hy28b-overlay.dts
Then click the pencil icon to edit.
Make change, preview to make sure you haven't messed up.
Add commit message, then Propose file change.
I don't remember the next steps.

mwilliams03, 2015-04-05:
Texy, if you want to test before making the changes.. download the latest copy of hy28b-overlay.dts to your Pi.  

Make the modifications on the local copy. Then compile with;
```sudo dtc -@ -I dts -O dtb -o /boot/overlays/hy28b-overlay.dts  hy28b-overlay-testing.dts```

reboot

guide here to downloading dtc;
https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays

This is a summary of what i did to get my change in the linux/raspberry repo;
1)go to https://github.com/raspberrypi/linux and create a fork.. top right button
2)download forked repo, which should be under your git account  https://github.com/mwilliams03/linux
3)make your changes to arch/arm/boot/dts/hy28a-overlay.dts
4)then `git add arch/arm/boot/dts/hy28a-overlay.dts`
5)`git commit -m "reason for change"`
6)`git push origin  'git push origin --force`      
7)create a pull request on https://github.com/raspberrypi/linux.  This tells the admin of raspberrypi/linux to pull the changes from your forked repo
8)The pull request should appear here https://github.com/raspberrypi/linux/pulls  waiting for someone to approve.

I am not sure if the above is the best or cleanest way... but it worked for me.
Here are two good videos that go over the process;
https://www.youtube.com/watch?v=1S_526C8Gkw
https://www.youtube.com/watch?v=NnBb9NTk-To

closed 2015-06-18 by notro