notro/fbtft #259: compilation error for piTFT 2.8''

abhi-84, 2015-04-01:
I am getting followinf error when I try to make after making appropriate selections in kernel configuration. What could be the possible issue? I am using gcc 4.8.1

arch/arm/kernel/asm-offsets.c:53:2: error: #error Your compiler is too buggy; it is known to miscompile kernels
 #error Your compiler is too buggy; it is known to miscompile kernels
  ^
arch/arm/kernel/asm-offsets.c:54:2: error: #error and result in filesystem corruption and oopses.
 #error and result in filesystem corruption and oopses.
  ^
make[1]: *** [arch/arm/kernel/asm-offsets.s] Error 1
make: *** [prepare0] Error 2

please help

notro, 2015-04-01:
gcc 4.8.1 isn't trusted: https://github.com/raspberrypi/linux/commit/7fc150543c73de71859631c8a6b17e3067fe7617

abhi-84, 2015-04-20:
got it. I am currently building it with gcc 4.9.1 and it works fine

closed 2015-04-20 by abhi-84