notro/fbtft #260: I am getting "FATAL: Module ads7846_device not found."

Ferite, 2015-04-02:
Hi.

I have installed a HY28A display to my Raspberry Pi B+, and I am having problems getting the Touchpanel to work. As the TFT worked without problems with the standard Foundation image, I just compiled the touch driver (runned make in the ads7846_device directory) and then installed them.

As I have not clear how installed then I just copied the resulting ads7846_device.ko in /lib/modules/3.18.0-trunk-rpi/kernel, and then runned $depmod in that directory. Then I tried to run the instruction in https://github.com/notro/fbtft/wiki/Touchpanel:

$sudo modprobe ads7846 
Seems to work ok.

$sudo modprobe ads7846_device 
Produces "FATAL: Module ads7846_device not found."

If I run:

$lsmod

I get ads7846 listed (not ads7846_device, off course) .

If I run:

$dmesg | grep -i ads7846

I don't get any message.

I hope I can get some help about what I am doing wrong.

Thanks.

notro, 2015-04-02:
The latest kernel doesn't have ads7846_device anymore. The same can be achieved with a Device Tree overlay. Lookup ads7846-overlay in [/boot/overlays/README](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README)
I'll make a note on that wiki page.

Ferite, 2015-04-08:
Hi Notro.

I have the TS working now, by loading ads7846-overlay.dtb. The only issue I still having is that in order to avoid having the X axe inverted, I have to to execute the comand:

$ DISPLAY=:0 xinput --set-prop 'ADS7846 Touchscreen' 'Evdev Axis Inversion' 1 0

It works ok if I run this command from a terminal within the session. However if I add this line to /etc/X11/xinit/xinitrc (before $. /etc/X11/Xsession), it has no effect.

Where else can I place this command?.

Thanks again for your help, and all the work you have put on fbtft.

notro, 2015-04-08:
Try /etc/xdg/lxsession/LXDE-pi/autostart

You can also use xinput-calibrator: https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian#x-windows-touch-calibration

johnnyquad, 2015-04-11:
Hi folks,
I have a new Waveshare 4 Spotpear

Screen works fine but can't get the touch bit working.
things done so far 

end of config.txt:
device_tree=bcm2708-rpi-b.dtb
dtparam=spi=on
dtparam=i2c_arm=on

device_tree=ads7846-overlay.dtb
dtoverlay=ads7846,cs=1, penirq=17, speed=500000, swapxy=0, pmax=255, xohms=100, xmin=200, xmax=3900, ymin=200, ymax=3900, verbose=3

end of /etc/modules:
snd-bcm2835
spi-bcm2708
i2c-bcm2708  
i2c-dev

flexfb nobacklight width=480 height=320 regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3 
fbtft_device debug=3 rotate=0 name=flexfb speed=16000000 gpios=reset:25,dc:24
ads7846

reboot
dmesg:
[   11.743981] bcm2708_spi bcm2708_spi.0: SPI Controller at 0x20204000 (irq 80)
[   11.746435] bcm2708_spi bcm2708_spi.0: SPI Controller running in dma mode
[   11.905681] bcm2708_i2c_init_pinmode(1,2)
[   11.908042] bcm2708_i2c_init_pinmode(1,3)
[   11.914519] bcm2708_i2c bcm2708_i2c.1: BSC1 Controller at 0x20804000 (irq 79) (baudrate 100000)
[   12.083345] i2c /dev entries driver
[   12.172550] pcm512x 1-004d: Failed to reset device: -5
[   12.175001] pcm512x: probe of 1-004d failed with error -5
[   12.243136] pcm512x 1-004c: Failed to reset device: -5
[   12.245665] pcm512x: probe of 1-004c failed with error -5
[   12.298376] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[   12.314008] flexfb: module is from the staging directory, the quality is unknown, you have been warned.
[   12.437981] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[   12.447194] fbtft_device:  SPI devices registered:
[   12.449740] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[   12.452155] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   12.454812] fbtft_device:  'fb' Platform devices registered:
[   12.457070] fbtft_device:      bcm2708_fb id=-1 pdata? no
[   12.459400] fbtft_device: Deleting spi0.0
[   12.465511] flexfb spi0.0: fbtft_request_gpios: 'reset' = GPIO25
[   12.467805] flexfb spi0.0: fbtft_request_gpios: 'dc' = GPIO24
[   12.469956] flexfb spi0.0: flexfb_verify_gpios_dc()
[   12.472058] flexfb spi0.0: fbtft_init_display()
[   12.492581] flexfb spi0.0: fbtft_reset()
[   12.629637] flexfb spi0.0: init: write(0xB0) 0x00 
[   12.631890] flexfb spi0.0: init: write(0x11) 
[   12.634065] flexfb spi0.0: init: mdelay(250)
[   12.885821] flexfb spi0.0: init: write(0x3A) 0x55 
[   12.887928] flexfb spi0.0: init: write(0xC2) 0x44 
[   12.889989] flexfb spi0.0: init: write(0xC5) 0x00 0x00 0x00 0x00 
[   12.892078] flexfb spi0.0: init: write(0xE0) 0x0F 0x1F 0x1C 0x0C 0x0F 0x08 0x48 0x98 0x37 0x0A 0x13 0x04 0x11 0x0D 0x00 
[   12.895954] flexfb spi0.0: init: write(0xE1) 0x0F 0x32 0x2E 0x0B 0x0D 0x05 0x47 0x75 0x37 0x06 0x10 0x03 0x24 0x20 0x00 
[   12.899769] flexfb spi0.0: init: write(0xE2) 0x0F 0x32 0x2E 0x0B 0x0D 0x05 0x47 0x75 0x37 0x06 0x10 0x03 0x24 0x20 0x00 
[   12.903676] flexfb spi0.0: init: write(0x36) 0x28 
[   12.905705] flexfb spi0.0: init: write(0x11) 
[   12.907583] flexfb spi0.0: init: write(0x29) 
[   13.078578] flexfb spi0.0: Display update: 1772 kB/s (169.165 ms), fps=0 (0.000 ms)
[   13.159816] graphics fb1: flexfb frame buffer, 480x320, 300 KiB video memory, 4 KiB DMA buffer memory, fps=50, spi0.0 at 16 MHz
[   13.164313] fbtft_device:  GPIOS used by 'flexfb':
[   13.165935] fbtft_device:    'reset' = GPIO25
[   13.167631] fbtft_device:    'dc' = GPIO24
[   13.169160] fbtft_device:  SPI devices registered:
[   13.170941] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   13.174404] fbtft_device:      flexfb spi0.0 16000kHz 8 bits mode=0x00
[... 52 lines cut ...]
snd_soc_pcm512x_i2c     2130  0 
snd_soc_tas5713         5866  0 
snd_soc_pcm512x         8508  1 snd_soc_pcm512x_i2c
regmap_spi              2307  1 snd_soc_wm8804
regmap_i2c              2944  3 snd_soc_wm8804,snd_soc_pcm512x_i2c,snd_soc_tas5713
i2c_dev                 6713  0 
i2c_bcm2708             6200  0 
spi_bcm2708             8684  0 
snd_bcm2835            21157  0 
snd_soc_bcm2708_i2s     7515  0 
regmap_mmio             3548  1 snd_soc_bcm2708_i2s
snd_soc_core          174135  4 snd_soc_pcm512x,snd_soc_wm8804,snd_soc_tas5713,snd_soc_bcm2708_i2s
snd_compress            8714  1 snd_soc_core
snd_pcm_dmaengine       5850  1 snd_soc_core
snd_pcm                90786  4 snd_bcm2835,snd_soc_wm8804,snd_soc_core,snd_pcm_dmaengine
snd_seq                61089  0 
snd_seq_device          7205  1 snd_seq
snd_timer              22990  2 snd_pcm,snd_seq
evdev                  11020  4 
snd                    66285  7 snd_bcm2835,snd_soc_core,snd_timer,snd_pcm,snd_seq,snd_seq_device,snd_compress
 

Any help greatly received

john

Ferite, 2015-04-13:
Hi John. I am not very advanced fbtft or Linux user but here are my recomendations:

1. Begin with a clean RPi fundation image on your SD.

2. From the .dts Generate the .dtb file. The .dts is here:

https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/boot/dts/ads7846-overlay.dts

the instructions for compiling the dtb are here:

https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays

3. Copy the resulting .dtb to  /boot/overlays

4. Add to /boot/config.txt the dtoverlay (dtoverlay=ads7846,cs=1...) line.

5. My config.txt only has uncommented "dtparam=spi=on" and "dtoverlay" lines.

johnnyquad, 2015-04-14:
@ferite
Thanks for the reply
Downloaded the compiler and compiled from source..
all was still the same :-(
It turns out I needed to comment out the line "device_tree=ads7846-overlay.dtb" in my config.txt file.
I also put in "dtdebug=1" in the config.txt file this gives a load of debug info when you run $sudo vcdbg log msg.

config.txt

dtdebug=1
dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=ads7846,cs=1, penirq=17, speed=500000, swapxy, pmax=255, xohms=100, xmin=200, xmax=3900, ymin=200, ymax=3900

Anyway all working now

closed 2015-06-18 by notro