notro/fbtft #262: A couple of general questions about fbtft.

Ferite, 2015-04-13:
Hi notro:

I have a couple of questions about your fbtft drivers:

1. Independently of speed and if you don't need a high frame update rate, what demands more proccesor usage, the drivers making use of DMA, or accesing memory in the "standard" way?. (as they work with the stock Raspberry Pi Foundation kernel).

2. What level of modification would take, making the drivers work using a USB to SPI bridge. It would be even possible?. I think the most popular chip is FTID.

http://www.ftdichip.com/Products/Cables/USBMPSSE.htm

It supposes to perform 30Mbps.

Thanks for your help, and best regards.

notro, 2015-04-14:
1. FBTFT using DMA will always perform better, but you will probably not need it if you don't play video or use fbcp. Use ```top``` to see cpu utilization.

2. I don't think there's a SPI kernel driver available for that chip, only serial driver or a userspace library. FBTFT needs a kernel driver.

Ferite, 2015-04-22:
Hi notro.

If there is an escenario where fbtft driver is going to be used on a board that implements SPI, but only has 1 chip select, how can be interfaced the display and touch at the same time?. Specifically I am thinking in MinnowboarMax (x86 dev board). I can think in two options:

+ Software solution. Like declaring chip select 2 to a different pin (as I understand any free GPIO can be chip select).
+ Hardware solution. Making a simple circuit that inverts the board's output of chip select 1, an take this signal to the touch screen controller.

What do you think of these approaches?. Please let me know if they are viable or if you could help me with some advise on a better way.

Thanks.

notro, 2015-04-22:
I don't know the board, but you need 2 chip selects.
If the SPI driver doesn't support it, there is a SPI GPIO driver: http://lxr.free-electrons.com/source/drivers/spi/spi-gpio.c
Maybe it will be fast enough for the touch controller.

Ferite, 2015-04-24:
Hi notro.

Your suggestion is to use the SPI-GPIO dirver only for touch, and use the hardware SPI for LCD, or use it for both?. How can one tell tft and touch drivers what port they must use?.

Sorry if my questiones are so much basic, but in my background working with microcontrollers, you have a SPI bus and the Chip Select was whatever gpio you got free, so I don't completely  understand why the Chip Select in this boards is tied so strongly to the bus, and you can not simply define what GPIO your going to use as Chip Select.

notro, 2015-04-25:
spi-gpio is slow, that's why I suggested it for touch.

> why the Chip Select in this boards is tied so strongly to the bus, and you can not simply define what GPIO your going to use as Chip Select.

That depends on the spi driver. Sometimes the spi controller handles CS directly, if not the driver toggles CS using gpiolib.
So you have to look at the spi driver you are using to determine the options you've got.

macmeck, 2015-05-04:
Ferite,

do you know the SPI driver for the MinnowBoard MAX? Find it here: https://github.com/MinnowBoard/minnow-max-extras/tree/master/modules/low-speed-spidev

There are working examples for a display with FBTFT over SPI on the MBM (although I also did not manage to get it running yet), find references here:
http://www.mjoldfield.com/atelier/2014/12/mbmx-spi.html
http://www.mjoldfield.com/atelier/2014/12/mbmx-display.html

Ferite, 2015-05-04:
Hi Markus.

Thanks for your comment. Actually I was going to follow the steps described in the links that you provided, but these links refers only to making work the LCD, not the touchscreen. So before I put my hands on work, I was hopping to "solve" the problem regarding the board's lack of chip select 2 to drive the TS.

If you are working on notro's driver on MinnowBoard, please let me know the advances you make.

Thanks.

macmeck, 2015-05-04:
Ferite, I'm not too experienced with SPI. You might want to check on the MinnowBoard MAX mailing list. These guys could be able to help you out from hardware or driver side...

macmeck, 2015-05-04:
okay, I think you already did ;-)

Ferite, 2015-05-04:
I wrote to the mailing list, and so far this was the most usefull/viable answer:

"The thing is that the Chip Select line provided  is part of the interface as defined.  The timing of the bus is established in the hardware, and the CS1 signal is part of that timing.  So CS1 will go low whenever the bus is  active.

So, while there are technical ways to provide multiplexing for the chip select, but it will most likely require some discrete logic (or two GPIOS).  It would be very difficult to use the current CS and the GPIO pin directly, as the CS pin will go low whenever the bus is in use.

Method one Example:
CS1 .AND. GPIOnn == CS.Device1
CS1 .AND. .NOT. GPIOnn == CS.Device2

Where  the terms .AND. and .NOT. are discrete logic components.  Basically, using the combination of the two signals to create two new chip selects.

GPIO and .NOT. GPIO establishes the device select, while the Chip Select line retains its function to regulate the overall chip select in regards to the data transfer.  (there is a question of  gate delays of the signal which could risk communication, but there are ways around this if it is an issue)

Method two: For using GPIOs:
GPIOnn1 = CS.Device1
GPIOnn2 = CS.Device2
Ignore CS1 completely, but use software to handle enabling and disabling chip select.

Then the driver to support the SPI IO either has to be modified, or a front end (shim) added to the interface procedures to add the parameter for device selection and manipulate the GPIO line(s) selected for the purpose (device select - device 1 or 2) in the mix.   How this works depends upon the method.

Also, remember doing this is playing around with bus timings.  Method one will delay the chip select to the device by the time delay of the gate involved.  Method two will most likely enable the slave device before the master interface actually signals the slave enable (making the communication susceptible to bus noise during the expanded window) and keep it enabled longer..

That is the technical (academic) overview of the general question

Michael Krau"

This implies:

1. Modify the driver to activate a GPIO only when it needs to talk with touchscreen.
2. A simple logic circuit to combine existing Chip Select and the GPIO into a Chip Select 2.

I hope that notro could read this, and maybe give us some advice regarding if he thinks there could be timming issues for adding a logic circuit for enabling touchscren, and hopefully give us an overall opinion on the proposed solution.

Thanks.

closed 2015-07-07 by notro