notro/fbtft #265: FBTFT kernel with SPI DMA (FBCP Support)

miesch1, 2015-04-19:
notro,

You're the man, I am up and running with your driver in OSMC on one of Texy's HY28B shields!

I am trying to wrap my head around the effects of the new Device Tree overlays introduced in the 3.18 kernel. Everything works fine when I define "dtoverlay=hy28b..." in my config.txt. However, in your Wiki you state that we should use the fbtft_device instead of a DT overlay for performance when using fbcp (https://github.com/notro/fbtft/wiki#install). Thus, I am trying to auto-load the driver using /etc/modules and dtparam=spi=on in config.txt.

I can manually initialize the shield using modprobe, however, when I put the corresponding values in /etc/modules and reboot, the fbtft_device remains uninitialized.

I know DT overlays are intended to replace values in /etc/modules. Is there something I need to do to force the kernel to load the devices from this file? I tried completely disabling the device tree using "device_tree=", but that didn't work either.

One more question, right now I auto start the fbcp service, but I only see the proper mirror output (console) while OSMC is booting. When Kodi is running, the TFT no longer mirrors the output. I am currently accomplishing this by placing "fbcp &" in /etc/rc.local file. Is there a better way I should do this?

Thanks again for the awesome driver and any insight you can provide!
-miesch1

notro, 2015-04-19:
> However, in your Wiki you state that we should use the fbtft_device instead of a DT overlay for performance when using fbcp 

That is not what I inteded to say :smile: 

First a say that the FBTFT drivers now are in the official kernel, but DMA support is lacking.
Then I list the DT overlays available.
Then I tell how to use fbtft_device instead of DT.

Next section tells about a kernel that has DMA support as well.
```
sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update
```
This kernel is identical to the official kernel from the foundation, except that it has a DMA version of the spi-bcm2708 driver.

miesch1, 2015-04-20:
Thanks for the prompt reply notro. Let me rephrase my question, I was not very clear.

I am interested in taking advantage of the increased performance offered by the DMA supported kernel, meaning I want to use the fbtft_device rather than the DT overlay. Following your instructions, I: 
1-Added the following line to "/boot/config.txt"

    dtparam=spi=on
2-Updated the kernel with the DMA version of the SPI driver using:

    sudo -E RPI_UPDATE_UNSUPPORTED=0 REPO_URI=https://github.com/notro/rpi-firmware rpi-update
3-Added the following lines to file "/etc/modules"

    fbtft dma
    fbtft_device name=hy28b rotate=90
4-When I reboot, the driver is not enabled (screen stays white). However, I can manually enable it by running:

    sudo modprobe fbtft dma
    sudo modprobe fbtft_device name=hy28b rotate=90
5-If I replace

    dtparam=spi=on
with

    dtoverlay=hy28b, rotate=90
in "/boot/config.txt", the driver is enabled when I reboot. However, it seems like I am using the DT instead of the fbtft_device at this point now.

Am missing something needed to force the kernel to load the devices from "/etc/modules"? Thanks again for your insight, it is greatly appreciated!

-miesch1

miesch1, 2015-04-21:
So I have a clue. I didn't realize the first line in the boot console reads:

    [FAILED] Failed to start Load Kernel Modules.
    See 'systemctl status systemd-modules-load.service' for details.
Running this command does not give me much to go off, but I am pretty sure this error was introduced after updating the kernel, as described in item 2 above (which is when the OSMC splash screen was replaced with the boot console).

    osmc@osmc:~$ systemctl status systemd-modules-load.service
    * systemd-modules-load.service - Load Kernel Modules
       Loaded: loaded (/lib/systemd/system/systemd-modules-load.service; static)
       Active: failed (Result: exit-code) since Tue 2015-04-21 05:36:57 UTC; 3min 22s ago
         Docs: man:systemd-modules-load.service(8)
               man:modules-load.d(5)
      Process: 69 ExecStart=/lib/systemd/systemd-modules-load (code=exited, status=1/FAILURE)
     Main PID: 69 (code=exited, status=1/FAILURE)
Any ideas? Thanks,

-miesch1

notro, 2015-04-21:
Ok, fbtft_device does the same as the DT overlay.
If you have a kernel with SPI DMA, you get that benefit also when using DT to load the device.

> fbtft dma

dma is the default so no need to specify it.
When you use a DT overlay, the necessary modules are loaded automatically.

closed 2015-07-07 by notro