notro/fbtft #268: waveshare spot pear 4inch screen

maty87, 2015-05-02:
Yeah when i do usr/share/X11/xorg.conf.d/99-fbturbo.conf i get blank empty file

maty87, 2015-05-02:
https://github.com/notro/fbtft/issues/215#issuecomment-98417963
 For the original posts

maty87, 2015-05-02:
heres a script of the info I see from the given image with screen working from following the guide steps could it be possible to just copy these into the original official rasb image

step1
file is empty

step 2
GNU nano 2.2.6 File: /etc/modprobe.d/raspi-blacklist.conf

blacklist spi and i2c by default (many users don't need them)
#blacklist spi-bcm2708
#blacklist i2c-bcm2708
blacklist snd-soc-pcm512x
blacklist snd-soc-wm8804

step 3
not done as screen working

ste 4
file is empty

step 5
GNU nano 2.2.6 File: /boot/cmdline.txt

dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 ro$
$0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont$
$roFont6x11 logo.nologo

step 6
GNU nano 2.2.6 File: /etc/rc.local

#!/bin/sh -e
#

rc.local
#

This script is executed at the end of each multiuser runlevel.
Make sure that the script will "exit 0" on success or any other
value on error.
#

In order to enable or disable this script just change the execution
bits.
#

By default this script does nothing.
Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
printf "My IP address is %s\n" "$_IP"
fi
su -l pi -c "env FRAMEBUFFER=/dev/fb1 startx -- -dpi 60 &"
[ Read 20 lines ]
GNU nano 2.2.6 File: /etc/rc.local Modified

fi
su -l pi -c "env FRAMEBUFFER=/dev/fb1 startx -- -dpi 60 &"
exit 0

                           [ Read 20 l

maty87, 2015-05-02:
https://github.com/notro/rpi-firmware/issues/6

MolarAmbiguity, 2015-05-03:
Which instructions were you following to get this info? If you do ```/usr/share/X11/xorg.conf.d``` then ```ls``` do you get any results?

maty87, 2015-05-03:
This is the guide i been following http://www.circuitbasics.com/setup-lcd-touchscreen-raspberry-pi/ ill try that 2moro if i get chance on normal image or the one with screen?

MolarAmbiguity, 2015-05-03:
Well it seems that you can't see the file on either image, so just try both.

maty87, 2015-05-05:
on orginal image or supplied image I get blank file 
[image]
wheres do I put the 1s bit? 
also found some other info while digin around

GNU nano 2.2.6             File: /etc/modules                                 

# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
# Parameters can be specified after the module name.

snd-bcm2835
i2c-bcm2708
i2c-dev
flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0$
$0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F$
$C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0$
$2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x$
$x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-$
$0x28,-1,0x11,-1,0x29,-3
fbtft_device debug=3 rotate=90 name=flexfb speed=16000000 gpios=reset:25,dc:24
ads7846_device model=7846 cs=1 gpio_pendown=17  keep_vref_on=1 swap_xy=0 pressure_max=255 x$
$=255 x_plate_ohms=60 x_min=200 x_max=3900 y_min=200 y_max=3900

maty87, 2015-05-08:
I tried following this guide and the guides on the links but ended up on reboot with a blank black screen flashing - 

Install Required C Libraries
In order to use the API examples we provide, related C libraries 
are required, which are pre-installed in our configured system 
image. If you're using your own system image, the C libraries 
should be installed manually.
Install WiringPi 
The WiringPi installation package file can be found on:
/home/pi/WringPi under the image we provide, or can be obtained 
from the official website: 
https://projects.drogon.net/raspberry-pi/wiringpi/download-and-in
stall/
Copy the installation package into your own system via a USB 
flash drive, enter the WiringPi folder, then do this to install:
chmod 777 build
./build
Run the following command to check the installation:
gpio –v

Install C Library bcm2835
The bcm2835 installation package file can be found on: 
/home/pi/bcm2835-1.25 under the image we provide, or can be 
obtained from the official website:
http://www.airspayce.com/mikem/bcm2835/
Copy the installation package into your own system via a USB 
flash drive, enter the bcm2835-1.25 folder, then do this to 
install:
./configure
make
sudo make check
sudo make install

MolarAmbiguity, 2015-05-09:
@maty87, you haven't followed my instructions. In your second last post it appears that you have done ```sudo nano /usr/share/X11/xorg.conf.d```. If you reread my previous posts you'll find this is not what I wanted you to try. I'm attempting to help you out, but I can't do that if you don't  enter the commands correctly.

maty87, 2015-05-10:
I did try it and seen file must of miss typed something not sure now and tried installing following circuitbasics

sazp96, 2015-05-19:
Hello @maty87 and @MolarAmbiguity, do you know what is the LCD controller chip that the 4" LCD uses?

I'm trying to use this screen with an Arduino and will need to port the controller libraries. I first need to know what controller that is :)

d-a-v, 2015-05-19:
> Hello @maty87 and @MolarAmbiguity, do you know what is the LCD controller chip that the 4" LCD uses?
> 
> I'm trying to use this screen with an Arduino and will need to port the controller libraries. I first need to know what controller that is :)

They use the flexfb driver as stated in LCD-show.tar.gz at http://spotpear.com/learn/EN/raspberry-pi/Raspberry-Pi-LCD/Drive-the-LCD.html
Their /etc/modules file (using flexfb) works with fresh raspbian and original kernel. I did not dig further about the touch screen.
I tested the 2.4' lcd.

sazp96, 2015-05-19:
Wow, this library is awesome!! Thanks @d-a-v.

It supports a wide array of chip controllers. Any idea which one is used by SpotPear? I'm trying to get their 4" TFT LCD screen going: http://spotpear.com/index.php/spotpear-raspberry-pi-lcd/spotpear-raspberry-pi-4-inch-lcd

Thanks

closed 2015-09-06 by notro