notro/fbtft #269: WaveShare SpotPear 3.5" on RPI 2. Extremely slow

sackcreator54, 2015-05-03:
So, I got a hold of one of these things, and I read about them a bit and heard they're a bit infamous. I've got the thing displaying X, and I even got fbcp working. I can play things like GBA emulators, Quake 3, anything. However, I cannot do any of this at a manageable rate. Movies, games, emulators or what-have-you draw extremely slow. It's almost like there's extreme screen tearing, you can see screen tearing, and modifing FPS doesn't help at all. It just seems like it's always drawing at like 10 FPS.
Here's my modprobes:

sudo modprobe flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3
sudo modprobe fbtft_device name=flexfb rotate=90 speed=48000000 gpios=reset:25,dc:24 fps=60

If I go above 48000000 on speed, it no longer works and just displays a white screen. Is this just my screen or fbtft?

sackcreator54, 2015-05-03:
Here's a video of me demonstrating this in Open Arena:
https://www.youtube.com/watch?v=4y-cpS4BOSQ

notro, 2015-07-07:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2015-07-07 by notro

dan200, 2015-10-29:
I also experienced this with a similar setup. Some reading suggested that lack of DMA support may be a cause?

billair, 2016-04-09:
Same here, Im having trouble and cannot get over better config in /boot/config.txt then:
dtoverlay=waveshare35a,speed=32000000,fps=60
even that mine OpenArena is slightly faster  ...

ian57, 2016-05-03:
Hi same issue here with a banggood waveshare 3.5 TFT screen. after speed=27000000, the image colours are degrated, and after 48000000 just a white screen