notro/fbtft #270: fb_pcd8544: probe of spi0.0 failed with error -12

macmeck, 2015-05-04:
Hi,

I'm trying to use the Nokia 3310/5110 display on a MinnowBoard Max with Ubuntu 15.04. With either compiling fbtft modules for the 3.19.0 stock kernel or using 4.0.1 kernel with fbtft modules from staging, after

modprobe fbtft_device name=nokia3310 gpios=reset:338,dc:339,led:340

I get the error

[  240.080298] fbtft_device:  SPI devices registered:
[  240.080322] fbtft_device:      spidev spi0.0 1000kHz 8 bits mode=0x00
[  240.080331] fbtft_device:  'fb' Platform devices registered:
[  240.080381] fbtft_device: Deleting spi0.0
[  240.080942] fbtft_device:  GPIOS used by 'nokia3310':
[  240.080957] fbtft_device:    'reset' = GPIO338
[  240.080966] fbtft_device:    'dc' = GPIO339
[  240.080975] fbtft_device:    'led' = GPIO340
[  240.080983] fbtft_device:  SPI devices registered:
[  240.080995] fbtft_device:      fb_pcd8544 spi0.0 400kHz 8 bits mode=0x00
[  240.108189] fb_pcd8544: probe of spi0.0 failed with error -12

Any idea what that can be? I cannot find any information on error -12.
SPI seems to be okay, before I modprobe fbtft_device, I have a /dev/spi0.0, after probing fails, /dev/spi0.0 is gone.

notro, 2015-05-05:
-12 == -ENOMEM
http://lxr.free-electrons.com/source/include/uapi/asm-generic/errno-base.h#L15

You can try without dma:
sudo modprobe fbtft dma=0
sudo modprobe fbtft_device ...

macmeck, 2015-05-05:
Genius! thanks a lot, works like a charm...

closed 2015-05-05 by macmeck