notro/fbtft #271: what's the deal with fb_[device name].c files?

Cupacoffee, 2015-05-06:
Hello Notro,

After quiet some time of trying different approaches to get a touchscreen running properly i've followed the development section in order to adapt the driver to my needs. It worked and so i was able to crank up the speed of the transaction, but it seems  it's not enought. My original idea was to reduce the color-depth  to 8-bit mode in order to reduce the mass of data.

I've checked the device-datasheet and found the section within the "fb_ra8875.c", so i changed the section and compiled again. Aperrently my pi ignores all changes i do in the "fb_ra8875.c", contrary to that changing the "fbtft_device.c" works just fine. Even if i change code in a way that is compilable but simply wrong the screen does not seem to mind and keeps working normally. (It should probably not ignore wrong border-parameters?)

I'm using the following steps (from a Raspberry Pi B+, for the ER-TFT-M070-5 with RA8875 - Controller and prepaired kernel sources according to development-section):

1. Change *.c - file
2. cd ~/linux
3. make modules_prepare
4. make -C ~/linux SUBDIRS=drivers/video/fbtft modules
5. sudo cp ~/linux/drivers/video/fbtft/{fbtft_device.ko,fb_hello.ko} /lib/modules/$(uname -r)/kernel/drivers/video/fbtft/
6. sudo depmod
7. sudo reboot
8. sudo modprobe fbtft_device name=er_tftm070_5
9. fbcp & FRAMEBUFFER=/dev/fb0 startx

Help would be appreciated.

notro, 2015-05-06:
> Aperrently my pi ignores all changes i do in the "fb_ra8875.c"

You change fb_ra8875.c

> sudo cp ~/linux/drivers/video/fbtft/{fbtft_device.ko,fb_hello.ko} /lib/modules/$(uname -r)/kernel/drivers/video/fbtft/

But you don't copy fb_ra8875.ko ?

If you don't change any of the module info macros (MODULE_*) or make use of an exported function you didn't before, then you don't have to run depmod.

No need to reboot, just make sure you have unloaded the module, before reloading.

```sudo modprobe --first-time ...``` will inform you if the module is currently loaded.
modprobe does a silent no-op by default if the module is already loaded.

```sudo modprobe -r fbtft_device``` should unload all the fbtft modules. Verify with ```lsmod```

Cupacoffee, 2015-05-07:
>>sudo cp ~/linux/drivers/video/fbtft/{fbtft_device.ko,fb_hello.ko} /lib/modules/$(uname -r)/kernel/drivers/video/fbtft/

>But you don't copy fb_ra8875.ko ?

Sorry, i was in a hurry at the time so i copied the wrong line of the development section; I wanted to write the following line that copies all fb_* files.

sudo cp -v ~/linux/drivers/video/fbtft/*.ko /lib/modules/$(uname -r)/kernel/drivers/video/fbtft/
sudo depmod

But i'll check if i made the same mistake on the pi, also thanks for the other hints.

closed 2015-09-06 by notro