notro/fbtft #272: 2.4 LCD ili9325 (touch)

playmonkey, 2015-05-09:
Hi,

I bought a 2.4 ili9325 lcd (34pin header) from ebay and connected it in parallel (8bit) mode using pins DB10:DB17 and cannot get it to work other than display the background light. I don't have the datasheet and neither does the ebay vendor.

I have another ili9325 2.4 lcd display (different pin markings) that is working fine using custom pin settings with fbtft_device.

Could the 'non-working' lcd be configured for 16bit and not 8it? The LCD has R1-R8 resistors.  Resistors R4-R2 are empty and 0Ohm (really links) are connected to R1 and R3. 

Does anyone have experience with this LCD?

mrmonteith, 2015-05-19:
If you can give the link to the ebay item or a pic we can probably find the datasheet for it.

playmonkey, 2015-05-20:
Thanks for responding :) Here is an image of the LCD pin header
[image]

closed 2015-05-25 by playmonkey

NicolasRam, 2020-06-24:
Hi! Did you succeed to make it work in 8bit parallel mode ? I'm searching some information for mine. Thanks