notro/fbtft #278: Having Trouble with HY28A and  ADS7846 TS

medrapal, 2015-05-15:
Notro-

First off, I've been a user of your original FBTFT devices (way back when the HY28A was supported!).  Thanks for all of the great work here - you probably don't hear that often enough.

I've started updating to the new kernel/device tree versions, and I'm having trouble getting the touchscreen to work - the HY28A worked as soon as I added the device tree entry, but the touch screen won't work no matter how hard I try.  I've included the bottom of my /boot/config.txt below:

[pi1]
device_tree=bcm2708-rpi-b-plus.dtb
[pi2]
device_tree=bcm2709-rpi-2-b.dtb
[all]
dtparam=spi=on
dtoverlay=ads7846,cs=1,penirq=17,speed=500000,swapxy=1,pmax=255,xohms=60,verbose=3
dtoverlay=hy28a,rotate=270,speed=42000000,fps=20

dmesg indicates (I think) that both devices were loaded:

[    6.853238] bcm2708_spi 20204000.spi: master is unqueued, this is deprecated
[    7.215952] ads7846 spi0.1: touchscreen, irq 411
[    7.561456] input: ADS7846 Touchscreen as /devices/soc/20204000.spi/spi_master/spi0/spi0.1/input/input2
[    8.144102] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    8.182383] fb_ili9320: module is from the staging directory, the quality is unknown, you have been warned.
[    8.224613] fbtft_of_value: buswidth = 8
[    8.236334] random: nonblocking pool is initialized
[    8.338508] fbtft_of_value: debug = 0
[    8.453389] fbtft_of_value: rotate = 270
[    8.594037] fbtft_of_value: fps = 20
[    8.712146] fbtft_of_value: startbyte = 112
[    9.620938] graphics fb1: fb_ili9320 frame buffer, 320x240, 150 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 42 MHz
[    9.654984] bcm2708_spi 20204000.spi: chipselect 1 already in use
[    9.668357] spi_master spi0: spi_device register error /soc/spi@7e204000/ads7846@1
[    9.683107] bcm2708_spi 20204000.spi: SPI Controller at 0x20204000 (irq 80)

Any ideas?  I've swapped the loading of the two device tree entries and the HY28A framebuffer always works, and the touchscreen always fails.

notro, 2015-05-16:
The hy28a overlay already has a touch device: https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/boot/dts/hy28a-overlay.dts
So when the device from the ads7846 overlay is to be added, the SPI chip select is already taken.

evtest is a good tool for testing your touch setup.

You don't have to set device_tree= either, the firmware sets a default for you.

medrapal, 2015-05-21:
Just got a chance to try what you suggested.  I still can't get it to
work.  Anything else to try?

*/boot/config.txt now has two active lines:*

dtparam=spi=on

dtoverlay=hy28a,rotate=270,penirq=17,swapxy=1,pmax=255,xohms=60,debug=3

*dmesg seems to report that everything loaded fine:*

[    6.982884] ads7846 spi0.1: touchscreen, irq 411

[    7.249937] input: ADS7846 Touchscreen as
/devices/soc/20204000.spi/spi_master/spi0/spi0.1/input/input0

[    7.799149] fbtft: module is from the staging directory, the quality is
unknown, you have been warned.

[    7.839546] fb_ili9320: module is from the staging directory, the
quality is unknown, you have been warned.

[    7.884367] fbtft_of_value: buswidth = 8

[    7.963643] fbtft_of_value: debug = 3

[    8.070233] fbtft_of_value: rotate = 270

[    8.203627] fbtft_of_value: fps = 50

[    8.213399] fbtft_of_value: startbyte = 112

[    8.443909] fb_ili9320 spi0.0: fbtft_gamma_parse_str() str=

[    8.556411] fb_ili9320 spi0.0: 07 07 6 0 0 0 5 5 4 0

[    8.556411] 07 08 4 7 5 1 2 0 7 7

[    8.663821] fb_ili9320 spi0.0: fbtft_request_one_gpio: 'reset-gpios' =
GPIO25

[    8.694395] fb_ili9320 spi0.0: fbtft_request_one_gpio: 'led-gpios' =
GPIO18

[    8.707906] fb_ili9320 spi0.0: fbtft_verify_gpios()

[    8.719051] fb_ili9320 spi0.0: init_display()

[    8.729501] fb_ili9320 spi0.0: fbtft_reset()

[    8.860003] fb_ili9320 spi0.0: Device code: 0x9320

[    9.225576] fb_ili9320 spi0.0: set_var()

[    9.242068] random: nonblocking pool is initialized

[    9.318299] fb_ili9320 spi0.0: Display update: 1818 kB/s (82.489 ms),
fps=0 (0.000 ms)

[    9.332507] fb_ili9320 spi0.0: set_gamma()

[    9.343743] fb_ili9320 spi0.0: fbtft_register_backlight()

[    9.356930] graphics fb1: fb_ili9320 frame buffer, 320x240, 150 KiB
video memory, 4 KiB DMA buffer memory, fps=50, spi0.0 at 32 MHz

[    9.375116] fb_ili9320 spi0.0: fbtft_backlight_update_status:
polarity=1, power=0, fb_blank=0

[    9.390089] bcm2708_spi 20204000.spi: SPI Controller at 0x20204000 (irq
80)

*evtest shows no activity when I try the touch screen:*

*pi@dz-pi* *~ $* sudo evtest /dev/input/event0

Input driver version is 1.0.1

Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0

Input device name: "ADS7846 Touchscreen"

Supported events:

  Event type 0 (EV_SYN)

  Event type 1 (EV_KEY)

    Event code 330 (BTN_TOUCH)

  Event type 3 (EV_ABS)

    Event code 0 (ABS_X)

      Value      0

      Min        0

      Max     4095

    Event code 1 (ABS_Y)

      Value      0

      Min        0

      Max     4095

    Event code 24 (ABS_PRESSURE)

      Value      0

      Min        0

      Max      255

Properties:

Testing ... (interrupt to exit)

^C*pi@dz-pi* *~ $*

notro, 2015-05-23:
Verify that you get interrupts from the touch controller when you touch the display: cat /proc/interrupts

medrapal, 2015-05-25:
Looks like it should be (interrupt 411):

pi@dz-pi ~ $ sudo cat /proc/interrupts 
           CPU0       
  3:       6424   ARMCTRL   3  BCM2708 Timer Tick
 16:          0   ARMCTRL  16  bcm2708_fb dma
 24:         87   ARMCTRL  24  DMA IRQ
 25:       1338   ARMCTRL  25  DMA IRQ
 32:      52946   ARMCTRL  32  dwc_otg, dwc_otg_pcd, dwc_otg_hcd:usb1
 49:          0   ARMCTRL  49  20200000.gpio:bank0
 50:          0   ARMCTRL  50  20200000.gpio:bank1
 65:         10   ARMCTRL  65  ARM Mailbox IRQ
 66:          2   ARMCTRL  66  VCHIQ doorbell
 75:          1   ARMCTRL  75
 77:         38   ARMCTRL  77  20204000.spi
 80:        117   ARMCTRL  80  20204000.spi
 83:          7   ARMCTRL  83  uart-pl011
 84:       4646   ARMCTRL  84  mmc0
411:          0  pinctrl-bcm2835  17  ads7846
FIQ:              usb_fiq
Err:          0
pi@dz-pi ~ $

notro, 2015-05-25:
Correct, and it's not generating any interrupts.
The ads7846 driver does not verify that it can talk to the touch controller during probe, so it will load even if there is no controller connected.

> dtoverlay=hy28a,rotate=270,penirq=17,swapxy=1,pmax=255,xohms=60,debug=3

Looking closer I see that you use parameters that are not supported by the overlay. I don't know what the firmware does with unsupported parameters. Try using the defaults and see what happens: ```dtoverlay=hy28a```

Doc: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L201
Source: https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/boot/dts/hy28a-overlay.dts

medrapal, 2015-05-26:
Using the defaults, the same failure with the touch screen - no
interrupts.  The display works fine.

I went back to the previous kernel/module configuration (changed out the SD
card) and the touch screen works fine.  So it's not a wiring issue (it's
wired as the default from the wiki page).

medrapal, 2015-05-26:
Sorry - just got it to fail with the old kernel/modules.  Looks like i must
have an intermittent on the ts_irq line somewhere.

medrapal, 2015-05-26:
Confirmed - flakey display module.  I changed it out with another one and
it works fine.  Sorry for all of the headaches.

closed 2015-09-06 by notro