notro/fbtft #279: JBtek Latest Version 3.5" TFT LCD 480x320 screen rotation

thenewuser, 2015-05-18:
I try to rotate the screen orientation by editing the file /etc/modules (parameter "rotate"), but the result that I get is this: https://cloud.githubusercontent.com/assets/12488067/7674206/27832a4c-fcec-11e4-9217-b7f54441e4ab.png

is there a problem with the driver for this display?

notro, 2015-05-19:
If it uses the flexfb driver, rotation does not work automatically. You have to change one of the register values in the init sequence as well.

mwilliams03, 2015-05-19:
The register Notro is talking about should be 0x36.
Try this;
 
Change this;
flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3

To this;
flexfb  width=480  height=320  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x48,-1,0x11,-1,0x29,-3

thenewuser, 2015-05-20:
notro/mwilliams03 thank for your help,
I change the "init" value and the display work properly with the flexfb driver.

But I realize that the "init" value is the same used by the fb_ili9486 driver, the only difference i see is that  the regwidth value is 8 for the fb_ili986 but 16 in my /etc/modules file.

closed 2015-09-06 by notro