notro/fbtft #280: Dual hy28a overlay

CaptainStouf, 2015-05-19:
Hi,
I'm trying to get 2 jy28a displays to work in the same time in framebuffer. I want to clone the displays, I may use fbcp once the displays are working.

I'm having some troubles getting the dual displays overlay to work.
Here is mine :
```
/*
 * Device Tree overlay for HY28A display shield by Texy
 *
 */

/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

	fragment@0 {
          target = <&spi0>;
          __overlay__ {
             status = "okay";

             spidev@0{
                status = "disabled";
             };

             spidev@1{
                status = "disabled";
             };
          };
       };

	fragment@1 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			hy28a@0{
				compatible = "ilitek,ili9320";
				reg = <0>;
				pinctrl-names = "default";
				pinctrl-0 = <&hy28a_pins>;

				spi-max-frequency = <32000000>;
				spi-cpol;
				spi-cpha;
				rotate = <270>;
				bgr;
				fps = <50>;
				buswidth = <8>;
				startbyte = <0x70>;
				reset-gpios = <&gpio 25 0>;
				led-gpios = <&gpio 18 1>;
				debug = <0>;
			};
			
			hy28a@1{
				compatible = "ilitek,ili9320";
				reg = <1>;
				pinctrl-names = "default";
				pinctrl-0 = <&hy28a_pins>;

				spi-max-frequency = <32000000>;
				spi-cpol;
				spi-cpha;
				rotate = <270>;
				bgr;
				fps = <50>;
				buswidth = <8>;
				startbyte = <0x70>;
				reset-gpios = <&gpio 24 0>;
				led-gpios = <&gpio 18 1>;
				debug = <0>;
			};
		};
	};
};
```
I got the direction with this post : https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=109065
With this ovelay, I could see both fragments in /proc/device-tree/soc/spi@7e204000/ but none is working (displays stay white).
- If I declare with "hy28a: hy28a@0{" and "hy28a: hy28a@1{", dts compilation fails because of duplicate labels.
- If I declare with "hy28a0: hy28a@0{" and "hy28a1: hy28a@1{", dts compilation doesn't fail but displays are not initialized.
- If I declare with "hy28a@0{" and "hy28a@1{", dts compilation doesn't fail but displays are not initialized.
- If I put the code in separate overlays, declaring them with "hy28a: hy28a@0{" and "hy28a: hy28a@1{", I could see both fragments but only one display is working (the last one). They are working separately when I load only one of them, so I assume my wiring is ok.

Could you see what I'm doing wrong ?
Thank you

notro, 2015-05-19:
You can drop these, since you don't have the pinctrl node (not really necessary):
```
                pinctrl-names = "default";
                pinctrl-0 = <&hy28a_pins>;
```
You have the same gpio number for "led-gpios", which will make the last one fail to load. ```dmesg``` will show you the error message.

CaptainStouf, 2015-05-19:
Wow, you're very fast !
I removed the lines you pointed, and added some debug in /boot/config.txt. So I could verify what you said about led-gpios. I thought it was used only for pwm, so I assumed the same gpio could work on both screens. I was wrong.
So I added another gpio for the second display led-gpio, and now both displays seem to work : I have fb0, fb1 and fb2 in /dev, I have the console display on one, and the other is black (so I assume it is initialized).
I'm now installing fbcp, and I will let you know.

Here is my updated code : 
```
/*
 * Device Tree overlay for HY28A display shield by Texy
 *
 */

/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

	fragment@0 {
          target = <&spi0>;
          __overlay__ {
             status = "okay";

             spidev@0{
                status = "disabled";
             };

             spidev@1{
                status = "disabled";
             };
          };
       };

	fragment@1 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			hy28a@0{
				compatible = "ilitek,ili9320";
				reg = <0>;

				spi-max-frequency = <32000000>;
				spi-cpol;
				spi-cpha;
				rotate = <270>;
				bgr;
				fps = <50>;
				buswidth = <8>;
				startbyte = <0x70>;
				reset-gpios = <&gpio 25 0>;
				led-gpios = <&gpio 18 1>;
				debug = <0>;
			};
			
			hy28a@1{
				compatible = "ilitek,ili9320";
				reg = <1>;

				spi-max-frequency = <32000000>;
				spi-cpol;
				spi-cpha;
				rotate = <270>;
				bgr;
				fps = <50>;
				buswidth = <8>;
				startbyte = <0x70>;
				reset-gpios = <&gpio 24 0>;
				led-gpios = <&gpio 23 1>;
				debug = <0>;
			};
		};
	};
};
```
Many thanks notro, you are of great help !

CaptainStouf, 2015-05-19:
I was wrong, one display is not working (the first one, on cs0). I get this dmesg : 
 fb_ili9320 spi0.0: cannot register backlight device (-17)
However it shows in /dev and in /proc/device-tree/soc/spi@7e204000/

Do you see what I could do ?

notro, 2015-05-19:
-17 is -EEXIST, so the backlight devices might end up with the same name in sysfs, I don't know.
'led-gpios' is optional, so you can drop one or both.

CaptainStouf, 2015-05-19:
Ok, so I removed led-gpios from the second display, and it is now working ! Thank you !

Now, I have to find a way to clone the displays. 
I see fbcp could only clone the main display (fb0) so I couldn't use it to clone fb1 on fb2. I'm currently looking into the sources to see if I could mod it (I don't think vc_dispmanx_display_open() allows fb1 as primary display).
Otherwise I may try with xorg.conf...

Do you have a better idea ? Ideally both displays should be synced.
But many thanks for your help !

notro, 2015-05-19:
Maybe xrandr can help (haven't tried it): http://linuxconfig.org/enable-multiple-clone-displays-to-vga-interface-projector-tv

CaptainStouf, 2015-05-19:
Oh yes, this is another possibility. This link will be useful in the future, thank you.
Right now, I got it to work with xorg and this /usr/share/X11/xorg.conf.d/99-fbturbo.conf : 
```
Section "Device"
        Identifier      "fbdev1"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb1"
EndSection

Section "Device"
        Identifier      "fbdev2"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb2"
EndSection

Section "Screen"
        Identifier      "lcd1"
        Device          "fbdev1"
        Monitor         "monitor1"
EndSection

Section "Screen"
        Identifier      "lcd2"
        Device          "fbdev2"
        Monitor         "monitor2"
EndSection

Section "ServerLayout"
        Identifier      "Default Layout"
        Screen          0 "lcd1"
        Screen          1 "lcd2"
        Option          "Xinerama" "on"
EndSection
```
You could replace 
Screen          1 "lcd2"
with
Screen          1 "lcd2" RightOf "lcd1"
for extension mode.

By the way, what is the difference between fbdev and fbturbo ? According to the sources, on RPi, the main advantage of fbturbo seems "Hardware accelerated window moving/scrolling on Raspberry Pi (using the BCM2835 DMA Controller)". Other stuff, like 3D acceleration seems to be reserved to AllWinner SoCs.

closed 2015-05-20 by CaptainStouf

HagenKnight, 2017-03-27:
Hello Captain

I'm working on a project very similar than yours. I'm working with two ili9341 chip display and I need to use both at the same time. Individually I can use each one and I configure them this way:

```
$ sudo nano /etc/modules-load.d/fbtft.conf
 	spi-bcm2835
 	fbtft_device
```
	
```
$ sudo nano /etc/modprobe.d/fbtft.conf
	options fbtft_device name=tm022hdh26 gpios=reset:25,dc:24,led:12 rotate=0 speed = 8000000 fps=60
	#options fbtft_device name=tm022hdh26 gpios=reset:23,dc:18,led:13 cs=1 rotate=0 speed = 8000000 fps=60
```

```
$	sudo nano /etc/modules
	fbtft_device name=tm022hdh26 gpios=reset:25,dc:24,led:12 rotate=0 speed=80000000 fps=60
	# fbtft_device name=tm022hdh26 gpios=reset:23,dc:18,led:13 cs=1 rotate=0 speed = 8000000 fps=60
```

I try using the second display individually, and it works! (the used configuration is the second line, commented one).
	
After reboot, I see in dmesg the follow lines (just for second display through SPI1, the commented lines on my configuration):

```
[  +0,002042] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[  +0,006316] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[  +0,001041] spidev spi0.0: spidev spi0.0 500kHz 8 bits mode=0x00
[  +0,000010] spidev spi0.1: spidev spi0.1 500kHz 8 bits mode=0x00
[  +0,000039] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
[  +0,000043] spidev spi0.1: Deleting spi0.1
[  +0,000741] fbtft_device: GPIOS used by 'tm022hdh26':
[  +0,000005] fbtft_device: 'reset' = GPIO23
[  +0,000003] fbtft_device: 'dc' = GPIO18
[  +0,000003] fbtft_device: 'led' = GPIO13
[  +0,000010] spidev spi0.0: spidev spi0.0 500kHz 8 bits mode=0x00
[  +0,000010] spi spi0.1: fb_ili9341 spi0.1 80000kHz 8 bits mode=0x00
[  +0,024909] fuse init (API version 7.23)
[  +0,021477] i2c /dev entries driver
[  +0,065268] systemd[1]: Reached target Slices.
[  +0,017458] systemd[1]: Mounted Debug File System.
[  +0,015334] systemd[1]: Mounted POSIX Message Queue File System.
[  +0,016389] systemd[1]: Started Increase datagram queue length.
[  +0,021702] systemd[1]: Started Create list of required static device nodes for the current kernel.
[  +0,036536] systemd[1]: systemd-modules-load.service: main process exited, code=exited, status=1/FAILURE
[  +0,025874] systemd[1]: Failed to start Load Kernel Modules.
[  +0,007424] systemd[1]: Unit systemd-modules-load.service entered failed state.
[  +0,016520] systemd[1]: Started File System Check on Root Device.
[  +0,016815] systemd[1]: Started udev Coldplug all Devices.
[  +0,137432] systemd[1]: Starting Apply Kernel Variables...
[  +0,055177] systemd[1]: Mounting Configuration File System...
[  +0,017675] systemd[1]: Mounting FUSE Control File System...
[  +0,011051] systemd[1]: Starting Create Static Device Nodes in /dev...
[  +0,011002] systemd[1]: Starting Syslog Socket.
[  +0,008260] systemd[1]: Listening on Syslog Socket.
[  +0,000164] systemd[1]: Starting Journal Service...
[  +0,017701] systemd[1]: Started Journal Service.
[  +0,020141] random: nonblocking pool is initialized
[  +0,181493] systemd-udevd[133]: starting version 215
[  +0,421390] fb_ili9341: module is from the staging directory, the quality is unknown, you have been warned.
[  +0,308389] graphics fb1: fb_ili9341 frame buffer, 240x320, 150 KiB video memory, 16 KiB DMA buffer memory, fps=100, spi0.1 at 80 MHz
[  +0,056189] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
```

In the exposed case, I don't need to call a dtoverlay. 

Well, I try to follow your example, writing an overlay like this:

```
/*
 * Device Tree overlay for 2x ili9341 2.2"
 *
 */

/dts-v1/;
/plugin/;

/ {
   compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

   fragment@0 {
      target = <&spi0>;
      __overlay__ {
         status = "okay";

         spidev@0{
            status = "disabled";
         };

         spidev@1{
            status = "disabled";
         };
      };
   };

   fragment@1 {
      target = <&spi0>;
      __overlay__ {
         /* needed to avoid dtc warning */
         #address-cells = <1>;
         #size-cells = <0>;

         cat9341@0{
            compatible = "ilitek,ili9341";
            reg = <0>;

            spi-max-frequency = <32000000>;
            rotate = <0>;
            buswidth = <8>;
            reset-gpios = <&gpio 25 0>;
            dc-gpios = <&gpio 24 0>;
			led-gpios = <&gpio 12 1>;
            debug = <0>;
         };

         cat9341@1{
            compatible = "ilitek,ili9341";
            reg = <1>;

            spi-max-frequency = <32000000>;
            rotate = <0>;
            buswidth = <8>;
            reset-gpios = <&gpio 23 0>;
            dc-gpios = <&gpio 18 0>;
			led-gpios = <&gpio 12 1>;
            debug = <0>;
         };
      };
   };
};
```

It was compiled as dual-ili9341.dtbo and moved to /boot/overlays/
On config.txt, the dtoverlay was:
`	dtoverlay=dual-ili9341.`
	
I had no success with overlay, It's like Raspbian ignore them and I just see two framebuffer as follows:
`/dev/fb0  /dev/fb1`

Please I need to know:
* what else it's needed for displays working?
* How did you do to have three FB (two of them for TFTDisplays)?

Thanks in advance.
Giovanni García

notro, 2017-03-27:
Both fragments use gpio 12 for led. Probing of the second instance will fail.

HagenKnight, 2017-03-27:
Hello Notro.
I checked and corrected the second led gpio, changed to 13 (edited on this post). I compiled again and move it to /boot/overlays.
Now, the previous setup that you recommend on notro wiki must go? I mean, now how I call the overlay also writing on config.txt? How do I change the files **/etc/modules-load.d/fbtft.conf** and **/etc/modprobe.d/fbtft.conf**?

Thanks,
Giovanni García

notro, 2017-03-27:
There are 2 ways to add "fbtft" devices:
- Using the fbtft_device module, but only one device can be added
- Using DT overlay, multiple device can be added

If you use an overlay, there's no need for setting the .conf files.
But most platforms doesn't support DT overlays, so they need the fbtft_device approach.

HagenKnight, 2017-03-30:
Finally I got running my two ILI9341 displays, getting two aditional framebuffers, /dev/fb1 and /dev/fb2. 
I have to start again the process with a fresh Raspbian image.  And I want to share my experience:

1. As a new image, you must enable SPI on raspi-config.

2. My overlay is the code below:
````
/*
* Device Tree overlay for 2x ili9341 2.2"
*
*/

/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";
	
	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			status = "okay";
			
			spidev@0{
				status = "disabled";
			};
			
			spidev@1{
				status = "disabled";
			};
		};
	};
	
	fragment@1 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;
			
			cat9341@0{
				compatible = "ilitek,ili9341";
				reg = <0>;
				
				spi-max-frequency = <32000000>;
				rotate = <0>;
				bgr;
				fps = <50>;
				buswidth = <8>;
				reset-gpios = <&gpio 25 0>;
				dc-gpios = <&gpio 24 0>;
				/*led-gpios = <&gpio 12 1>;*/
				debug = <0>;
			};
			
			cat9341@1{
				compatible = "ilitek,ili9341";
				reg = <1>;
				
				spi-max-frequency = <32000000>;
				rotate = <0>;
				bgr;
				fps = <50>;
				buswidth = <8>;
				reset-gpios = <&gpio 23 0>;
				dc-gpios = <&gpio 18 0>;
				led-gpios = <&gpio 13 1>;
				debug = <0>;
			};
		};
	};
};
````

It's very important use spi-max-frequency at 32MHz: on testing I was using 8MHz and the mplayer was very slow; I thought it was kernel issue or another thing related to SPI bus. The option bgr is also important. At first time I didn't put on the overlay, supposed fbtft uses another by default, and the colors were wrong! Everything was blue. FInally I got it and fixed the display behaviour. 
I have to comment the led-gpio for second display backlight. fbtft only register one backlight; to turn on the second backlight, I have to use the **gpio** command. 

3. Save the overlay file with a nice name and compile it:
`sudo dtc -@ -I dts -O dtb -o /boot/overlays/dual-ili9341.dtbo ili9341.dts`

4. Register the overlay calling in config.txt
`sudo nano /boot/config.txt`

5. Find the last line of your file and write:
`dtoverlay=dual-ili9341`

6. Disable the X server boot and run from console (I need in that way on my project)

7. sudo reboot!!!

8. Test each display with mplayer (for videos) or fbi (static images) and enjoy it!
[image]

Thanks to @notro to correct me at first time and fast answer.

Now the challenge is to play videos at the same time for each framebuffer, any idea @notro ?
I'm trying with GNU parallel and no success at this moment.
Thanks again.

x3n0n-dev, 2017-09-16:
HI guys,
i have some problems with my dual ili9341 configuration.
i downloaded latest stretch lite version from raspian website,
created  a dts file copying one of the solution provided for a similar issue
 
```
/*
* Device Tree overlay for 2x ili9341 2.2"
*
*/
 
/dts-v1/;
/plugin/;
 
/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";
	
	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			status = "okay";
			
			spidev@0{
				status = "disabled";
			};
			
			spidev@1{
				status = "disabled";
			};
		};
	};
	
	fragment@1 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;
			
			cat9341@0{
				compatible = "ilitek,ili9341";
				reg = <0>;
				
				spi-max-frequency = <32000000>;
				rotate = <0>;
				bgr;
				fps = <50>;
				buswidth = <8>;
				reset-gpios = <&gpio 25 0>;
				dc-gpios = <&gpio 24 0>;
				led-gpios = <&gpio 12 1>;
				debug = <0>;
			};
			
			cat9341@1{
				compatible = "ilitek,ili9341";
				reg = <1>;
				
				spi-max-frequency = <32000000>;
				rotate = <0>;
				bgr;
				fps = <50>;
				buswidth = <8>;
				reset-gpios = <&gpio 23 0>;
				dc-gpios = <&gpio 18 0>;
				led-gpios = <&gpio 13 1>;
				debug = <0>;
			};
		};
	};
};`
```
compiled with 

`sudo dtc -@ -I dts -O dtb -o /boot/overlays/dual-ili9341.dtbo ili9341.dts
`
and i modified /boot/config.txt  adding ad the very bottom

`dtoverlay=dual-ili9341`

i'm sure the wiring is correct : 
3.3v
miso 
mosi
sclk

are in common

c0 for one screen and c01 for the other 

25 24 12

23 18 13

i've been trying for 3 days now and i simply cannot make them work.

i suppose it's something linked to led-gpios .

when i try executing dmesg this is what i get

```
[    4.589928] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    4.604245] fb_ili9341: module is from the staging directory, the quality is unknown, you have been warned.
[    4.605140] fbtft_of_value: buswidth = 8
[    4.605156] fbtft_of_value: debug = 0
[    4.605162] fbtft_of_value: rotate = 0
[    4.605174] fbtft_of_value: fps = 50
[    4.614365] fb_ili9341: module is from the staging directory, the quality is unknown, you have been warned.
[    4.968852] graphics fb1: fb_ili9341 frame buffer, 240x320, 150 KiB video memory, 16 KiB DMA buffer memory, fps=50, spi0.1 at 32 MHz
[    4.969001] fbtft_of_value: buswidth = 8
[    4.969009] fbtft_of_value: debug = 0
[    4.969015] fbtft_of_value: rotate = 0
[    4.969021] fbtft_of_value: fps = 50

...

   5.288309] fb_ili9341 spi0.0: cannot register backlight device (-17)
[    5.288715] graphics fb2: fb_ili9341 frame buffer, 240x320, 150 KiB video me$

```

and that's probably the point . what can i do to solve the problem ? 

i followed what suggested in https://github.com/notro/fbtft/issues/280

but 

i found something on https://github.com/notro/fbtft/wiki/How-it-works
but my linux knowledge is not deep enough to understand what i should do to solve this.

any hint?

ty for the support and sorry for my terrible english which is not my main language

HagenKnight, 2017-09-17:
Dear @x3n0n-dev 
When I began to compile my overlay, I must disable one of the LED pin. The another LED pin I must turn on by software (using gpio and pwm mode).  If you need to start the RaspPi with TFT display, you must edit 99-fbturbo.conf file:

`sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf`

The current file looks like this:

```
# This is a minimal sample config file, which can be copied to
# /etc/X11/xorg.conf in order to make the Xorg server pick up
# and load xf86-video-fbturbo driver installed in the system.
#
# When troubleshooting, check /var/log/Xorg.0.log for the debugging
# output and error messages.
#
# Run "man fbturbo" to get additional information about the extra
# configuration options for tuning the driver.

############### ORIGINAL SECTION
#Section "Device"
#        Identifier      "Allwinner A10/A13 FBDEV"
#        Driver          "fbturbo"
#        Option          "fbdev" "/dev/fb0"
#        Option          "SwapbuffersWait" "true"
#EndSection

#### DEFAULT HDMI
Section "Device"
	Identifier	"HDMI_Main"
	Driver		"fbturbo"
	Option		"fbdev" "/dev/fb0"
	Option		"SwapbuffersWait" "true"
EndSection

Section "Device"
        Identifier      "my_tft1"
        Driver          "fbdev
        Option          "fbdev" "/dev/fb1"
EndSection

Section "Device"
	Identifier 	"my_tft2"
	Driver 		"fbdev"
	Option 		"fbdev" "/dev/fb2"
EndSection
	
Section "Screen"
	Identifier      "HDMI"
	Device          "HDMI_Main"
	Monitor         "Monitor TV"
EndSection
	
Section "Screen"
	Identifier	"TFT_RIGHT"
	Device		"my_tft1"
	Monitor		"Monitor TFT"
EndSection

Section "Screen"
	Identifier      "TFT_LEFT"
	Device          "my_tft2"
	Monitor         "Monitor TFT"
EndSection

	
Section "ServerLayout"
	Identifier      "Default Layout"
#	Screen		0 "HDMI"
	Screen		0 "TFT_RIGHT"
	Screen		1 "TFT_LEFT" RightOf "TFT_RIGHT"
	Screen		2 "HDMI" RightOf "TFT_LEFT"
	Option 		"Clone" "off"
	Option          "Xinerama" "on"
EndSection
```

I have to run my project on desktop GUI and launch the application from OS start.

notro, 2017-09-17:
-17 is -EEXIST, so something exists the second time fbtft tries to register a backlight.

The reason seems to be that the backlight name is derived from the driver name instead of the device name:
```c
void fbtft_register_backlight(struct fbtft_par *par)
{

	bd = backlight_device_register(dev_driver_string(par->info->device),
				       par->info->device, par,
				       &fbtft_bl_ops, &bl_props);
}
```
Ref: http://elixir.free-electrons.com/linux/latest/ident/fbtft_register_backlight

Maybe it would have been better to use ```dev_name(dev)``` instead.
No one really maintains fbtft, so it probably won't be fixed.

x3n0n-dev, 2017-09-17:
Wow such fast answers. I'll try as soon as I'll be home tomorrow. Amazing . Ty all

x3n0n-dev, 2017-09-17:
Anyway I need to use it console from python with pygame and pygameui so no x11 but still I suppose I can enable by code while launching the code

x3n0n-dev, 2017-09-18:
i tried with

gpio -g mode 12 pwm
gpio -g pwm 12 300

screen is now turned on but still con2fbmap 1 2 doesn't work at all so it seems i cannot comunicate with the dev/fb created

CaptainStouf, 2017-09-18:
When I did it, I used the same pwm pin on both display (this is the display backlight led and there's only one hw pwn pin on the pi, so it doesn't matter if is the same one).
I will look if I could find the overlay and module.conf I used to make it work.

x3n0n-dev, 2017-09-18:
i tried with 
`led-gpios = <&gpio 13 1>;`
on both configuration on overlay

but one is working while the other is full white.

CaptainStouf, 2017-09-20:
It's been a long time since I did it... I think these are the files I was using :
the dts :

```
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

	fragment@0 {
          target = <&spi0>;
          __overlay__ {
             status = "okay";

             spidev@0{
                status = "disabled";
             };

             spidev@1{
                status = "disabled";
             };
          };
       };

	fragment@1 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			hy28a@0{
				compatible = "ilitek,ili9320";
				reg = <0>;

				spi-max-frequency = <32000000>;
				spi-cpol;
				spi-cpha;
				rotate = <180>;
				bgr;
				fps = <50>;
				buswidth = <8>;
				startbyte = <0x70>;
				reset-gpios = <&gpio 25 0>;
				led-gpios = <&gpio 18 1>;
				debug = <0>;
			};
			
			hy28a@1{
				compatible = "ilitek,ili9320";
				reg = <1>;

				spi-max-frequency = <32000000>;
				spi-cpol;
				spi-cpha;
				rotate = <180>;
				bgr;
				fps = <50>;
				buswidth = <8>;
				startbyte = <0x70>;
				reset-gpios = <&gpio 24 0>;
				/*led-gpios = <&gpio 23 1>;*/
				debug = <0>;
			};
		};
	};
};
```

the 99-fbturbo.conf :

```
Section "Device"
        Identifier      "fbdev0"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb0"
        Option          "SwapbuffersWait" "false"
EndSection

Section "Device"
        Identifier      "fbdev1"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb1"
        Option          "SwapbuffersWait" "false"
EndSection

Section "Device"
        Identifier      "fbdev2"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb2"
        Option          "SwapbuffersWait" "false"
EndSection

Section "Screen"
        Identifier      "lcd1"
        Device          "fbdev1"
        Monitor         "monitor1"
        SubSection "Display"
        Depth 24
        Virtual 800 600
        Modes "800x600@60"
        EndSubSection
EndSection

Section "Screen"
        Identifier      "lcd2"
        Device          "fbdev2"
        Monitor         "monitor2"
        SubSection "Display"
        Depth 24
        Virtual 800 600
        Modes "800x600@60"
        EndSubSection
EndSection

Section "ServerLayout"
        Identifier      "Default Layout"
        Screen          0 "lcd1"
        Screen          1 "lcd2"
        Option          "Xinerama" "on"
EndSection
```
I wrote a blog article about it : [http://hardware-libre.fr/2015/05/openvar-le-casque-vr-do-it-yourself-v1-etape-1/](url)
and also another, with a camera pass-through script : [http://hardware-libre.fr/2015/05/openvar-le-casque-vr-do-it-yourself-mark-i-camera-pass-through/](url)