notro/fbtft #282: Additional SPI slave with Sainsmart 3.2 in parallel mode and touchscreen

iomode, 2015-05-25:
Hi,

In my current setup I have connected a Sainsmart 3.2 to my Raspberry Pi using the parallel interface (modprobe fbtft_device name=sainsmart32 ) and the touchscreen using the SPI device tree overlay (ads7846-overlay.dts). 

I need to connect an additional SPI slave in my project but i cannot figure out how to do that.
When I load the ads7846 overlay the /dev/spidev0.x device links disappear. 
I need to get the /dev/spidev0.1 interface for my application.

I already tried this https://github.com/raspberrypi/linux/wiki/Device-Tree-on-ARCH_BCM2708#spi with no luck. 

Has anybody an idea how to use the ads7846 overlay/driver and spidev at the same time?

Thanks

This is my current configuration:
/boot/config.txt
```
dtdebug=on
dtoverlay=
dtparam=spi=on
dtoverlay=ads7846,cs=0,penirq=15,pmax=255,xohms=60
```
pi@raspberrypi ~ $ dmesg | grep spi
```
[    4.889197] bcm2708_spi 20204000.spi: master is unqueued, this is deprecated
[    5.011366] ads7846 spi0.0: touchscreen, irq 409
[    5.059906] input: ADS7846 Touchscreen as /devices/soc/20204000.spi/spi_master/spi0/spi0.0/input/input0
[    5.184204] bcm2708_spi 20204000.spi: SPI Controller at 0x20204000 (irq 80)
```
pi@raspberrypi ~ $ find /dev/ -name "*spi*"
```
/dev/input/by-path/platform-20204000.spi-event
```
pi@raspberrypi ~ $ find /sys/ -name "*spi*"
```
/sys/bus/spi
/sys/bus/spi/devices/spi0.0
/sys/bus/spi/drivers/stmpe-spi
/sys/bus/spi/drivers/ads7846/spi0.0
/sys/bus/spi/drivers/spidev
/sys/bus/platform/devices/20204000.spi
/sys/bus/platform/drivers/bcm2708_spi
/sys/bus/platform/drivers/bcm2708_spi/20204000.spi
/sys/devices/soc/20204000.spi
/sys/devices/soc/20204000.spi/spi_master
/sys/devices/soc/20204000.spi/spi_master/spi0
/sys/devices/soc/20204000.spi/spi_master/spi0/spi0.0
/sys/class/spi_master
/sys/class/spi_master/spi0
/sys/class/spidev
/sys/firmware/devicetree/base/soc/gpio/spi0_pins
/sys/firmware/devicetree/base/soc/spi@7e204000
/sys/firmware/devicetree/base/soc/spi@7e204000/spidev@0
/sys/firmware/devicetree/base/soc/spi@7e204000/spidev@0/spi-max-frequency
/sys/firmware/devicetree/base/soc/spi@7e204000/spidev@1
/sys/firmware/devicetree/base/soc/spi@7e204000/spidev@1/spi-max-frequency
/sys/firmware/devicetree/base/soc/spi@7e204000/ads7846@1/spi-max-frequency
/sys/firmware/devicetree/base/__symbols__/spi0
/sys/firmware/devicetree/base/__symbols__/spi0_pins
/sys/firmware/devicetree/base/__symbols__/clk_spi
/sys/firmware/devicetree/base/aliases/spi0
/sys/firmware/devicetree/base/__overrides__/spi
/sys/module/spi_bcm2708
/sys/module/ads7846/drivers/spi:ads7846
/sys/module/spidev
```
pi@raspberrypi ~ $ lsmod 
```
Module                  Size  Used by
snd_bcm2835            20818  0 
snd_pcm                90778  1 snd_bcm2835
snd_seq                61097  0 
snd_seq_device          7209  1 snd_seq
snd_timer              23007  2 snd_pcm,snd_seq
snd                    66325  5 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device
evdev                  11000  1 
joydev                  9766  0 
ads7846                10632  0 
spi_bcm2708             6018  0 
8192cu                569561  0 
uio_pdrv_genirq         3666  0 
uio                     9897  1 uio_pdrv_genirq
```

notro, 2015-05-25:
Create an overlay that enables spidev and load it after ads7846:
```
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			status = "okay";

			spidev@1{
				status = "okay";
			};
		};
	};
};
```
Or edit the source file: https://github.com/raspberrypi/linux/blob/rpi-3.18.y/arch/arm/boot/dts/ads7846-overlay.dts

https://www.raspberrypi.org/documentation/configuration/device-tree.md

iomode, 2015-05-26:
Thank you, it worked ;)
I compiled your code and loaded it after ads7846, /dev/spidev0.1 is back!
I am currently not able to test the interface but it looks good so far.
Never worked with device tree overlays before but I think it is time to have a look on it.

iomode, 2015-05-27:
Now I can verify that the spidev0.1 interface works without problems.
spidev handles chip select switching between the slaves. (prevents race condition).

closed 2015-05-27 by iomode