notro/fbtft #283: How to activate stagging driver for display?

abhi-84, 2015-05-25:
Sir
How to activate the driver for piTFT2.8-resistive display? Since now the driver is already in stagging mode in main kernel, i hope all i need to do is activate it some how and make it usable?
Can you please specify me how to do so? I am using Raspberrypi Model B

Thank you

notro, 2015-05-25:
If you use the foundation kernel: dtoverlay=pitft28-resistive
https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L325

abhi-84, 2015-05-26:
Where i need to make this change?in which file?do i need to recompile the
kernel after making this change?do i need to modify the cmdline.txt file
also?please elaborate

notro, 2015-05-26:
https://github.com/notro/fbtft/wiki#install

closed 2015-09-06 by notro