notro/fbtft #285: Seeed Studio 2.8" TFT LCD v2.0 (ILI9341) - Mirrored/Flipped Image

buzzdennis, 2015-06-03:
Hi notro,
Amazing work you've done here. Let me start by saying thank you. I've been lurking the issues pages for the last few days and I've gotten pretty far in making the Seeed Studio TFT Shield work with the pi. This little sucker was a clearance item at many of the Radio Shacks that shut down so I get the sense a few people have picked these up with similar hopes of getting it running on the pi.

One of the main issues that gave me a white screen for the longest time was that the MOSI, MISO and SCK lines are labeled poorly in the Seeed Studio documentation. These lines are not on any of the Arduino Digital Header Pins like their documentation says they are. They are actually on the ICSP header. I've since figured that out and got the right connection. 

The next issue I had was in configuring a proper init sequence since the default ILI9341 driver does not load it properly and still displays a white screen.

I was finally able to get it running using the below sequence:
```sudo modprobe fbtft_device custom name=fb_ili9341 gpios=reset:25,dc:24 speed=48000000 width=240 height=320 fps=24 rotate=90 bgr=1 init=-2,500,-1,0x01,-2,200,-1,0xCF,0x00,0x8B,0X30,-1,0xED,0x67,0x03,0X12,0X81,-1,0xE8,0x85,0x10,0x7A,-1,0xCB,0x39,0x2C,0x00,0x34,0x02,-1,0xF7,0x20,-1,0xEA,0x00,0x00,-1,0xC0,0x1B,-1,0xC1,0x10,-1,0xC5,0x3F,0x3C,-1,0xC7,0XB7,-1,0x36,0x08,-1,0x3A,0x55,-1,0xB1,0x00,0x1B,-1,0xB6,0x0A,0xA2,-1,0xF2,0x00,-1,0x26,0x01,-1,0xE0,0x0F,0x2A,0x28,0x08,0x0E,0x08,0x54,0xA9,0x43,0x0A,0x0F,0x00,0x00,0x00,0x00,-1,0XE1,0x00,0x15,0x17,0x07,0x11,0x06,0x2B,0x56,0x3C,0x05,0x10,0x0F,0x3F,0x3F,0x0F,-1,0x11,-2,120,-1,0x29,-3```

One of my problems is that I really have no idea what each of those strings means in the init sequence. I pulled the pattern from the Seeed Studio arduino library and when I loaded it up, the screen worked. However, everything I'm seeing is flipped/mirrored exactly like this guy's screen https://github.com/notro/fbtft/issues/83.

Any idea what string might fix that?

notro, 2015-06-03:
Register 0x36 controls mirror/flip on these controllers. Sadly the value is hardcoded in the set_var() function in the driver. You can look through the fbtft drivers and find the ones that have a similar set_addr_win() function and uses 0x36 in set_var(). Then try these instead of fb_ili9341. If none of the drivers work for you, you need to use flexfb. flexfb does not support rotation, so it doesn't mess up 0x36.

buzzdennis, 2015-06-04:
Awesome. Got it working using the fb_hx8353d driver. I'll be writing up a blog post on how to get this screen working. Below is the code I used to get it running in case others want to get it going as well. I'll be trying different refresh rates and fps values just to see what it can handle. Thanks again @notro 

fbtft_device custom name=fb_hx8353d gpios=reset:25,dc:24 speed=48000000 width=240 height=320 fps=25 rotate=90 bgr=1 init=-2,500,-1,0x01,-2,200,-1,0xCF,0x00,0x8B,0X30,-1,0xED,0x67,0x03,0X12,0X81,-1,0xE8,0x85,0x10,0x7A,-1,0xCB,0x39,0x2C,0x00,0x34,0x02,-1,0xF7,0x20,-1,0xEA,0x00,0x00,-1,0xC0,0x1B,-1,0xC1,0x10,-1,0xC5,0x3F,0x3C,-1,0xC7,0XB7,-1,0x36,0x08,-1,0x3A,0x55,-1,0xB1,0x00,0x1B,-1,0xB6,0x0A,0xA2,-1,0xF2,0x00,-1,0x26,0x01,-1,0xE0,0x0F,0x2A,0x28,0x08,0x0E,0x08,0x54,0xA9,0x43,0x0A,0x0F,0x00,0x00,0x00,0x00,-1,0XE1,0x00,0x15,0x17,0x07,0x11,0x06,0x2B,0x56,0x3C,0x05,0x10,0x0F,0x3F,0x3F,0x0F,-1,0x11,-2,120,-1,0x29,-3

closed 2015-09-06 by notro