notro/fbtft #286: ssd 1963 raspberry pi b tft 5"

edsondragon, 2015-06-11:
Hello I have a raspberry pi B and an LCD 5 "with ssd1963.
The connection between pins am in doubt as to:
RST --- GPIO 29
WR ---- GPIO 31

In raspberry B has no such GPIO 29 to 31, in which I connect?

It has tutorials but it is difficult to solve .....

I'm sorry google translator -I'm from Brazil

notro, 2015-06-18:
Those gpios are only present on boards with a P5 header: http://elinux.org/RPi_Low-level_peripherals
Pick some other gpios from the 40-pin connector: http://pi.gadgetoid.com/pinout

edsondragon, 2015-06-18:
Hello
Thanks for answering, yesterday I managed to make work, but missing features colors and not resized to 800x480, I wanted to work with touch XPT2046 my lcd ... but the first step I got through their tutorials

closed 2015-09-06 by notro