notro/fbtft #290: Install the  FBTFT driver on Retropie

mjimack360omg, 2015-06-18:
Hi I just got a Waveshare 3.5 Inch TFT screen and a raspberry pi 2. I want to install the drive on the retropie and it dosn't seem to work. Can anyone help?

mjimack360omg, 2015-06-18:
My specs are :
Waveshare 3.5 inch
Raspberry Pi 2 Model B
Retropie 3.0 beta for Raspberry Pi 2

mjimack360omg, 2015-06-18:
I had followed the steps on installing the drive on post#215 but whenever i reach step 5 i get the ERROR: could not insert 'fbtft_device': Invalid argument. Could someone explain and solution and the complete steps on install the driver for the tft display.

mjimack360omg, 2015-06-19:
I have made some progress right now, I used the DVD driver that came with the TFTscreen and I boot into the fresh rasbian. So i script installed retropie and it worked! Not yet. Instead of the TFTscreen showing the emulator the TFTscreen is showing the console and the emulator is being outputed through the HDMI port. Anyway?

Konoa, 2015-09-06:
i tried using someone else's on getting a 2.8" tft and it didn't work so if you could figure yours out it would help a ton as i have the same exact specs as you.

notro, 2015-12-04:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2015-12-04 by notro

benplum, 2016-01-05:
Having the same issue. Console is output to TFT, emulation station to HDMI.

bolsunovskyi, 2017-01-05:
> Console is output to TFT, emulation station to HDMI.

Having same problem :(

MarSoft, 2018-12-07:
I think EmulationStation uses hardware graphics acceleration which is only available with HDMI. But you can use `fbcp` which will run in background and duplicate image from hdmi framebuffer to tft framebuffer. Not very beautiful solution but it works.