notro/fbtft #294: waveshare22 enable backlight

haiko21, 2015-06-28:
Dear notro,

would it be possible to enable backlight for waveshare22?

I'm using it with MZTX-PI-EXT, http://elinux.org/MZTX-PI-EXT, and with modified fbtft_device.c, the backlight works perfectly.

Module settings are: fbtft_device name=waveshare22 rotate=270 gpios=reset:23,dc:25,led_:18

The patch is trivial:

```
pi@raspberrypi ~/linux/drivers/video/fbtft $ git diff
diff --git a/fbtft_device.c b/fbtft_device.c
index b9f4c30..ecf63cd 100644
--- a/fbtft_device.c
+++ b/fbtft_device.c
@@ -1038,6 +1038,7 @@ static struct fbtft_device_display displays[] = {
                        .platform_data = &(struct fbtft_platform_data) {
                                .display = {
                                        .buswidth = 8,
+                                       .backlight = 1,
                                },
                                .gpios = (const struct fbtft_gpio []) {
                                        { "reset", 24 },
```

Thank you very much.

notro, 2015-06-29:
See here for where to send the patch: https://github.com/notro/fbtft/wiki/Development

haiko21, 2015-06-29:
I will try :)
Thank you.

closed 2015-06-29 by haiko21