notro/fbtft #298: Raspberry Pi 2 with Ubuntu Mate and Waveshare 32b

bkems, 2015-07-06:
Hi notro

I'm confused about the status of the fbtft drivers and the debian kernel. Can I install the fbtft drivers on Ubuntu Mate? Or don't I need to? lsmod doesn't show fbtft listed. I bricked a Pi 2 running Raspian Jessie by installing the fbtft drivers and I'd like not to do that again id possible.

If someone reads this who has done this before or knows what I need to do I'd love to get some help. :-)

Guy

jonklinger, 2015-08-04:
this is a comment to see when the solution comes.

notro, 2015-11-23:
This looks to be solved now, from changelog:

2015-10-14 - Ubuntu MATE 15.10 for Raspbery Pi 2 Release Candidate
* Updated to Linux 4.1.10.
Now using the kernel, firmware and drivers from Raspberry Pi Foundation and includes rpi-update to easily update the kernel and firmware.

https://ubuntu-mate.org/raspberry-pi/

closed 2016-04-28 by notro