notro/fbtft #299: Rpi 2 FBTFT  240x320 3.2  9-bit spi ili9341 ,but it seems the init is not successed......

gaosong00, 2015-07-07:
[   50.806739] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[   50.812630] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[   50.813869] fbtft_device:  SPI devices registered:
[   50.813894] fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00
[   50.813914] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   50.813924] fbtft_device:  'fb' Platform devices registered:
[   50.813956] fbtft_device:      soc:fb id=-1 pdata? no
[   50.814012] fbtft_device: Deleting spi0.0
[   50.814447] spi spi0.0: setting up native-CS0 as GPIO 8
[   50.814594] fbtft_device:  GPIOS used by 'mi0283qt-9a':
[   50.814609] fbtft_device:    'reset' = GPIO25
[   50.814621] fbtft_device:    'led' = GPIO18
[   50.814631] fbtft_device:  SPI devices registered:
[   50.814647] fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00
[   50.814666] fbtft_device:      fb_ili9341 spi0.0 32000kHz 8 bits mode=0x00
[   50.831372] fb_ili9341: module is from the staging directory, the quality is unknown, you have been warned.
[   50.953406] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958442] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958463] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958481] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958499] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958516] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958534] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958551] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958568] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958586] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958603] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958620] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958638] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958655] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958672] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958689] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958707] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   50.958724] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   51.058762] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   51.078793] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   51.078816] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   51.078835] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   51.078852] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   51.078979] fb_ili9341 spi0.0: fbtft_update_display: write_vmem failed to update display buffer
[   51.078998] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   51.079016] fb_ili9341 spi0.0: fbtft_write_reg8_bus9: write() failed and returned -22
[   51.079739] graphics fb1: fb_ili9341 frame buffer, 240x320, 150 KiB video memory, 16 KiB DMA buffer memory, fps=20, spi0.0 at 32 MHz

when i dmesg，it shows the above
i have opened the SPI by sudo raspi-config

i need help

notro, 2015-07-07:
Yes, and I can see why it happens. The official kernel has moved to the mainline SPI driver spi-bcm2835. This driver does not verify the bits_per_word value in it's setup function.
FBTFT relies on this to know if it should switch to emulating 9-bit (see [fbtft_probe_common](http://lxr.free-electrons.com/ident?i=fbtft_probe_common)).

The spi-bcm2708 driver is still available, so this Device Tree overlay should enable it instead of spi-bcm2835.
```
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2836", "brcm,bcm2708", "brcm,bcm2709";
	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			status = "okay";
			compatible = "brcm,bcm2708-spi";
		};
	};
};

```
https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays

yifanshu02, 2015-07-08:
hi，notro，i am a fish about linux,so i just understand a bit about above
I think i should do  this on  my PRI2:
wget -c https://raw.githubusercontent.com/RobertCNelson/tools/master/pkgs/dtc.sh
(wait...)
chmod +x dtc.sh
(wait...)
./dtc.sh
(wait...)
sudo dtc -@ -I dts -O dtb -o /boot/overlays/spi-bcm2708-overlay.dtb spi-bcm2708-overlay.dts

but when  i run the last code,i get:
Couldn't open "spi-bcm2708-overlay.dts":No such file or directory 

aho...please help me,men

notro, 2015-07-08:
This should go in the *spi-bcm2708-overlay.dts* file:
```
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2836", "brcm,bcm2708", "brcm,bcm2709";
	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			status = "okay";
			compatible = "brcm,bcm2708-spi";
		};
	};
};

```

closed 2015-10-18 by notro