notro/fbtft #300: fbtft + device tree on sam9g25 cpu ssd1331 mirrored display

mani009, 2015-07-11:
Hello everybody,

I have an arietta g25 with ssd1331 oled display with mirrored display.

I have start the Kernel modul with this changes : 

modprobe fbtft_device name=ssd1331 busnum=32766 gpios=reset:67,dc:68 speed=20000000

dmesg output : 

fbtft: module is from the staging directory, the quality is unknown, you have been warned.
fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
fbtft_device:  SPI devices registered:
fbtft_device:  'fb' Platform devices registered:
fbtft_device:  GPIOS used by 'ssd1331':
fbtft_device:    'reset' = GPIO67
fbtft_device:    'dc' = GPIO68
fbtft_device:  SPI devices registered:
fbtft_device:      fb_ssd1331 spi32766.0 20000kHz 8 bits mode=0x03
fb_ssd1331: module is from the staging directory, the quality is unknown, you have been warned.
graphics fb0: fb_ssd1331 frame buffer, 96x64, 12 KiB video memory, 4 KiB DMA buffer memory, fps=21, spi32766.0 at 20 MHz

I have start die Module again with rotate option and have the same problem : 
modprobe fbtft_device name=ssd1331 busnum=32766 gpios=reset:67,dc:68 speed=20000000 rotate=270

dts:

                spi1: spi@f0004000 {
                status = "okay";
                cs-gpios = <&pioA 8 0>;

                        /*device@1 {
                        compatible = "spidev";
                        spi-max-frequency = <50000000>;     // 5 MHz
                        reg = <1>;
                        };*/
            };

Can anyone help me what i`am doing wrong ???

irfansehic, 2015-07-30:
What is the problem? I also use sam9g25 (Arietta G25 by Acme).

mani009, 2015-07-31:
Hello irfansehic,

the problem is the display is mirrored . I take a photo form the display , but at the moment i have borrowed my arietta board.

And the Rotate option doesn`t change anything.

notro, 2015-07-31:
fb_ssd1331 does not support rotation (no fbtftops.set_var function).
A mirrored display means that the default init sequence does not match your display.
From the datasheet I can see that register 0xA0 controls mirroring.
Try changing the value in the driver init_display() function:
```
        write_reg(par, 0xa0, 0x70 | (par->bgr << 2)); /* Set Colour Depth */
```
Driver: http://lxr.free-electrons.com/source/drivers/staging/fbtft/fb_ssd1331.c
Datasheet page 37: http://www.adafruit.com/datasheets/SSD1331_1.2.pdf

mani009, 2015-07-31:
Thanks a lot ,

I'll try it and write back as soon as possible .

quinte17, 2015-08-01:
and yes changing the initialisation helps a bit.
atm using kernel 4.1 with dts.
```
                        spi1: spi@f0004000 {
                                status = "okay";
                                cs-gpios = <&pioA 8 0>;

                                device@0 {
                                        rotate = <270>;
                                        bgr;
                                        fps = <30>;
                                        compatible = "solomon,ssd1331";
                                        spi-max-frequency = <20000000>;
                                        reg = <0>;
                                        regwidth = <8>;
                                        buswidth = <8>;
                                        verbose = <3>;
                                        reset-gpios = <&pioC 3 0>; /* PC3 - J4.33 */
                                        dc-gpios = <&pioC 4 0>;    /* PC4 - J4.31 */
                                };
                        };
```
the white bottom is some random painting. memory not initialized or something.
and the mirrored text on the right is some old value from the boot before

[image]

quinte17, 2015-08-01:
without the rotate option in the dts and with the flip bit for rotation set, the display seems working, at least for console.
```
                        spi1: spi@f0004000 {
                                status = "okay";
                                cs-gpios = <&pioA 8 0>;

                                device@0 {
                                        bgr;
                                        fps = <30>;
                                        compatible = "solomon,ssd1331";
                                        spi-max-frequency = <20000000>;
                                        reg = <0>;
                                        regwidth = <8>;
                                        buswidth = <8>;
                                        verbose = <3>;
                                        reset-gpios = <&pioC 3 0>; /* PC3 - J4.33 */
                                        dc-gpios = <&pioC 4 0>;    /* PC4 - J4.31 */
                                };
                        };
```
[image]

quinte17, 2015-08-01:
still having trouble with displaying bitmaps.
is there an example file anywhere which can be dd'ed to fb0?

notro, 2015-08-01:
I have used this from time to time:
```
git clone https://git.kernel.org/pub/scm/linux/kernel/git/geert/fbtest.git
cd fbtest
make
./fbtest -f /dev/fb1
```

quinte17, 2015-08-01:
notro, thx that worked like a charm.
is it possible to add an option to the devicetree to be able to set that bit?
how could a generic way look like?
```
# changed line
write_reg(par, 0xa0, 0x02 | 0x70 | (par->bgr << 2)); /* Set Colour Depth */
```

notro, 2015-08-02:
There is an option already to change the whole initialisation: ```init=<..>;```
See https://github.com/raspberrypi/linux/blob/rpi-4.0.y/arch/arm/boot/dts/overlays/mz61581-overlay.dts#L67

Rules:
A command byte must be or'ed with 0x1000000
A delay is or'ed with 0x2000000

The function [fbtft_init_display_dt](http://lxr.free-electrons.com/ident?i=fbtft_init_display_dt) is responsible for parsing the init property.

quinte17, 2015-08-02:
so as i understand it:
if an init-section exists then it is executed instead of the ```static int init_display(struct fbtft_par *par)```

does that also mean that some options in the device-tree, which are only handled in the ```.. init_display ..```, wont have any effect anymore?
so if someone wants to have that option set he has to manually change the init-sequence of the devicetree.

quinte17, 2015-08-02:
notro, thx again.
the init section in the dts is something i think i like.
i believe you can close this issue now.
for completeness the dts part:
```
			spi1: spi@f0004000 {
				status = "okay";
				cs-gpios = <&pioA 8 0>;

				device@0 {
					fps = <30>;
					compatible = "solomon,ssd1331";
					spi-max-frequency = <20000000>;
					reg = <0>;
					regwidth = <8>;
					buswidth = <8>;
					verbose = <3>;
					reset-gpios = <&pioC 3 0>; /* PC3 - J4.33 */
					dc-gpios = <&pioC 4 0>;    /* PC4 - J4.31 */

					init = <0x10000ae /* Display Off */
						0x10000a0 0x72 /* Set Colour Depth and left to right */
						0x10000a0 0x72 /* RGB colour */
						0x10000a1 0x00 /* Set Display Start Line */
						0x10000a2 0x00 /* Set Display Offset */
						0x10000a4 /* NORMALDISPLAY */
						0x10000a8 0x3f /* Set multiplex */
						0x10000ad 0x8e /* Set master */
					     /* 0x10000b0 0x0b  Set power mode */
						0x10000b1 0x31 /* Precharge */
						0x10000b3 0xf0 /* Clock div */
						0x100008a 0x64 /* Precharge A */
						0x100008b 0x78 /* Precharge B */
						0x100008c 0x64 /* Precharge C */
						0x10000bb 0x3a /* Precharge level */
						0x10000be 0x3e /* vcomh */
						0x1000087 0x06 /* Master current */
						0x1000081 0x91 /* Contrast A */
						0x1000082 0x50 /* Contrast B */
						0x1000083 0x7d /* Contrast C */
						0x10000af /* Set Sleep Mode Display On */
						>;
				};
			};
```

closed 2015-10-18 by notro

denghongcai, 2017-04-18:
@notro I faced the same problem with @quinte17 . how to rewrite init sequence using `modprobe fbtft_device name=ssd1331 gpios=reset:8,dc:7 init=?` command instead of dts? for example, `0x10000a0 0x72` equals to what? (sorry, i'm a driver noob...)

notro, 2017-04-18:
> 0x10000a0 0x72

init=-1,0xa0,0x72,-1,reg2,val,...,-3

See: https://github.com/notro/fbtft/wiki/fbtft_device#parameters