notro/fbtft #301: 2.4" Arduino shield possible ? (SPFD5408)

Ardakilic, 2015-07-14:
Hello,

I've this arduino shield:

[http://www.aliexpress.com/item//32254339975.html](http://www.aliexpress.com/item//32254339975.html)

[Here](https://alselectro.wordpress.com/2014/11/28/touch-screen-tft-shield-for-arduino-uno/)'s a how to for this module for Arduino.

This is the back of the module
[image]

It has SPFD5408 controller over it.

A commenter on aliexpress mentioned this is based on ILI9341 driver.

I've tried to google it, but I couldn't find an exact match of this shield or controller being used on Raspberry Pi.

Can you assist me here please?

Thank in advance,

Ardakilic, 2015-07-16:
I have wired like this:

lcd_rst = GPIO17
lcd_cs = GPIO27
lcd_rs = 
lcd_wr = GPIO3
lcd_rd = 

GND = GND

5V = +5V
3v3 =

LCD_D2 = GPIO9
LCD_D3 = GPIO11
LCD_D4 = GPIO18
LCD_D5 = GPIO23
LCD_D6 = GPIO24
LCD_D7 = GPIO25
LCD_D0 = GPIO8
LCD_D1 = GPIO7

SD_SS = 
SD_DI =
SD_DO =
SD_SCK =

Blanks are not wired.

Because @notro suggested **itdb24** for similar modules in #67 , I've checked [this source code](https://github.com/notro/fbtft/blob/e67014490a9df34b9a4bf04e49c50254aebc10a8/fbtft_device.c#L520-L550), and wired like above.

This is my code to load the module:

```
sudo modprobe fbtft_device name=itdb24 gpios=db0:8,db1:7,db2:9,db3:11,db4:18,db5:23,db6:24,db7:25,cs:27,dc:2,wr:3,reset:17 rotate=90
```

The framebuffer appears, but like as stated in [this forum thread](https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=98442), it's blank.

This is my dmesg output:

```
[   48.016136] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[   48.021729] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[   48.022988] fbtft_device:  SPI devices registered:
[   48.023009] fbtft_device:  'fb' Platform devices registered:
[   48.023040] fbtft_device:      soc:fb id=-1 pdata? no
[   48.023260] fbtft_device:  GPIOS used by 'itdb24':
[   48.023277] fbtft_device:    'db0' = GPIO8
[   48.023289] fbtft_device:    'db1' = GPIO7
[   48.023300] fbtft_device:    'db2' = GPIO9
[   48.023311] fbtft_device:    'db3' = GPIO11
[   48.023323] fbtft_device:    'db4' = GPIO18
[   48.023334] fbtft_device:    'db5' = GPIO23
[   48.023345] fbtft_device:    'db6' = GPIO24
[   48.023356] fbtft_device:    'db7' = GPIO25
[   48.023367] fbtft_device:    'cs' = GPIO27
[   48.023378] fbtft_device:    'dc' = GPIO2
[   48.023389] fbtft_device:    'wr' = GPIO3
[   48.023401] fbtft_device:    'reset' = GPIO17
[   48.023411] fbtft_device:  'fb' Platform devices registered:
[   48.023433] fbtft_device:      soc:fb id=-1 pdata? no
[   48.023465] fbtft_device:      fb_s6d1121 id=0 pdata? yes
[   48.036910] fb_s6d1121: module is from the staging directory, the quality is unknown, you have been warned.
[   48.403319] graphics fb1: fb_s6d1121 frame buffer, 320x240, 150 KiB video memory, 4 KiB DMA buffer memory, fps=20
```

What may be the issue, how can I run this? Any ideas?

notro, 2015-10-18:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2015-10-18 by notro

larsenglund, 2016-04-25:
Was any solution found for this?

Shin2D, 2016-11-02:
Any solution?

jorgeolivares, 2017-02-07:
No solution yet ?

spawnmax, 2017-04-12:
You can find [here](http://www.orangepi.org/orangepibbsen/forum.php?mod=viewthread&tid=879) correct commands to init the lcd. Also, don't forget to wire LCD_RD with HIGH.

asianshakespeare, 2018-01-15:
Any Solution Yet

Ardakilic, 2018-01-15:
Personally, I gave up, couldn't find any proper solution to run it on latest Raspbian.

elitvinov, 2019-08-07:
@Ardakilic If it still actual
I managed to run lcd part using flexfb in 8-bit parallel mode. 
see wiring below in second solution
```
# cat /etc/modprobe.d/fbtft.conf
options fbtft_device name=flexpfb gpios=dc:4,reset:17,wr:15,cs:27,db00:9,db01:11,db02:18,db03:23,db04:24,db05:25,db06:8,db07:7
options flexfb width=320 height=240 buswidth=8 init=\
-1,0x28,-2,20,-1,0xCF,0x00,0x83,0x30,-1,0xED,0x64,0x03,0x12,0x81,-1,0xE8,0x85,0x01,0x79,\
-1,0xCB,0x39,0x2c,0x00,0x34,0x02,-1,0xF7,0x20,-1,0xEA,0x00,0x00,-1,0xC0,0x26,-1,0xC1,0x11,\
-1,0xC5,0x35,0x3E,-1,0xC7,0xBE,-1,0xB1,0x00,0x1B,-1,0xB6,0x0a,0x82,0x27,0x00,-1,0xB7,0x07,\
-1,0x3A,0x55,-1,0x36,0x38,-1,0x11,-2,120,-1,0x29,-2,20,-3
```

2nd solution: using https://github.com/sammyizimmy/ili9341 
thanks to @sammyizimmy 

touchpanel still in work..

CRImier, 2019-08-07:
wait, there's a touchpanel? The original post doesn't show any. If so, I'd guess it has some kind of off-the-shelf controller.

elitvinov, 2019-08-07:
@CRImier This one I have:
https://www.aliexpress.com/item/32919273566.html
see descriprion:
> 4. 4-wire resistive touch screen
stilus present as well

Another description I guess is http://www.raspberrypiwiki.com/index.php/2.4_inch_TFT_LCD_for_Arduino_SKU:394081
In zip archive there are examples for Arduino and STM32. 

Maybe it must be another issue for this shield's touchscreen support?

CRImier, 2019-08-07:
In the ZIP on the raspberrypiwiki, I found this:

```
 #define YP A3  // must be an analog pin, use "An" notation!
 #define XM A2  // must be an analog pin, use "An" notation!
 #define YM 9   // can be a digital pin
 #define XP 8   // can be a digital pin
```

So, it doesn't seem like they have any kind of touchscreen controller - just connecting the resistive touchscreen pins to the GPIOs. In that case, it will be hard to use that kind of touchscreen with a Raspberry Pi, as it doesn't have analog inputs - to put it simply, unless you add some kind of controller (i.e. XPT2046 ) to read data from it, seems like it won't work.

elitvinov, 2019-08-07:
Yesm you correct, next this "internal" pins go to "external" (you can see them on photo, there are no separate pins for touch unfortunately):
```
// For better pressure precision, we need to know the resistance
// between X+ and X- Use any multimeter to read it
// For the one we're using, its 300 ohms across the X plate
TouchScreen ts = TouchScreen(XP, YP, XM, YM, 300);

#define LCD_CS A3
#define LCD_CD A2
#define LCD_WR A1
#define LCD_RD A0
```
Well, am I right you mean that to read touch the resistance should be made between LCD_CS and LCD_CD pins on shield?

CRImier, 2019-08-07:
It indeed seems to be the case that they're connected together. In that case, you need cooperation from the display driver - not send anything to the display while you're reading the touch state from the CS and CD pins (and switch them to being inputs, of course). I don't see any way how that could reasonably work with a Raspberry Pi, to be honest - unless you, at the very least, rewire the pins, or do  some buffering of the CS and CD pins. Cheaper to just get a 3.5" RPi-ready waveshare-like display from eBay, it'd be supported by fbtft too =)

elitvinov, 2019-08-08:
Ok, thank you!
As for me touch for RPi is not a problem. The shield was bought for my STM32 labs and waits when I find time for it. Raspberry Pi is another lab, much simple because of real linux os.
Nevertheless LCD solution works.