notro/fbtft #303: xorg-macros 1.8

tyler-kalevra, 2015-07-18:
I'm running pidora on a rev 1 pi.
trying to run the autorun is failing. 

Any input or missing information is appreciated.

[root@raspi xf86-video-fbturbo-master]# ./autogen.sh 
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal 
configure.ac:38: error: must install xorg-macros 1.8 or later before running autoconf/autogen
configure.ac:38: the top level
autom4te: /usr/bin/m4 failed with exit status: 1
aclocal: error: /usr/bin/autom4te failed with exit status: 1
autoreconf: aclocal failed with exit status: 1

notro, 2015-10-18:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2015-10-18 by notro

tyler-kalevra, 2015-10-23:
I love how this was closed with 0 input from dev?

notro, 2015-10-25:
Which dev? This isn't a xorg project.

tyler-kalevra, 2015-10-28:
I don't understand the statement "This isn't a xorg project." or how it relates to the reported issue with this driver compiling.

notro, 2015-10-31:
AFAIKT you are trying to build a xorg driver, a userspace driver for X windows. fbtft is a framebuffer kernel driver project.