notro/fbtft #306: Waveshare32b (touch interferes with LCD rendering)

vbenso, 2015-08-01:
First of all, thanks for the awesome work.

Now comes the bad news, I've configured the LCD+touch using the waveshare32b overlay in /boot/config.txt:
- dtoverlay=waveshare32b

Calibrations being done, X comes up, I can see the desktop, everything looks fine, until I touch the lcd using the stylus (actually using my finger the same problem happens). 
When the lcd is touched the screen get's all messed up. It shows multiple pointers, regions with inverted colors and so on. After some touches, when the rendered screen is looking like a mess, if I connect my mouse and move it, the rendered desktop looks perfectly fine. Until I touch it again and all hell breaks loose again.
(From my POV looks like the touch controller is getting in the way of the communication between the pi and the LCD)

[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.0.7-v7+ (pi@raspi2) (gcc version 4.8.3 20140106 (prerelease) (crosstool-NG linaro-1.13.1-4.8-2014.01 - Linaro GCC 2013.11) ) #1 SMP PREEMPT Sat Jul 11 20:44:05 CEST 2015
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 808404c0, node_mem_map b9fa6000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (8660->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 11 pages/cpu @b9f6d000 s16128 r8192 d20736 u45056
[    0.000000] pcpu-alloc: s16128 r8192 d20736 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa21041 bcm2709.serial=0x13d4774c smsc95xx.macaddr=B8:27:EB:D4:77:4C bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbtft_device.custom fbtft_device.name=tm022hdh26 fbtft_device.gpios=dc:22,reset:27 fbtft_device.bgr=1 fbtft_device.speed=32000000 fbcon=font:ProFont6x11 logo.nologo dma.dmachans=0x7f35 console=tty1 consoleblank=0 fbtft_device.fps=50 fbtft_device.rotate=0
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939664K/966656K available (5885K kernel code, 378K rwdata, 1784K rodata, 400K init, 757K bss, 18800K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff000000   (1080 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x8078575c   (7670 kB)
[    0.000000]       .init : 0x80786000 - 0x807ea000   ( 400 kB)
[    0.000000]       .data : 0x807ea000 - 0x80848a64   ( 379 kB)
[    0.000000]        .bss : 0x80848a64 - 0x809061cc   ( 758 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (virt).
[    0.000015] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 3579139424256ns
[    0.000038] Switching to timer-based delay loop, resolution 52ns
[    0.000304] Console: colour dummy device 80x30
[    0.002255] console [tty1] enabled
[    0.002323] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.002425] pid_max: default: 32768 minimum: 301
[    0.002804] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002873] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.004184] Initializing cgroup subsys blkio
[    0.004267] Initializing cgroup subsys memory
[    0.004337] Initializing cgroup subsys devices
[    0.004397] Initializing cgroup subsys freezer
[    0.004477] Initializing cgroup subsys net_cls
[    0.004602] CPU: Testing write buffer coherency: ok
[    0.004727] ftrace: allocating 20007 entries in 59 pages
[    0.053609] CPU0: update cpu_capacity 1024
[    0.053693] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.053740] [bcm2709_smp_prepare_cpus] enter
[    0.053897] Setting up static identity map for 0x54ef70 - 0x54efa4
[    0.113502] [bcm2709_boot_secondary] cpu:1 started (0) 16
[... 155 lines cut ...]
[    2.939520] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:d4:77:4c
[    3.183456] udevd[177]: starting version 175
[    3.731789] bcm2708_spi 3f204000.spi: DMA channel 2 at address 0xf3007200 with irq 77
[    3.751284] bcm2708_spi 3f204000.spi: DMA channel 4 at address 0xf3007400 with irq 20
[    3.804997] ads7846 spi0.1: touchscreen, irq 497
[    3.816207] input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input0
[    3.832967] spi_master spi0: /soc/spi@7e204000/waveshare35a-ts@1 has no valid 'spi-max-frequency' property (-22)
[    3.850508] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/waveshare35a-ts@1
[    3.925320] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    3.948259] fb_ili9340: module is from the staging directory, the quality is unknown, you have been warned.
[    3.972172] fbtft_of_value: buswidth = 8
[    3.989094] fbtft_of_value: debug = 0
[    4.009608] fbtft_of_value: rotate = 90
[    4.020175] fbtft_of_value: fps = 25
[    4.390262] graphics fb1: fb_ili9340 frame buffer, 320x240, 150 KiB video memory, 4 KiB DMA buffer memory, fps=25, spi0.0 at 16 MHz
[    4.407226] bcm2708_spi 3f204000.spi: SPI Controller at 0x3f204000 (irq 80)
[    4.416745] bcm2708_spi 3f204000.spi: SPI Controller running in dma mode
[    5.979503] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    6.228541] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   10.071087] random: dd urandom read with 111 bits of entropy available
[   11.298332] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   12.902488] random: nonblocking pool is initialized
[   12.902586] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0x45E1
[   14.618635] cfg80211: Calling CRDA to update world regulatory domain
[   15.173116] Adding 102396k swap on /var/swap.  Priority:-1 extents:2 across:2134012k SSFS

notro, 2015-10-18:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2015-10-18 by notro

vbenso, 2017-02-09:
Any news on this issue? Just tested with raspbian jessie and the is still interference on lcd rendering when touched.
```
Driver for 1-wire Dallas network protocol.
[    7.979807] w1-gpio onewire@0: gpio pin 4, external pullup pin -1, parasitic power 0
[    7.979867] w1_add_master_device: set_pullup requires write_byte or touch_bit, disabling
[    7.996837] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
[    8.017866] bcm2835-wdt 3f100000.watchdog: Broadcom BCM2835 watchdog timer
[    8.025856] spi-bcm2835 3f204000.spi: chipselect 1 already in use
[    8.025871] spi_master spi0: spi_device register error /soc/spi@7e204000/waveshare32b_ts@1
[    8.025886] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/waveshare32b_ts@1
[    8.025911] spi_master spi0: /soc/spi@7e204000/waveshare35a-ts@1 has no valid 'spi-max-frequency' property (-22)
[    8.025922] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/waveshare35a-ts@1
[    8.028765] bcm2708_i2c 3f804000.i2c: BSC1 Controller at 0x3f804000 (irq 83) (baudrate 100000)
[    8.389638] ads7846 spi0.1: touchscreen, irq 183
[    8.401061] input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input2
[    8.429690] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    8.436144] fb_ili9340: module is from the staging directory, the quality is unknown, you have been warned.
[    8.437305] pinctrl-bcm2835 3f200000.gpio: pin gpio17 already requested by spi0.1; cannot claim for spi0.0
[    8.437341] pinctrl-bcm2835 3f200000.gpio: pin-17 (spi0.0) status -22
[    8.437366] pinctrl-bcm2835 3f200000.gpio: could not request pin 17 (gpio17) from group gpio17  on device pinctrl-bcm2835
[    8.437384] fb_ili9340 spi0.0: Error applying setting, reverse things back
[    8.437490] fbtft_of_value: buswidth = 8
[    8.437513] fbtft_of_value: debug = 0
[    8.437527] fbtft_of_value: rotate = 270
[    8.437540] fbtft_of_value: fps = 25
```

maGitty, 2017-10-31:
Same here, I'm also searching, but couldn't find anything yet:

[    7.072948] bcm2835-wdt 20100000.watchdog: Broadcom BCM2835 watchdog timer
[    7.143239] gpiomem-bcm2835 20200000.gpiomem: Initialised: Registers at 0x20200000
[    7.216844] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    7.501666] spi-bcm2835 20204000.spi: chipselect 1 already in use
[    7.501707] spi_master spi0: spi_device register error /soc/spi@7e204000/waveshare32b_ts@1
[    7.501733] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/waveshare32b_ts@1
[    7.501767] spi_master spi0: /soc/spi@7e204000/waveshare35a-ts@1 has no valid 'spi-max-frequency' property (-22)
[    7.501788] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/waveshare35a-ts@1
[    7.539589] bcm2708_i2c 20804000.i2c: BSC1 Controller at 0x20804000 (irq 77) (baudrate 100000)
[    8.462304] cdc_ether 1-1:2.0 usb0: register 'cdc_ether' at usb-20980000.usb-1, CDC Ethernet Device, 02:1e:10:1f:00:00
[    8.491251] usbcore: registered new interface driver cdc_ether
[    8.525930] usbcore: registered new interface driver brcmfmac
[    8.710179] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7
[    8.767443] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    9.692423] systemd-journald[109]: Received request to flush runtime journal from PID 1
[   11.661139] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[   11.678122] ads7846 spi0.1: touchscreen, irq 177
[   11.691650] input: ADS7846 Touchscreen as /devices/platform/soc/20204000.spi/spi_master/spi0/spi0.1/input/input0
[   11.731079] fb_ili9340: module is from the staging directory, the quality is unknown, you have been warned.
[   11.732078] pinctrl-bcm2835 20200000.gpio: pin gpio17 already requested by spi0.1; cannot claim for spi0.0
[   11.732109] pinctrl-bcm2835 20200000.gpio: pin-17 (spi0.0) status -22
[   11.732127] pinctrl-bcm2835 20200000.gpio: could not request pin 17 (gpio17) from group gpio17  on device pinctrl-bcm2835
[   11.732143] fb_ili9340 spi0.0: Error applying setting, reverse things back
[   11.732228] fbtft_of_value: buswidth = 8
[   11.732246] fbtft_of_value: debug = 0
[   11.732259] fbtft_of_value: rotate = 180
[   11.732269] fbtft_of_value: fps = 25
[   12.700133] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[   12.700198] cfg80211: World regulatory domain updated:
[   12.700212] cfg80211:  DFS Master region: unset
[   12.700222] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[   12.700238] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[   12.700253] cfg80211:   (2457000 KHz - 2482000 KHz @ 20000 KHz, 92000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   12.700267] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[   12.700298] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[   12.700313] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[   12.700325] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[   12.700336] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[   12.700347] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[   13.266082] graphics fb1: fb_ili9340 frame buffer, 240x320, 150 KiB video memory, 4 KiB DMA buffer memory, fps=25, spi0.0 at 16 MHz