notro/fbtft #307: Driver feature suggestion

irfansehic, 2015-08-02:
Hello notro..

If you look at my issue 305: https://github.com/notro/fbtft/issues/305
you will notice that I had the problem that the kernel did not tell me my SPI driver does not support 9bit mode, but the driver kept sending 9bit frames over an 8bit bus... This resulted in every other pixel not getting its color set properly (missing DC=1 bit !?). 
My display does not have a DC pin, so I had to use 9bit mode.
So I am not sure where the problem is, but here is an idea..

Suggestion/Question: Would it make sense to add a new (debug!?) flag called e.g. emulate9bit which, when set, would force sending  9bit packages over a 8bit bus (so, for RasPi, instead of LoSSi, normal 8bit would be used), and when not set, the driver would just work as before..

If the answer is yes, or if you have a better suggestion how this problem could be mitigated, I would love to contribute to your repo..

notro, 2015-08-03:
fbtft relies on the setup() function of the SPI master controller driver to fail if 9 bits_per_word is not supported: http://lxr.free-electrons.com/source/drivers/staging/fbtft/fbtft-core.c#L1434
This works with the raspberry pi spi_bcm2708 driver, but not with the new spi_bcm2835.

The spi master driver has a variable: bits_per_word_mask, that tells what the driver supports. But this is quite new so all drivers does not set this.

I have done some work on redesigning fbtft and there I use this variable in adition to an override module argument: https://github.com/notro/fbdbi/blob/master/core/lcdreg-spi.c

Development happens in the Linux kernel now: https://github.com/notro/fbtft/wiki/Development

irfansehic, 2015-08-03:
So you are already a step ahead of me. :+1:

closed 2015-08-03 by irfansehic

notro, 2015-08-03:
It will take some time before this work enters the kernel and even even more time before the drivers switch to using the new io path. So don't hold your breath :-)